tuya-cloudcutter/tuya-cloudcutter #143: Generic RGBCCW CB2L Bulb (Daybetter Brand)

W-Floyd, 2022-09-16:
Dumped a paired bulb. Can also dump a stock (unpaired) bulb if it would help. Note that the stock bulb works with the `Tuya-Generic/E27-RGBCW-Smart-Life-WB2L_M1` profile, while the paired bulb does not (SSL errors).

[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9584150/dump.zip)

[image]
[image]
[image]
[image]

openshwprojects, 2022-09-16:
"while the paired bulb does not (SSL errors)." this is interesting, does it mean that the firmware is remotely updated to the one without the exploit?

W-Floyd, 2022-09-16:
That's what I'm fearing. Perhaps I have mixed up batches of devices, but I really do not believe I have.

kuba2k2, 2022-09-17:
Not necessarily "without the exploit". It's enough if it updates to a different build of the same firmware, and the profile stops working already. @openshwprojects

W-Floyd, 2022-09-25:
Turns out there were two variants of this bulb, most likely depending on if it's a 12 pack or lesser quantity.

For those that identify as V2.9.16 once paired, these use a WB2L, and the Phopollo SBE2606 profile works.

For those that identify as V1.2.16 once paired, these use a CB2L, and the Tuya-Generic E27-RGBCW-Smart-Life-WB2L_M1 profile works.

Thanks to @Cossid and @kuba2k2 for helping me figure this out!

closed 2022-09-25 by W-Floyd