tuya-cloudcutter/tuya-cloudcutter #145: 3000273-970717 supported but no profile?

pkoretic, 2022-09-25:
Hi all.

3000273-970717 is listed in supported devices, however, when I tried to flash it I get the following error:

`Could not load profile /work/device-profiles/LSC/3000273-970717/profile. Are you sure the profile directory and file exist?`

And indeed, there is no  profile file in the directory. I see from the git history that it was never introduced. Is it possible to do this somehow without the profile?

I have flashed 3 other bulb types (3001686-970709,3000272-970716, 2578539-970719) without problem (including OTA for OpenBK7231T) and I tried to use those profiles as they are similar enough but that always results in DECRYPTION_FAILED_OR_BAD_RECORD_MAC.

Let me know what I'm doing wrong or if I can provide more information.

Thanks!

Cossid, 2022-10-15:
It looks like the profile is probably the same as the Arlec ALD092CHA

```
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQWWmCyJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QkJCQh3VCgAd1QoAHdUKAB3VCgAd1QoAHdUKAB3VCgAd1QoAHdUKAA=="
}
```

pkoretic, 2022-10-15:
Thanks @Cossid, let me try it out and report back.

ReanuKeeves01, 2022-10-28:
@pkoretic i too have a 3000273-970717 device - i managed to flash it by adding the profile as mentioned above. It's running OpenBK7231T (version 1.14.71) - so far so good. 

I just cannot get the bulb to do anything. I tried some default pin out configs for the 3000272 (seems super similar) and might have the same pin layout, but to no avail. 

Did you ever manage to get this device up and running? Mind giving me some advice on the right config in OpenBK in case you DID get it working? 

thanks !

ferbulous, 2022-10-28:
@ReanuKeeves01 is there an LED driver on the bulb? Since the usual PWN pin’s not working for you

ReanuKeeves01, 2022-10-28:
@ferbulous I'm honestly not entirely sure, i haven't opened the bulb yet. 

I did however get some good tips from someone on the Elektroda forum, that helped me in finding the correct channels/pins 

First, the correct pins were:
[image]

How i figured it out, was by putting all the possible PWM pins (6,7,8,9,24 and 26) on REL (Relay) and gave them value 0,1,2,3,4 and 5. 

Reference is on the config page:
"BK7231N/BK7231T supports PWM only on pins 6, 7, 8, 9, 24 and 26!"

The above gave me 6 buttons/toggles on the homepage, of which 2 worked. For me P7 and P8, i then noticed the values were all reversed (color temperature)... so i ended up with the flipped values as per screenshot above.

ferbulous, 2022-10-28:
@ReanuKeeves01 so only 2 pins out of 6?
I take it that’s not RGBCW bulb (only CW) since you only get warm and cold light?

ReanuKeeves01, 2022-10-28:
@ferbulous that is correct. only CW.

ReanuKeeves01, 2022-10-28:
@pkoretic You mentioned that you also flashed 3001686-970709 bulbs. 
which should be this one: 
[image]

What did you flash it with? Or was it perhaps one of the older versions still with an ESP chip?? I already (seemingly) bricked one of these trying to flash OpenBK on it...

ferbulous, 2022-10-28:
@ReanuKeeves01 ok, so you’ve got all the pins needed. But you still can’t control anything from the bulbs? That’s odd

For the (seemingly) bricked led, does it still light up at least? Otherwise you might’ve flashed the wrong firmware type for the device

pkoretic, 2022-10-28:
@ReanuKeeves01 indeed I flashed it in the same way with same firmware (OpenBK7231T_UG_1.12.60.bin, upgraded now to latest OpenBK7231T_UG_1.14.72.bin). 
Those were all bought maybe a year ago so probably they have older chips.
Only ones I couldn't manage to flash / cut from the cloud are 3001702-970727 and 3001700-970739.

ReanuKeeves01, 2022-10-28:
@ferbulous no actually I did eventually solve it. The 3000273-970717 bulb is working as intended now. The help from the forum solved it for me.

For the 2nd bulb, which is the 3001686-970709 type; No it doesn't come on at all. But i think i might have flashed the  OpenBK7231N-variant on that. So most likely that one is a goner seeing that @pkoretic confirmed he used a OpenBK7231T.

Thanks all, I have two more 3001686-970709 brand new in a box. I'm gonna give it another go tonight with the correct firmware variation.

ReanuKeeves01, 2022-10-28:
@pkoretic Do you recall if you used the existing device profile for the 3001686-970709 or did you tweak anything? 
I'm getting a seemingly successful flash, no errors. Yet the procedure doesn't seem to have succeeded as no AP is created by the bulb. It seems to remain with the Original Firmware as i can still operate it by hand and put it in pairing or ap mode. So at least it doesn't seem bricked this time.

pkoretic, 2022-10-28:
@ReanuKeeves01 for some bulbs I had to repeat process 2 times. It said it will reset in 30 seconds flashed but nothing happened even after few minutes. So I repeated the proces and then it worked. I think I never had to do it more than 2x times though.

ReanuKeeves01, 2022-10-28:
@pkoretic alright, let me try that. It does exactly that, but if i recall correctly from the succesful one earlier today, the fireware sending took longer and mentioned several of those 'Device will download and reset' .... will let you know how it goes!

ReanuKeeves01, 2022-10-28:
@pkoretic are these the PIN settings you're using?

[image]

I've tried these two channels and can config they are the right ones. Yet once i set them up as above it won't work anymore. Setting the above channels on REL allows me to toggle it on for either yellow/white.

ReanuKeeves01, 2022-10-30:
@pkoretic just realized i never followed up. Perhaps it could be useful for a future reader. 

After 3 or 4 tries, I noticed the script would always halt at the exact same moment. What worked for me, was copying out the file(s) from the `configured_devices` folder and then emptying the folder out. Kinda forcing the script to see the device 'as new'. After this, it flashed correctly first try.

I have a feeling, after running it numerous times and while it never gracefully exited, certain stuff just doesn't get stopped or halted properly, causing things to go wonky on next runs so i went one step further to avoid this with future bulbs.

For my personal workflow (i use Proxmox locally) I now have a prepared container template that has everything needed for cloudcutter to run pre-installed and setup. I found that spinning up a fresh container for every new device really sped up the process and only takes me seconds to do.

notkmhn, 2022-10-30:
Closed automatically by the PR merge, but seems like the issue is "resolved"? Please let me know if a re-open is needed!

commit df52c21e (2022-10-30): Add LSC 3000273-970717 profile - closes #145

Same as Arlec ALD092CHA
commit f6aa1c97 (2022-10-30): Merge pull request #165 from pkoretic/main

Add LSC 3000273-970717 profile - closes #145
closed 2022-10-30 by notkmhn