tuya-cloudcutter/tuya-cloudcutter #149: Calex Smart Power Plug (UK) 900011.1 v1.1 - dump attached

btsimonh, 2022-10-10:
Presenting the UK calex power plug (tesco £8 on clubcard).
A profile would be great for future purchasers!

[image]

[image]

Full dump:
[calexstrip.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9748157/calexstrip.zip)

Cossid, 2022-10-15:
This dump looks incomplete, probably from hid tools?  Please dump with [bk7231tools](https://github.com/tuya-cloudcutter/bk7231tools) and also please tell us which tuya module (wb#x, cb#s) is inside.

openshwprojects, 2022-10-23:
@Cossid Hid tools and bkWriter has nothing to do with "incomplete dumps", I even created profile that way.

One just has to remember to read full 2MB and that's it. As per that filesystem util, one just has to change the default offset because bkWriter reads from 0x11000 by default.

Also preferably pair before dumping, so the schema is in the profile.

kuba2k2, 2022-10-23:
Still, the much preferred way is to use bk7231tools as it's currently the best tool for that.

btsimonh, 2022-10-23:
it's everything apart from bootloader... i.e. starts at 0x10000
The chip is wb2s.

openshwprojects, 2022-10-23:
@kuba2k2 as far as I know, those tools are still commandline only and for me it's faster to work with bkWriter 1.60. "There is no point in reinventing the wheel", did you forget? 

@btsimonh  I know, they just have to shift the offset by 0x10000 at that python util, how it was called, decrypt_settings.py or something instead of using it blindly

kuba2k2, 2022-10-23:
> as far as I know, those tools are still commandline only and for me it's faster to work with bkWriter 1.60.

Yes, they are CLI only. BkWriter is extremely buggy, fails to read randomly, can't read/write specific offsets, or just doesn't link at all (fails at start). `bk7231tools` just works as it should. 

> There is no point in reinventing the wheel

hey, the entire Tuya ecosystem also works and is easier to use, why bother with custom firmwares at all?

> I know, they just have to shift the offset by 0x10000 at that python util, how it was called, decrypt_settings.py or something instead of using it blindly

(It's 0x11000, by the way) My tool can work with bootloader-less binaries. It's worse if binaries have BL/APP swapped, like the ones you upload, but the tool can also fix that. Also, people should read on how to produce full, usable dumps, *instead of using the defaults blindly.*

> since you know Python, have you considered doing a more automatic Haxomatic? Or are you on it right now? It would be indeed helpful.

I am working on something bigger right now - eliminating the need of profiles entirely.

openshwprojects, 2022-10-23:
That's strange, for me bkWriter 1.60 works ok and I had no troubles with decrypting it's binaries.

So, what kind of approach are you planning to take with to do it without profiles? Another vulnerability? I hope at least it's a fair work and not the deal that you just chimed in at the right time by following someone else trail.

Or define - "the need of profiles" - are you saying that OTA will work for device that was never dumped before (and has not known yet build)?

notkmhn, 2022-10-23:
@openshwprojects this is your first and final warning, snarky comments will absolutely not be tolerated. You're welcome to interact with people on issues on this repository and any other cloudcutter repositories in a civil manner but otherwise find yourself another way to moan at people that you disagree with. Better yet, stop moaning already - everyone is free to do whatever the hell they want.

kuba2k2, 2022-10-23:
> So, what kind of approach are you planning to take with to do it without profiles? Another vulnerability?

Kind of - the possibility of remote code execution on the chip and dumping/writing firmware remotely.

> I hope at least it's a fair work and not the deal that you just chimed in at the right time by following someone else trail.

Haha, funny. It really was a kind of teamwork of myself and @notkmhn - his expertise and insight in lower-level CPU's inner workings, and my idea of payload contents and calling the right functions.

> Or define - "the need of profiles" - are you saying that OTA will work for device that was never dumped before (and has not known yet build)?

Correct, that's the plan.

openshwprojects, 2022-10-23:
@notkmhn  Ah, ok, I will refrain from posting here and will just keep sending profiles only. For the sake of transparency, I will just quote here the message I send you, because people here might not have the full context and backstory about the @kuba2k2 extremely disrespectful and deceitful behaviour that goes against open source spirit:
[image]
so next time you do something with him, be aware about the backstabbing possibility

@kuba2k2 wow, that's a good idea, remember to also look into RTL chips which have the same vulnerability. Was any more work done in that field?

kuba2k2, 2022-10-23:
> wow, that's a good idea, remember to also look into RTL chips which have the same vulnerability. Was any more work done in that field?

They probably do. It should be even easier on RTL because they supposedly can execute code directly from RAM. Currently, I have an Android app that can connect to BK7231T (no N support yet), exploit it, and dump the entire flash contents.

Cossid, 2022-12-01:
The included dump has no storage data (dump should be 2048KB, not 1980KB), and therefore a schema cannot be generated.

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid