tuya-cloudcutter/tuya-cloudcutter #151: Treatlife DS02S - WB3S

peddamat, 2022-10-14:
Hi,

Attached is a firmware dump of the Treatlife DS02S 3-way Smart Dimmer.

I tried OTA exploiting the device using the DS02F profile, but it didn’t work, only realizing the DS02F is for a fan controller, despite the naming convention.

Thank you!

Best

Sam
[treatlife-ds02s.bin.gz](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9790645/treatlife-ds02s.bin.gz)

Cossid, 2022-10-15:
It looks like using the DS02F profile, you should at least still be getting an A- AP, is that correct?
Is your intent to continue to use the stock Tuya firmware, or replace it with an alternative?

In order to use the stock Tuya firmware, a dump would be needed after you already join it to Smart Life (note, this would include the ssid and password of network you connect to, so create a throw-away guest network with a dummy password and use that).

If you plan to use an alternative firmware, you shouldn't need to do anything, just use the DS02F profile and flash your desired firmware.

peddamat, 2022-10-15:
That’s correct, I do get an A- AP, so the exploit worked in that respect.

I do intend to flash with libretuya’s ESPHome, but will go ahead and provide a dump for folks who want to use the stock firmware.

The issue I’m having is that the device isn’t triggering a firmware update after the exploit and reconnect to the cloudcutterflash SSID.

All I see is the DHCP output, and the device queries for h3-iot-dns.com a few times.

I’ll provide a full log output.  I was thinking of running tcpdump to see what’s happening over the wire.

peddamat, 2022-10-15:
```
pi@raspberrypi:~/tuya-cloudcutter $ sudo ./run_flash.sh wlan0 "TreatLife/DS02F" "bk7231t_app.ota.ug.bin"
Building cloudcutter docker image..
Sending build context to Docker daemon  36.59MB
Step 1/10 : FROM navikey/raspbian-bullseye:latest AS base
 ---> 2760a3d8ce38
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto python3-pip libpython3.9-dev
 ---> Using cache
 ---> ea3cd704ac61
Step 3/10 : FROM base AS python-deps
 ---> ea3cd704ac61
Step 4/10 : RUN pip3 install --upgrade pipenv
 ---> Using cache
 ---> 4d207b028594
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 31a9286d722c
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> f46d3b098a71
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> f7747cd47f17
Step 8/10 : FROM python-deps AS cloudcutter
 ---> f7747cd47f17
Step 9/10 : COPY src /src
 ---> adf2dbaef07f
Step 10/10 : WORKDIR /src
 ---> Running in 5d7c999115a3
Removing intermediate container 5d7c999115a3
 ---> 404997854c27
Successfully built 404997854c27
Successfully tagged cloudcutter:latest
Successfully built docker image
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "TreatLife-SL-B90E", trying to connect..
Device 'wlan0' successfully activated with '468bab0b-495e-427d-a4c5-8ae48b0e0cb5'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Exploit run, saved device config to!
output=/work/configured-devices/WHjB9yc8GMG9.deviceconfig
Saved device config in /work/configured-devices/WHjB9yc8GMG9.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
[... cut ...]
Oct 14 03:33:50 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Oct 14 03:33:50 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Oct 14 03:33:51 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Oct 14 03:33:51 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Oct 14 03:33:52 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Oct 14 03:33:52 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Oct 14 03:33:54 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Oct 14 03:33:54 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
```

Cossid, 2022-10-15:
That is strange, as there appear to be many DHCP requests.  Have you tried rebooting the device running cloudcutter?  Sometimes processes get stuck in the background and cause trouble.

You also appear to have more than one device attempting to connect to the cloudcutter AP, I would not recommend doing more than one at a time.

peddamat, 2022-10-15:
Yes, rerunning after reboot didn’t solve the issue.  Also, I wasn’t sure how to get the device to connect to the ‘cloudcutterflash’ ssid using the SmartLife app, as that ssid isn’t connected to the internet.

As a workaround, I created a temporary ‘cloudcutterflash’ wifi network and used that, but to no avail.

The multiple device connects was due to my iPhone connecting to the ssid after aforementioned process.

I’ll give it a few more tries and use tcpdump, as well.

Glad to know that the DS02F profile should have worked, helps me narrow things down.

Cossid, 2022-10-15:
No, you don't create a separate 'cloudcutterflash' network, the cloudcutter script does that for you.  It starts an AP with network-manager and hostapd.  If the device doesn't connect to the AP started by cloudcutter, it doesn't continue on.

peddamat, 2022-10-15:
Yup, in understand, i worded things poorly.

After running the initial exploit and having things hang after the dhcp offer, the instructions say to reboot the device and use a SmartLife-compatible application to get it to reconnect to the cloudcutterflash AP.

I couldn’t accomplish this because the SmartLife and Tuya iOS apps wouldn’t let me pair the devices to an SSID that didn’t have internet access.

peddamat, 2022-10-15:
Humble request, do you know how I can decrypt the TLS traffic that I captured with tcpdump?  

Since my raspberry pi is communicating with the Treatlife device over TLS, I’m assuming that we have everything necessary, but can’t figure out what to feed Wireshark.

Cossid, 2022-10-15:
Sorry, I do not know how to do that.

kuba2k2, 2022-10-15:
> Humble request, do you know how I can decrypt the TLS traffic that I captured with tcpdump?

The TLS connection uses the PSK which should have been written to a `.deviceconfig` file. I don't know specifically how to do the decryption, but it may be possible that Wireshark can do this, given the PSK and having the handshake packets.

kuba2k2, 2022-10-15:
By the way: this dump fits the `bk7231t-common-user-config-ty-1.1.71-sdk-1.0.2-40.00` profile, which is the same for the following devices:
- TreatLife DS03 Fr fan 15h 484
- Mlambert 49Ft Outdoor RGB
- MartinJerry US FC 01
- TreatLife DS02F

You can try these other devices and see if it downloads the upgrade.

I'd also appreciate testing [my PR](https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/148) which simplifies device/profile selection a bit. You can download it by using `git clone https://github.com/kuba2k2/tuya-cloudcutter.git -b feature/profiles-api`.

closed 2022-11-30 by Cossid