tuya-cloudcutter/tuya-cloudcutter #152: [Profile request] Immax NEO E14 RGBCCT bulb (Czech rebrand)

leoshusar, 2022-10-15:
Hi, about 2 months ago I got some Immax RGBCCT bulbs and I've been using them with Tuya. I tried to ask the rebrander which chip it uses and they told me CB2L - great, I don't have to open it before trying some device profiles.
Yesterday I got another one for testing and finally tried it. No profiles worked but I believe it's exploitable since some profiles crashed it (it stopped blinking).

So I opened it (it was quite easy, rotated and popped it) and to my surprise it uses [CBLC5](https://developer.tuya.com/en/docs/iot/cblc5-module-datasheet?id=Ka07iqyusq1wm) which I haven't seen yet, it only has 6 pins in total and several test pins. I guess they've changed it recently or the manufacturer lied to me :) but I don't think so, he was very helpful.

I had to desolder it so I can access UART test pins but I managed to dump the firmware.
(photos taken after desoldering)

[image]
(Tuya chip goes to those left holes)
[image]
[image]
[image]

FW: [led_tuya.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9791868/led_tuya.zip)

If this one works well, I'll try my older ones with the same profile and see if they really changed something in it.

OT: how do I put it back? :D Do I just insert the wire under that thing on the right so it just touches the screw from inside?
[image]

leoshusar, 2022-10-15:
The reseller wrote he asked the manufacturer directly, so I don't know, maybe they also don't know.

That's unfortunate; I checked my other bulbs, now connected to cloud, and they all have 1.3.21 without new update available, so I guess Tuya doesn't really care about fixing the exploit? Or is it up to the reseller to update those? But I sure won't be updating if there appears one.

Anyway thank you! Hopefully someone will look at this soon.

Cossid, 2022-10-15:
Generally speaking, they don't patch existing products, they'll only make the fix moving forward for new product lines or if a major hardware revision is needed to an existing product.

ferbulous, 2022-10-18:
> The reseller wrote he asked the manufacturer directly, so I don't know, maybe they also don't know.
>
> That's unfortunate; I checked my other bulbs, now connected to cloud, and they all have 1.3.21 [...]

Is there a version that officially blocks the exploits?

Cossid, 2022-10-18:
> Is there a version that officially blocks the exploits?

None have been observed yet.  I suspect we might possibly see the first in the coming months on new devices though.

ferbulous, 2022-10-18:
> None have been observed yet. I suspect we might possibly see the first in the coming months on new devices though.

I see, at least that confirms that some of my newer tuya devices just don’t have compatible profiles yet

closed 2022-11-30 by Cossid