tuya-cloudcutter/tuya-cloudcutter #154: [Profile request] Aubess 20A wifi 2.4Ghz plug

ziporah, 2022-10-19:
Would it be possible to create a profile for the Aubess 20A plug?
Attached is the original firmware and the paired firmware ssid = hotspot, pass = abcdabcd
Pictures of the device:
[image]
[image]
[image]
[image]
[image]
The device has a CB2S module inside. Cloudcutter works untill the final part where it tries to reconnect

This is the haxomatic output, I noticed in the code there is a section <not working for bk7231N> and that's also where it fails:
```
jo@meegopad:/net/samba/jo/tasmota/tuya/cloudcutter-bk7231-haxomatic$ pipenv run python haxomatic.py ../aubess-paired/aubess-20a-paired.bin_app_1.00_decrypted.bin
[!] Loading and disassembling code - may take a moment
[+] Code loaded!
[!] Searching for post-vuln code patterns
[+] Found a post-vuln code pattern match!
[+] Matched instructions:
        0xb625c: adds r5, #0xfc
        0xb625e: ldr r3, [r5, #0x50]
        0xb6260: ldr r0, [sp, #0xc]
        0xb6262: ldr r1, [pc, #0x48]
[+] Identified lan object register as r5
[!] Searching for JSON object register
[+] Identified JSON object register as r4
[!] Searching for ty_cJSON_Parse function address
[+] ty_cJSON_Parse address: 0xa0604
[!] Searching for mf_cmd_process gadget address
Traceback (most recent call last):
  File "haxomatic.py", line 243, in <module>
    walk_app_code(appcode)
  File "haxomatic.py", line 209, in walk_app_code
    raise RuntimeError("Failed to find the mf_cmd_process gadget address")
RuntimeError: Failed to find the mf_cmd_process gadget address
```

The cloudcutter output:
```
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-0A2B", trying to connect..
Device 'wlp2s0' successfully activated with 'cff1576b-89af-49e9-838e-6d5a38932559'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Exploit run, saved device config to!
output=/work/configured-devices/IlZD2lv3uygI.deviceconfig
Saved device config in /work/configured-devices/IlZD2lv3uygI.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
....
<snip>
...
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 tags: wlp2s0
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 DHCPACK(wlp2s0) 10.42.42.35 fc:67:1f:3d:0a:2b wlan0
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 next server: 10.42.42.1
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  1 option: 53 message-type  5
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 51 lease-time  12h
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 58 T1  6h
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 59 T2  10h30m
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Oct 18 10:12:30 dnsmasq-dhcp[16]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Oct 18 10:12:51 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.35
Oct 18 10:12:51 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:51 iostream:1406] SSL Error on 12 ('10.42.42.35', 59751): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:52 iostream:1406] SSL Error on 12 ('10.42.42.35', 59752): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:53 iostream:1406] SSL Error on 12 ('10.42.42.35', 59753): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:54 iostream:1406] SSL Error on 12 ('10.42.42.35', 59754): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:55 iostream:1406] SSL Error on 12 ('10.42.42.35', 59755): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:56 iostream:1406] SSL Error on 12 ('10.42.42.35', 59756): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221018 10:12:57 iostream:1406] SSL Error on 12 ('10.42.42.35', 59757): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
```
From what I understood this SSL Error indicates a wrong profile. I used the Tuya-Generic/LSPA9-Smart-Plug-BL0937 profile

[aubess.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9818151/aubess.zip)
layout of the zipfile:
```
./aubess-clean => decrypted files from clean bin file
./aubess-paired => decrypted files from paired bin file
./aubess-20a-paired.bin => extracted with hid-downloader, throwed ctc error in the end
./aubess-20a-paired-bk7231tools.bin => extracted with bk7231tools
./aubess-20a.bin => extracted with hid-downloader, throwed crc error in the end
```

Cossid, 2022-10-19:
The Tuya-Generic LSPA9-Smart-Plug-BL0937 profile should work for this device.  Unfortunately, you snipped out a bit that might better indicate what is going on.  Can you please provide the full cloudcutter log?

ziporah, 2022-10-19:
The full log without my ssid and password
```
Using  and
Building cloudcutter docker image..
Sending build context to Docker daemon  36.08MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> ab00e24b64b6
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> cccf5bf44b57
Step 3/10 : FROM base AS python-deps
 ---> cccf5bf44b57
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 2eaf37e118f2
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 642fc0160de1
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 2a92829041d5
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 0f54c4bcdf4f
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 0f54c4bcdf4f
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 4f9e9362ae24
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 906c5394e9ed
Successfully built 906c5394e9ed
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select the brand of your device: Tuya-Generic
   LEDVANCE
   Lightway
   Lonsonho
   LSC
   MartinJerry
   MINCO_HEAT
   Mlambert
   Mycket
   PHOPOLLO
   QIACHIP
   STITCH
   TreatLife
 > Tuya-Generic

[?] Select the article number of your device: LSPA9-Smart-Plug-BL0937
   E27-A60-RGBCWW-V1
   E27-RGBCW-Smart-Life-WB2L_M1
 > LSPA9-Smart-Plug-BL0937
   WiFi-Curtain-Switch-WF-CS01
   WiFi-Switch-2-Gang
   WiFi-Switch-3-Gang-CB3S
   WiFi-Switch-4-Gang

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-0A2B", trying to connect..
Device 'wlp2s0' successfully activated with 'cff1576b-89af-49e9-838e-6d5a38932559'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Exploit run, saved device config to!
output=/work/configured-devices/wPfdHc7W6vzP.deviceconfig
[... 72 lines cut ...]
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Oct 19 17:45:55 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Oct 19 17:46:31 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.35
Oct 19 17:46:31 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:31 iostream:1406] SSL Error on 12 ('10.42.42.35', 59751): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:32 iostream:1406] SSL Error on 12 ('10.42.42.35', 59752): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:33 iostream:1406] SSL Error on 12 ('10.42.42.35', 59753): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:34 iostream:1406] SSL Error on 12 ('10.42.42.35', 59754): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:36 iostream:1406] SSL Error on 12 ('10.42.42.35', 59755): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:37 iostream:1406] SSL Error on 12 ('10.42.42.35', 59756): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:38 iostream:1406] SSL Error on 12 ('10.42.42.35', 59757): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221019 17:46:39 iostream:1406] SSL Error on 12 ('10.42.42.35', 59758): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
```

ziporah, 2022-10-19:
And the generated file:
```
 cat configured-devices/wPfdHc7W6vzP.deviceconfig
{"uuid": "wPfdHc7W6vzP", "auth_key": "QUUZK1gcEw1L0Mbm", "local_key": "v8SwV97ukAoMCw08", "sec_key": "m4OWXPJEufvi1mWf", "device_id": "Avdrh2TDADAvJyQklpEa", "chip_family": "BK7231T"}
```
The chip family is wrong, it states BK7231T, while there is a BK7231N device in the plug

Cossid, 2022-10-19:
Ah, unfortunately, I was mixing up the LSPA7 and LSPA9.  Your plug matches the LSPA7, not the 9.  Unfortunately, there is no profile for the LSPA7 yet either.

There is currently no working profile for this device.

ziporah, 2022-10-19:
indeed, I couldn't upload a tgz or tar.gz, so I renamed it to .zip
I have some more simular devices, so I can pull more images if that helps in identifying some stuff. I am also able to connect the rx2/tx2 if that helps

Cossid, 2022-10-19:
The firmware images and the module type (CB2S for example) are the most useful pieces of information.  Feel free to create an issue for each unique device/model.

ziporah, 2022-10-19:
I have 6 devices of the same device/model. I meant to say I can pull more firmware images from these devices if that in some way helps. Are there other developers in this project that are able to step in or should I ask on the elektroda forum ?

Cossid, 2022-10-19:
I'm not sure how much help the elektroda forum will be specifically for cloudcutter profiles, they usually open tickets up here.  There are other devs (who wrote the tools) that are more capable, but it does appear there hasn't been a focus on profile creating as of late, so I can't speak to when there will be action.  All you can do is wait and see I think.

You are also free to serial flash the devices directly with 3rd party firmwares and skip cloudcutter altogether.

ziporah, 2022-10-19:
I was planning on flashing the ones I have opened, while I have them connected. 
It's just easier while I'm in the process of opening them, a profile can be created so next time I won't have to open all of them and can just use cloudcutter. I also have a bunch of tywe2s for replacing the cb2s if I screw up on a device.

ziporah, 2022-10-24:
hey @notkmhn any chance you could assist on creating a profile? I looked at your video explaining the exploit and tried using Ghidra to open the bin file with the xtensa plugin, but I must be missing something because I can't seem to decompile any of the code or determine any functions.

kuba2k2, 2022-10-24:
BK7231 is not xtensa, it's ARM.

ziporah, 2022-10-28:
I tried decompiling with arm, that does work better indeed. Any idea on what specific version should be best? I discovered the cpuset is armv5te, but the little endian ones available for decompiling are armv5 and armv5t. 
Is there some pointer to get more info or what to search for in order to create the profile? I have a tuya account and can register devices there to get more debugging information and id's directly from the cloud. I have used it before to register devices with localtuya. I will also go trough the full presentation of mch2022 again https://rb9.nl/posts/2022-08-28-mch2022-smart-light-hacking-journey/  hoping to find some more pointers there.

kuba2k2, 2022-10-28:
Any ARM instruction set will probably work, it's best to just use the default and let the decompiler choose.

christi-ko, 2022-10-28:
Just jumping into this to keep up to date. Also have 6 of those plugs around (but the 16A version) with same point stuck. Hoping we get a working profile soon. 
If i can support, please let me know ...

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

ziporah, 2022-12-01:
Thanks a lot ! The cloudcutter works. Tested on 3 devices and they all were correctly connected to my local wifi.

christi-ko, 2022-12-02:
It also worked for me ...
Thank you very much. This is a great tool.