tuya-cloudcutter/tuya-cloudcutter #155: WB3S dimmer 1 channel TY-02-1CH

openshwprojects, 2022-10-20:
That device:
[image]
[image]
[image]
[image]
[image]
[wb3s_dimmer_singlechannel_ TY-02-1CH_paired.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9830296/wb3s_dimmer_singlechannel_.TY-02-1CH_paired.zip)

kuba2k2, 2022-10-23:
These dumps are "invalid" for one reason: APP starts at byte 0x0 and bootloader is placed by the end of the file. I don't know what tool you're using to dump firmware, but I'd recommend fixing it so that it reads from 2MB starting at 0x0, and not starting at 0x11000.

kuba2k2, 2022-10-23:
Also, these dumps match the `oem-bk7231s-strip-ty-1.1.6-sdk-1.0.2-40.00` profile, which is compatible with LSC 3004154-970742 profile. Try this and let us know if it works.

btsimonh, 2022-10-23:
It does sound like the wrong dump, does it not?  Especially with the name 'strip'.  I'll search to see if I have the correct dump...

kuba2k2, 2022-10-23:
It's an RGB controller, so it's not unusual for the firmware to be used for LED strips as well. The attached dump was verified to CRC-match the profile I mentioned.

closed 2022-11-30 by Cossid