tuya-cloudcutter/tuya-cloudcutter #156: Device still has original firmware after trying to flash

BlackDragon-B, 2022-10-22:
I am trying to flash OpenBeken to a LSC Moodlight (970742) but after doing so and the script stopping the original firmware still is present on the device (and I can notice that by the ability of going into AP mode and not seeing the OpenBeken AP)
I've also included the log

```
sudo ./run_flash.sh "wlan0" "LSC/3004154-9                                                                                                                                                             70742" "OpenBK7231T_UG_1.12.30.bin"
Building cloudcutter docker image..
Sending build context to Docker daemon   37.8MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 6f3c3a33a982
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommend                                                                                                                                                             s git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 69a30826d278
Step 3/10 : FROM base AS python-deps
 ---> 69a30826d278
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> d0899cabcbc5
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> cf8c0833c2e5
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 9249791bb3a6
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 127d5b5d6844
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 127d5b5d6844
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 5383bd982307
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 45451e3ab7c3
Successfully built 45451e3ab7c3
Successfully tagged cloudcutter:latest
Successfully built docker image
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in b                                                                                                                                                             etween, to enable AP mode. Repeat if your device's SSID doesn't show up within ~                                                                                                                                                             30 seconds. For smart plugs long press the reset button on the device for about                                                                                                                                                              5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more i                                                                                                                                                             nformation.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home                                                                                                                                                             -" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "A-D8B1", trying to connect..
Device 'wlan0' successfully activated with '400761c8-0d58-4144-9298-5de29e649650                                                                                                                                                             '.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Exploit run, saved device config to!
output=/work/configured-devices/h3r3WL9T0U5e.deviceconfig
Saved device config in /work/configured-devices/h3r3WL9T0U5e.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's                                                                                                                                                              SSID doesn't show up within ~5 seconds. For smart plugs long press the reset but                                                                                                                                                             ton on the device for about 5 seconds. See https://support.tuya.com/en/help/_det                                                                                                                                                             ail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home                                                                                                                                                             -" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "A-D8B1", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home                                                                                                                                                             -" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "A-D8B1", trying to connect..
Device 'wlan0' successfully activated with '400761c8-0d58-4144-9298-5de29e649650                                                                                                                                                             '.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP i                                                                                                                                                             s 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This                                                                                                                                                              script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Oct 22 05:44:02 dnsmasq[15]: started, version 2.80 cachesize 150
Oct 22 05:44:02 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Oct 22 05:44:02 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Oct 22 05:44:02 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Oct 22 05:44:02 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:29:08:71 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
[... 207 lines cut ...]
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1666417486, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}, 't': 1666417486, 'success': True}
[I 221022 05:44:46 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417486&v=1.0&sign=06cd754e019e5289363fd5a7b70c197c (10.42.42.38) 167.89ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.ack&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417486&v=1.0&sign=62133e46d6ae5860e7c6417885417995
[LOG] ==== Client request body ===
{'ackId': '0-0', 't': 1666417486}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.ack - response: {'success': True, 't': 1666417489}
[I 221022 05:44:49 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.ack&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417486&v=1.0&sign=62133e46d6ae5860e7c6417885417995 (10.42.42.38) 104.52ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.silent.get&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417489&v=4.3&sign=26ed452be53276e5b00422024fc88fd9
[LOG] ==== Client request body ===
{'subId': None, 't': 1666417489}
[LOG] ==== End client request body ===
tuya.device.upgrade.silent.get - response: {'result': {'url': 'http://10.42.42.1:80/files/OpenBK7231T_UG_1.12.30.bin', 'hmac': 'D60D7A39A99B87FC92819FFBD9A5BB52A394731E9E804E14F806A177DA24C748', 'version': '9.0.0', 'size': '385680', 'type': 0}, 'success': True, 't': 1666417490}
[I 221022 05:44:50 web:2239] 200 POST /d.json?a=tuya.device.upgrade.silent.get&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417489&v=4.3&sign=26ed452be53276e5b00422024fc88fd9 (10.42.42.38) 129.52ms
[... cut ...]
tuya.device.timer.count - response: {'result': {'devId': 'GdEiJdKmaHaRyLYpg666', 'count': 0, 'lastFetchTime': 0}, 't': 1666417490, 'success': True}
[I 221022 05:44:50 web:2239] 200 POST /d.json?a=tuya.device.timer.count&devId=GdEiJdKmaHaRyLYpg666&et=1&t=1666417490&v=4.0&sign=454b61a6c207cdabb927f6101e80d2c5 (10.42.42.38) 105.73ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1666417491} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
```

botioni, 2022-10-29:
same here with a ledavance  bulb any news or i need to open it up and flash manualy

tony-fav, 2022-10-29:
I had a similar issue where the script shows it is telling the device to OTA but there's never a DEVICE_OTA log message. I'm not sure what the root issue was but I was successful with a fresh pi os and clean install of TCC.

botioni, 2022-10-29:
ok i will try thnks.

botioni, 2022-10-29:
tryd stele the same profile Ledadvance 1.1.4 firmware not changing

Using PSK v1 - Received PSK ID version 01
[I 221029 19:53:28 web:2239] 200 POST /v1/url_config (10.42.42.14) 623.42ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=82&uuid=x3U2EuWvcCPe&v=4.4&sign=2b6657dd25a3ded9d61dbd9703e9d352
[LOG] ==== Client request body ===
{'token': 'AAAAAA', 'softVer': '1.1.4', 'productKey': 'keyhxhxkrkkxcvt7', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'keyhxhxkrkkxcvt7', 'devId': 'ycSIG3QJdBOl7jiKvQiJ', 'hid': '7cf666162b07', 'modules': '[{"type":9,"softVer":"1.1.4","online":true}]', 'devAttribute': 515, 'cadVer': '1.0.2', 'cdVer': '1.0.0', 'options': '{"isFK":true}', 't': 82}
[LOG] ==== End client request body ===
tuya.device.active - response: {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":20},{"type":"obj","mode":"rw","property":{"range":["white","colour","scene","music"],"type":"enum"},"id":21},{"type":"obj","mode":"rw","property":{"min":10,"max":1000,"scale":0,"step":1,"type":"value"},"id":22},{"type":"obj","mode":"rw","property":{"min":0,"max":1000,"scale":0,"step":1,"type":"value"},"id":23},{"type":"obj","mode":"rw","property":{"type":"string","maxlen":255},"id":25},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":26},{"type":"obj","mode":"wr","property":{"type":"string","maxlen":255},"id":28},{"mode":"rw","id":30,"type":"raw"},{"mode":"rw","id":31,"type":"raw"},{"mode":"rw","id":32,"type":"raw"},{"mode":"rw","property":{"type":"bool"},"id":101,"type":"obj"}]', 'devId': 'WmZwMh9RGfNDnsPopo1u', 'resetFactory': False, 'timeZone': '+02:00', 'capability': 1025, 'secKey': 'wdHSiEyiYtuIgBqa', 'stdTimeZone': '+01:00', 'schemaId': '000002eher', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200], [1711846800, 1729990800], [1743296400, 1761440400], [1774746000, 1792890000]], 'localKey': 'zcIqflc0u6Kzsjs6'}, 't': 1667073209, 'success': True}
[I 221029 19:53:29 web:2239] 200 POST /d.json?a=tuya.device.active&et=1&t=82&uuid=x3U2EuWvcCPe&v=4.4&sign=2b6657dd25a3ded9d61dbd9703e9d352 (10.42.42.14) 29.56ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073209&v=1.0&sign=f4a180377b4f9cdf2cb758266fb64f0d
[LOG] ==== Client request body ===
{'t': 1667073209}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1667073210, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}, 't': 1667073210, 'success': True}
[I 221029 19:53:30 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073209&v=1.0&sign=f4a180377b4f9cdf2cb758266fb64f0d (10.42.42.14) 22.86ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.uuid.pskkey.get&et=1&t=1667073212&uuid=x3U2EuWvcCPe&v=1.0&sign=5fe3d70b6d0a45c3fcd5e9c2a2953814
[LOG] ==== Client request body ===
{'t': 1667073212}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1667073212, 'success': True}
[I 221029 19:53:33 web:2239] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1667073212&uuid=x3U2EuWvcCPe&v=1.0&sign=5fe3d70b6d0a45c3fcd5e9c2a2953814 (10.42.42.14) 12.19ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1667073215} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.get&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073213&v=4.3&sign=af241f266b76bea647b06288c9a4111e
[LOG] ==== Client request body ===
{'type': 0, 't': 1667073213}
[LOG] ==== End client request body ===
tuya.device.upgrade.get - response: {'result': {'url': 'http://10.42.42.1:80/files/OpenBK7231T_UG_1.14.85.bin', 'hmac': '18DF1FB98F5CD0B4885A6CAAA9500D59E84A261E717ABD0F3C9FEA5F3B35B932', 'version': '9.0.0', 'size': '448976', 'type': 0}, 'success': True, 't': 1667073216}
[I 221029 19:53:36 web:2239] 200 POST /d.json?a=tuya.device.upgrade.get&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073213&v=4.3&sign=af241f266b76bea647b06288c9a4111e (10.42.42.14) 38.32ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.status.update&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073216&v=4.1&sign=461c3ee449f64c4191e1d3179e4bcb88
[LOG] ==== Client request body ===
{'type': 0, 'upgradeStatus': 2, 't': 1667073216}
[LOG] ==== End client request body ===
tuya.device.upgrade.status.update - response: {'success': True, 't': 1667073216}
[I 221029 19:53:36 web:2239] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073216&v=4.1&sign=461c3ee449f64c4191e1d3179e4bcb88 (10.42.42.14) 10.43ms
[LOG] Client request PATH: /d.json - QS: a=atop.online.debug.log&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073220&sign=524c9546168a7bdc63b1dcaea39d9d9f
[LOG] ==== Client request body ===
{'data': '1', 't': 1667073220}
[LOG] ==== End client request body ===
atop.online.debug.log - response: {'result': True, 't': 1667073222, 'success': True}
[I 221029 19:53:42 web:2239] 200 POST /d.json?a=atop.online.debug.log&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073220&sign=524c9546168a7bdc63b1dcaea39d9d9f (10.42.42.14) 26.70ms
[... cut ...]
[I 221029 19:53:43 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.ack&devId=WmZwMh9RGfNDnsPopo1u&et=1&t=1667073222&v=1.0&sign=a161e02c7a8865943eb54bb1a1012d4b (10.42.42.14) 11.63ms
[I 221029 19:53:44 web:2239] 200 GET /files/OpenBK7231T_UG_1.14.85.bin (10.42.42.14) 7907.22ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-448975/448976
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1667073227} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.

Cossid, 2022-11-21:
Please try copying the [tuya.device.dynamic.config.get.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-profiles/CASENL/CA-K1-04-WiFi-4-Gang-Switch/tuya.device.dynamic.config.get.json) file from the CASENL profile and try again.

Cossid, 2022-12-01:
The profile for this has been updated and should now work correctly.

closed 2022-12-01 by Cossid