tuya-cloudcutter/tuya-cloudcutter #157: Nous P8 wrongly cut from the cloud and now I cannot do anything

stelizap99, 2022-10-22:
Since Nous is not on the list of device profiles, I used tuya-generic profile, e27-rgbcw-smart-life_wb2l_m1, even though the bulb is GU10 and it worked, the device was cut from the tuya cloud. But, later I realised that this profile that I used had the chip listed as BK7321N, while my bulb has the BK7321T inside. So naturally, before I tried to flash the bulb with openbeken, I tried to reuse the detaching script to put the correct profile on my bulb. But this didnt succeed after many tries, getting with 2 types of errors: padding is incorrect or address already in use. Not even flashing works, it blocks at the steps where the bulb is connecting to the cloudcutter AP.

kuba2k2, 2022-10-22:
You can't really "break" a device by using an incorrect cloudcutter profile (well, unless you manage to get it to flash wrong firmware).

A non-matching profile will make the device hang - and that's the worst case. If you used BK7231N profile on BK7231T, it most likely didn't actually work - cloudcutter doesn't (can't) check for success of the operation (unless you start firmware flashing, in which case it waits for the device to communicate with it).

"address already in use" might mean that cloudcutter is trying to run the HTTP/MQTT server, and some other software is blocking the port. Try restarting your device (laptop, a Pi, or whatever you're using).

stelizap99, 2022-10-22:
So the first time, when using the BK7231N profile, cloudcutter said the exploit worked and from then the device's ap was changed to A-..., which would imply that it worked. From then it was that the exploit didn't succeed again, I think not even with the original profile that worked the first time.

stelizap99, 2022-10-22:
I would upload logs, but it doesnt seem to be saving any.

stelizap99, 2022-10-22:
I also tried with an LSC GU10 bulb profile, which also has BK7231T and that also didnt work.

kuba2k2, 2022-10-22:
If it created the `A-` AP that means the exploit is compatible - so either the profile is for BK7231T, or your device actually has a BK7231N. It's not possible for one profile to be compatible with N and T at once.

stelizap99, 2022-10-22:
I thought the device has a T because i asked NOUS about it, maybe its a different version that the guy didnt have in his list idk...

kuba2k2, 2022-10-22:
Yeah the devices change unpredictably, the manufacturer may not know what they ship inside the products.

stelizap99, 2022-10-22:
Lol i think you are right. I actually tried the original profile with N once again and ... once again it worked!
If i get openbeken to flash too I will close this issue

stelizap99, 2022-10-22:
Openbeken flashed apparently succesfully so closing this:))

closed 2022-10-22 by stelizap99

blakadder, 2022-11-18:
Which firmware did you end up using? T or N?

blakadder, 2022-11-18:
hmmm, managed to get mine to the flashing part (used LibreTuya binary) but it didn't flash, its still in the A- SSID mode

kuba2k2, 2022-11-18:
Did you use the new profiles API PR branch? You can try using the master instead, or the new API instead. Problems like this also point towards having incorrect schema.

Cossid, 2022-11-18:
Just to clarify here, the board labeled "WB2L_M1" despite being named with a leading W (which normally indicates T) is actually a BK7231N chipset, not a T.

You should be flashing firmware designed for BK7231N on WB2l_M1 modules.