tuya-cloudcutter/tuya-cloudcutter #158: EDM-01AA-EU 300W incandescent dimmer smart switch WB3S with TuyaMCU [BK7231T]

openshwprojects, 2022-10-23:
in this switch, you have to cut the tracess to flash
[image]
[image]
[image]
[image]
[image]

[EDM-01AA-EU.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9846977/EDM-01AA-EU.zip)

kuba2k2, 2022-10-23:
This dump matches the `bk7231t-common-user-config-ty-1.1.71-sdk-1.0.2-40.00` profile, which can be used with one of the following devices:
- TreatLife DS03 Fr fan 15h 484
- Mlambert 49Ft Outdoor RGB
- MartinJerry US FC 01
- TreatLife DS02F

openshwprojects, 2022-10-23:
@kuba2k2 How do you compare dumps? Are you using that fest online database with automated tool or by hand? 

For the record, I never flash devices by air, I just do a firmware dumps of each device I buy in order to help people.

kuba2k2, 2022-10-23:
I don't know what "fest online database" is. I have written my own tool, that fetches all GitHub issues, downloads dump files from them, fixes/decrypts the dumps and analyzes them. Doesn't produce profiles automatically, because I was too lazy to do that.

openshwprojects, 2022-10-23:
@kuba2k2  wow, this is very nice, finally some good work. Why give up on that? If you have run into some issue with profile generation, tell me, I can help you with that. That tool would be a gold idea, especially that I make a lot firmware dumps to help users who are unable to solder or to open device without destroying it.

kuba2k2, 2022-10-23:
Answered similar questions here: https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/149#issuecomment-1288175382

`finally some good work` yes! *finally* we did something good!

closed 2022-11-30 by Cossid