tuya-cloudcutter/tuya-cloudcutter #160: does not exist or not a file

esnet146, 2022-10-24:
Firmware /work/custom-firmware//work/custom-firmware/51.bin does not exist or not a file.
Oh no, something went wrong with updating firmware! Try again I guess..

Yes! is not file /work/custom-firmware//work/custom-firmware/
File work/custom-firmware/51.bin exist

Step 10/10 : WORKDIR /src
 ---> Running in 73bb5439b595
Removing intermediate container 73bb5439b595
 ---> 893565bdd281
Successfully built 893565bdd281
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: 51.bin
 > 51.bin

[?] Select the brand of your device: Arlec
 > Arlec
   Avatar
   Ayococr
   Calex
   CASENL
   CLOUDY_BAY
   DIY
   Euarne
   Fcmila
   Feit
   FLSNT
   Gosund
   Helloify

[?] Select the article number of your device: ALD092CHA
 > ALD092CHA

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.

kuba2k2, 2022-10-24:
59963d7bfd16c3723d5be76f9f94d8a9af8bc89d introduced an error, try checking out the repository prior to that change.

esnet146, 2022-10-24:
checked when running from the command line. can be used. if you fix it - it will be good, but not critical

closed 2022-10-24 by esnet146