tuya-cloudcutter/tuya-cloudcutter #162: Zmai-90 with CBU module (the earlier version has WB3S)

openshwprojects, 2022-10-26:
Flash dump from a fellow user.
[image]

[ Zmai-90 with CBU module.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9867141/Zmai-90.with.CBU.module.zip)

Cossid, 2022-12-10:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-10 by Cossid

Liionboy, 2023-01-03:
i am trying to extract dump now but i get only this:

root@adrian-huawei-d16:/home/adrian# bk7231tools read_flash -d /dev/ttyUSB0 -s 0 -l 0x200000 dump_zmai90.bin
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/bk7231tools/__main__.py", line 468, in cli
    with closing(connect_device(args.device, args.baudrate, args.timeout, args.debug)) as device:
  File "/usr/local/lib/python3.10/dist-packages/bk7231tools/__main__.py", line 270, in connect_device
    s.connect()
  File "/usr/local/lib/python3.10/dist-packages/bk7231tools/serial/__init__.py", line 39, in connect
    raise TimeoutError("Timed out attempting to link with chip")
TimeoutError: Timed out attempting to link with chip

[image]

Cossid, 2023-01-03:
try adding `--timeout 9999`, executing the command, then briefly shorting the CEN pin to reboot the device.  You may need to use an external power source (not mains) if your ttl is being reset by that process.

Liionboy, 2023-01-03:
Same problem connected to 220v i dont know if is ok pin 15 and 16 for tx1 and rx1, on datasheet this write and CEN is pin 18

Cossid, 2023-01-03:
You should absolutely not be attempting to read flash when powered by mains.  If you did, you may have killed the module completely.

Liionboy, 2023-01-03:
is working :D but wtill can't read, i will give up now maybe someone will succesfully extract from CBU ver 2.3.3