tuya-cloudcutter/tuya-cloudcutter #163: Ledvance Smart+ 6W Is it exploitable?

JakubBrachTieto, 2022-10-26:
I'm running into trouble while using script on LEDVANCE Smart+
I ripped the device apart and I know it uses WBLC5 module with BK7231TQN32 and exploitability test script passes, however main script fails to complete. 

Logs from script and bulld attached.

I'm using Raspberry Pi 3 prepared according to the instruction.

`==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Job for NetworkManager.service failed because the control process exited with error code.
See "systemctl status NetworkManager.service" and "journalctl -xe" for details.
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Found access point name: "LDV SMART+-DB49", trying to connect..
Device 'wlan0' successfully activated with '52863403-7643-438a-a389-13c25ff1ddf7'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and 
Exploit run, saved device config to!
output=/work/configured-devices/UsA9nhSQl2oe.deviceconfig
Saved device config in /work/configured-devices/UsA9nhSQl2oe.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Job for NetworkManager.service failed because the control process exited with error code.
See "systemctl status NetworkManager.service" and "journalctl -xe" for details.
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Scanning for "A-" "GRID-" "Geeni-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "iHome-" SSID...
Found access point name: "LDV SMART+-DB49", trying to connect..
Device 'wlan0' successfully activated with '52863403-7643-438a-a389-13c25ff1ddf7'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
Using WLAN adapter: wlan0
Oct 26 20:22:32 dnsmasq[15]: started, version 2.80 cachesize 150
Oct 26 20:22:32 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Oct 26 20:22:32 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Oct 26 20:22:32 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Oct 26 20:22:32 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:33:65:a1 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Oct 26 20:22:37 dnsmasq-dhcp[15]: 1093620482 available DHCP range: 10.42.42.10 -- 10.42.42.40
Oct 26 20:22:37 dnsmasq-dhcp[15]: 1093620482 client provides name: wlan0
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 DHCPDISCOVER(wlan0) 50:8a:06:32:db:49 
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 tags: wlan0
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 DHCPOFFER(wlan0) 10.42.42.38 50:8a:06:32:db:49 
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 next server: 10.42.42.1
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  1 option: 53 message-type  2
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option: 54 server-identifier  10.42.42.1
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option: 51 lease-time  12h
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option: 58 T1  6h
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option: 59 T2  10h30m
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option:  1 netmask  255.255.255.0
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option: 28 broadcast  10.42.42.255
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option:  3 router  10.42.42.1
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 sent size:  4 option:  6 dns-server  10.42.42.1
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 available DHCP range: 10.42.42.10 -- 10.42.42.40
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 client provides name: wlan0
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 DHCPDISCOVER(wlan0) 50:8a:06:32:db:49 
Oct 26 20:22:40 dnsmasq-dhcp[15]: 1093620482 tags: wlan0
[... 723 lines cut ...]
rc_init: station_id=0 format_mod=0 pre_type=2 short_gi=529229 max_bw=1001086
rc_init: nss_max=0 mcs_max=0 r_idx_min=7 r_idx_max=529271 no_samples=1001153
sm_assoc_ind set stats to SM_IDLE
WPA: TK 058bd39c2bc8be72ecbe41682dfa1fc0
sta_mgmt_add_key
WPA: GTK 793bc2f6a7ec38a49d2e6970df34a473
ctrl_port_hdl:1
me_set_ps_disable:840 0 0 0 518373 999756

configuring interface mlan (with DHCP client)oper_state: state 0, operstate 1
me_send_ps_req 2 0 0
set_ps_mode_cfm:963 1 0 4 517961 999700

old ie: 0 : 63 6c 6f 75 64 63 75 74 74 65 72 66 6c 61 73 68 
new ie: 0 : 63 6c 6f 75 64 63 75 74 74 65 72 66 6c 61 73 68 
old ie: 1 : 82 84 8b 96 24 30 48 6c 
new ie: 1 : 82 84 8b 96 24 30 48 6c 
old ie: 3 : 1 
new ie: 3 : 1 
old ie: 30 : 1 0 0 f ac 4 1 0 0 f ac 4 1 0 0 f ac 2 c 0 
new ie: 30 : 1 0 0 f ac 4 1 0 0 f ac 4 1 0 0 f ac 2 c 0 
old ie: 2d : 21 0 17 ff 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
new ie: 2d : 21 0 17 ff 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

`

kuba2k2, 2022-10-26:
The profile you're trying to use is incompatible. Either the device has different hardware or got a software update which makes it incompatible.

Cossid, 2022-10-26:
> **********[oem_bk7231s_light_ty] [2.9.16] compiled at Apr 22 2021 19:25:54**********

The Phopollo-RGBW-SBE2606 should be compatible with this device (and probably a few others too).

ydac, 2022-10-28:
Many Thanks!

Profile you suggested worked like a charm.
There's one weird behaviour sill. 

Few minutes after startup, the device is very unresponsive on local interface (6668 TCP port).

Device still tries to connect somewhere:
[01-01 01:10:43 TUYA Err][httpc.c:532] tcp connect failed 10.42.42.1:80 errno=0
[01-01 01:10:43 TUYA Err][http_inf.c:47] Open session failed: http://10.42.42.1/d.json?a=tuya.device.info.sync&devId=EhbEd9RDb4rvvqpvgUZG&et=1&t=627&v=1.0&sign=8d7133c5d58eb664c4507aa1)
[01-01 01:10:43 TUYA Err][iot_httpc.c:1035] Post Data Return Fail.-7

Can it be configurable somehow? For example this  10.42.42.1: IP and port can be somethin else ? Or even disabled?

Cossid, 2022-10-28:
Unfortunately, I believe that is a pitfall of using the original firmware, in order to "cut" it from Tuya, it needs to be changed to something else.  "10.42.42.1" is the intermediate AP cloudcutter sets up, but unfortunately, it also needs that set to actually run the exploit, so I'm not sure it can easily be changed afterwards.  Nothing you could change it to should work, and you'll always see this "timeout" trying to resolve.  After it times out, it should be quickly responsive.

The best option is to replace the Tuya firmware with a 3rd party firmware such as LibreTuya/ESPHome or OpenBeken, in which case you'll have more full control.

closed 2022-11-21 by Cossid