tuya-cloudcutter/tuya-cloudcutter #164: Moes WiFi Smart Switch WA2 module BEKEN BK7231QN40 / Beken/HLK-B30 WA2?

openshwprojects, 2022-10-27:
A dump from a fellow Elektroda user who wants to flash OpenBeken.
[image]
[image]
[image]
[WA2 module BEKEN BK7231QN40  Beken HLK-B30 WA2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9875859/WA2.module.BEKEN.BK7231QN40.Beken.HLK-B30.WA2.zip)

_Disclaimer: I do not even flash by tuya-cloudcutter. I but one device of each kind to analyze it. I am posting here just to help community._

Cossid, 2022-11-20:
Posting for reference.  This dump contains only the decrypted app partition.  This app is very non-standard confirming in terms of how it was compiled, and there appears to be corruption within (obvious known strings have random corruption for example).  It appears to be BK7231N based on the SDK version, but I can't verify without the bootloader.  It also has a reference to an older Realtek SDK (similar to really old BK7231T devices).

This device is unlikely to get a profile, especially without a full encrypted bin dump.

kuba2k2, 2022-11-21:
There is no full encrypted binary. The code partitions are not encrypted.

This is a "BK7231QN40" chip, 40 pin QFN (notice no T or N). It's an older version of T, from what I can see. It's unlikely that any custom firmware (OBK/LT) will work with this device, even after removing the code encryption. The SDK for it was called tysdk_for_bk7231 or something; it's still available on GitHub in LibreTuya organization.

I like to call this chip "BK7231Q" as it's neither N nor T, and saying simply "BK7231" is too ambiguous. It would be great if we could further refer to this using the Q letter.

Also, the Tuya SDK (and blob library) differs a lot from what Cloudcutter can exploit, so it may even be impossible to create a profile for this right now.

Cossid, 2022-12-10:
Adding as won't fix, as I don't think it's feasible to invest reverse engineering a device that has so little public exposure.

closed 2022-12-10 by Cossid