tuya-cloudcutter/tuya-cloudcutter #168: Exploit possible on device running ewelink/smartlife software?

TurkeyMan, 2022-10-30:
Hi, I'm guessing this exploit is in the tuya firmware, and not from the hardware's wifi driver?
I have a couple of devices where I bought identical copies, one flashed with tuya software, and the other flashed with ewelink/smartlife software, so that i could compare the software.
Is it possible to exploit the ewelink/smartlife devices in the same way?

ferbulous, 2022-10-30:
Most probably not with ewelink since they're not using tuya chips (bk7231n/bk7231t).

TurkeyMan, 2022-10-30:
Oh are they a Tuya exclusive chip? Weird; this Chinese device gave me the option ewelink or tuya, the device looks identical i just choose which OS to run... when I cracked open the Tuya one, it has a bk7231t, and I just assumed the ewelink one would be identical inside.

kuba2k2, 2022-10-30:
The chips may (and likely are) the same in ewelink devices, but the firmware is probably different. Cloudcutter only works for Tuya firmwares, so if a device is not compatible with SmartLife/TuyaSmart apps, it will never work with Cloudcutter.

TurkeyMan, 2022-10-31:
Okay, so the software, not the wifi driver for the chips.

closed 2022-11-21 by Cossid