tuya-cloudcutter/tuya-cloudcutter #169: LSC 3001700 - Device Profile needed

ReanuKeeves01, 2022-10-30:
Hi all,

I can see this specific bulb is in the `device_pictures` folder ( [link](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-pictures/3001700.jpeg) )

However I cannot see a corresponding device profile to go with this product number. I could try a few other LSC profiles. I have a feeling it shouldn't be too different from the [3001686](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/device-pictures/3001686.jpeg), which as a device profile and is practically a smaller version of this exact product. 

Anyone can point me in the right direction as to which device profile works with this bulb before I end up just trying all different LSC profiles ^^

Many thanks in advance.

ReanuKeeves01, 2022-10-30:
@pkoretic didn't you mention you already tried to flash this device? as mentioned the product image is available and thus looks supported, yet not device profile. Looking at your original comment, i'm a bit wary to try anyway since i got them working in TuyaLocal and cut off from the internet, so there's no real rush.

Cossid, 2022-12-01:
This has been fixed, and a profile should now exist.

closed 2022-12-01 by Cossid

ReanuKeeves01, 2022-12-01:
@Cossid Great, thanks for the follow up! 
I will try the profile immediately tonight and confirm the same!