tuya-cloudcutter/tuya-cloudcutter #170: Hama integretion if posible

botioni, 2022-10-30:
Hama integretion if posible ?
tryd tuya generic not working .
Using PSK v1 - Received PSK ID version 02
[W 221030 17:02:21 iostream:1406] SSL Error on 12 ('10.42.42.24', 53720): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221030 17:02:22 iostream:1406] SSL Error on 12 ('10.42.42.24', 53721): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221030 17:02:23 iostream:1406] SSL Error on 12 ('10.42.42.24', 53722): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
[W 221030 17:02:24 iostream:1406] SSL Error on 12 ('10.42.42.24', 53723): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)

ReanuKeeves01, 2022-10-30:
With such little info, it's kinda difficult for people to be of any help mate.

botioni, 2022-10-31:
What you need? Because I don't know what is needed.

tony-fav, 2022-10-31:
To start:

What device do you have?

What profile did you try?

botioni, 2022-10-31:
I have a hama gu10 rgbcw bulb and I tryd tuya generic

tony-fav, 2022-10-31:
When you use the Tuya app with it, what version firmware does it have?

botioni, 2022-10-31:
I have it linked to ldv app not in tuya and it has v 1.1.2

Cossid, 2022-10-31:
Which "tuya generic" profile did you use?  Tuya-Generic E27-A60-RGBCWW-V1 is likely the correct profile, though that schema has extra RGB stuff if you plan to use stock firmware just cut.  LSC 3000272-970715 would have the correct schema.

botioni, 2022-10-31:
It E27-RGBCW-SMARTLIFE WB2L_M1 I used

botioni, 2022-10-31:
I wil try lsc you mentioned thx

Cossid, 2022-12-10:
Closing as stale.  Please provide additional information if you re-open this issue.

closed 2022-12-10 by Cossid