tuya-cloudcutter/tuya-cloudcutter #171: german Aldi Süd casalux Wifi LED RGB Light Strip Dump (Tuya CBU, BK7231N,RGB PWM and IR receiver)

twobeseen, 2022-11-03:
Here is the (german) no name brand casalux with the “Wifi LED RGB Light Strip”, vendor is german Aldi Süd.
I bought it in 2022-10 for ca. 17 €. Seems to be it was sold also in early 2022 for 10€, but I don’t know if it had the same controller back then, seems to be identical.

Early 2022 info: https://www.aldi-sued.de/de/p.casalux-wifi-led-rgb-light-strip.490000000000711151.html
Youtube Video/Review about early 2022 (packing is still looking the same): https://www.youtube.com/watch?v=W_QRa3aQdN4
2022-10 info: https://www.aldi-sued.de/de/p.casalux-wifi-led-rgb-light-strip.490000000000818507.html

Packing states: Imported by Horstmann GmbH & Co KG, Hopfens. 1d, 24114 Kiel, Germany www.horstmann-kiel.com
Markings on the packing: 818507, Modell 9A22044,   X/17/2022
Technical specs on the packing:
 - 9 W
 - 0,5 W standby
 - 5 m LED strip
 - 150 LEDs

The LED strip only has 4 connections (12 V, R-PWM,G-PWM,B-PWM), it only has RGB, so no extra white LEDs

The “Smart Wi-Fi controller" in the white plastic box has the following information on the sticker:
Model: DW-RGB-WI01
Input: 12 V DC
Output: 3x2A (Max)
2.4 GHz ISM band
protection class III
Dongguan D-win Light ing Co., Ltd
Room 101, No.17,Dongkeng West Yanhe
Road 3. Dongkeng Town, Dongguan City. Guangdong Province China

Besides this controller there is the LED strip, a simple IR remote and a wall wart power plug (12 V DC / 1 A) in the box.

I opened the controller and found a Tuya CBU module (or CBU IPEX?): https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u which wasn’t a surprise because the handbook tells you to download one of the Tuya apps.

The CBU module (which has Beken BK7231N chip) has only 7 connections soldered to the PCB:
3.3 V: CBU Pin 14 
GND: CBU Pin 13

PWM Green: CBU Pin 8
PWM Blue: CBU Pin 9
PWM Red: CBU Pin 10

IR Receiver: CBU Pin 1

?: CBU Pin 21 (maybe for mechanical stability reasons because the right edge of the module has no solder connection, seems to be not connected to anything else on the PCB)

I set up my old Android smartphone, changed MAC address for safety reasons and set up a test access point with SSID “test1234” and used the “Smart Life”  android app to connect the device to the tuya cloud, but i didn’t create a user profile (skipped profile creation in the app). I could then use the LED strip with the smart life app, so i suppose the readout dump should contain a Datapoint ID (DPID).
I found the access point SSID and PSK in cleartext in the dump.bin (see below) at offset 0x33CAA7 to 0x33CB58. Didn’t find my access point MAC (BSSID) in the dump.bin.

For readout I connected my Raspi 3B to UART (CBU Pins 15 and 16) (redirected the Raspi UART /dev/ttyAMA0 to the Raspi Pins and therefore deactivated Bluetooth on the Raspi to get a “real” UART, checked if /dev/serial0 now points to /dev/ttyAMA0) and used python script:

bk7231tools read_flash -d /dev/serial0 -s 0 -l 0x200000 ./dump.bin

Further attempts (dump2.bin and all further attempts) gave a slightly changed readout (changed 1 time, stayed then this way), but looking into the dump file reveal that the change doesn’t affect the boot loader or the main program but maybe is some sort of boot counter. (change between dump.bin and dump2.bin starts at offset 0x36A1E3, ends at offset 0x36A1FF).

At 0x1EF000 there seems to be no DPID data. (it is all blank there) but at 0x1EEA8C to 0x1EEB07 there is something, maybe it is it? If yes: This could be for BK7231N devices other than BK7231T.

I am also curious why free areas in the flash have the data 0x87CB in it instead of 0x00 or 0xFF.

Because I couldn’t write my CBU module with bk7231tools (see my issue https://github.com/tuya-cloudcutter/bk7231tools/issues/7) i successfully wrote my flash with https://github.com/OpenBekenIOT/hid_download_py with the OpenBeken FW. Still trying that out, but seems to work so far, boots up, WLAN works, everything seems fine, haven’t tried to configure PWM/LEDs or IR receiver yet.

[Aldi_casalux_Wifi_RGB_LED.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9927782/Aldi_casalux_Wifi_RGB_LED.zip)

Have fun!

kuba2k2, 2022-11-03:
> Further attempts (dump2.bin and all further attempts) gave a slightly changed readout (changed 1 time, stayed then this way), but looking into the dump file reveal that the change doesn’t affect the boot loader or the main program but maybe is some sort of boot counter

That's normal, the firmware writes to upper flash areas frequently, so it's usual to get different data after the device boots.

I'm not sure why are you referring to the offsets like 0x33CB58; the flash is 2 MiB in size, which is 0x200000 so there's no data at that offset. It's true that bk7231tools prints larger offsets - that's to trick the bootloader to allow reading the entire flash. If you do `0x33CB58 % 0x200000` it will show the correct offset, which would be `0x13CB58` in this case.

---

> I found the access point SSID and PSK in cleartext in the dump.bin (see below) at offset 0x33CAA7 to 0x33CB58. Didn’t find my access point MAC (BSSID) in the dump.bin.

It's at offset 0x1D1000 - not sure where do the offsets you mention come from. The MAC is also there - at offset 0x1D1021 - `C0:EE:FB:C2:D9:C5`.

---

> At 0x1EF000 there seems to be no DPID data. (it is all blank there) but at 0x1EEA8C to 0x1EEB07 there is something, maybe it is it? If yes: This could be for BK7231N devices other than BK7231T.

The data starts at 0x1EE000 and goes all the way to 0x200000. It's an encrypted block that contains DPIDs, network SSID, pinouts and other things like that.

---

> I am also curious why free areas in the flash have the data 0x87CB in it instead of 0x00 or 0xFF.

I don't see that in the dump. All empty flash data is filled with 0xFF, some blocks (which are used) are filled with 0x00 if the data takes less space.

---

I'll look into the issue with bk7231tools. Meanwhile, I think you're using some kind of a weird hex viewer - the offsets at which you find the data are wrong. On Linux, you may simply use:

```
cat dump.bin | hexdump -C | less
```

kuba2k2, 2022-11-03:
For the record, here's your encrypted block data, but decrypted and put into JSON. I believe there are the DPIDs you're looking for.

[gh-171-9927782-aldi-casalux-wifi-rgb-led-dump.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9928037/gh-171-9927782-aldi-casalux-wifi-rgb-led-dump.json.txt)

twobeseen, 2022-11-03:
I used Textwrangler on mac os to view the .bin files and used File -> Hex Dump Front Document, so all the offsets come from there. Don't know why all the offsets would be completely off.

Will look into it later, also why i obviously wrote some offset out of 2MB area which doesn't make sense...
Thanks for your reply !

twobeseen, 2022-11-04:
I edited my original post to correct my wrong offsets and wrong assumptions based on the bad hexdump i saw. Thanks kuba2k2 :-)

closed 2022-11-30 by Cossid