tuya-cloudcutter/tuya-cloudcutter #172: KHSUIN 16W RGBW

W-Floyd, 2022-11-04:
Found here: https://www.amazon.com/dp/B08FML629L

[Dump (using LightLeak)](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9934626/16W.zip)

Could never actually get these to pair with the app...

Cossid, 2022-11-04:
Can't give you a schema since it's unpair, but you probably don't need it anyway.

Profile should be (does not match any other prior device)
```
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQYX4CSJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QkJCQg0rCQANKwkADSsJAA0rCQANKwkADSsJAA0rCQANKwkADSsJAA=="
}
```

W-Floyd, 2022-11-08:
@Cossid profile didn't work unfortunately.

W-Floyd, 2022-11-08:
Interesting. I'll try again later and post the log (though it's uninteresting). When I tried, it would just hang - no errors, but no progress be (AP name changes, stops blinking). Using other profiles would produce errors as expected.

Cossid, 2022-11-08:
Try clearing out your configured-devices folder (backup anything you might need).

If the AP name changes, that means the profile is working at least.

W-Floyd, 2022-11-09:
Same result:

```
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Nov  9 00:57:49 dnsmasq[15]: started, version 2.80 cachesize 150
Nov  9 00:57:49 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Nov  9 00:57:49 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Nov  9 00:57:49 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Nov  9 00:57:49 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 38:de:ad:71:c3:5c and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Nov  9 00:57:52 dnsmasq-dhcp[15]: 1716252376 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov  9 00:57:52 dnsmasq-dhcp[15]: 1716252376 client provides name: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPDISCOVER(wlan0) 38:1f:8d:cb:82:c8 
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 tags: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPOFFER(wlan0) 10.42.42.11 38:1f:8d:cb:82:c8 
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 next server: 10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  1 option: 53 message-type  2
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 54 server-identifier  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 51 lease-time  12h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 58 T1  6h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 59 T2  10h30m
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  1 netmask  255.255.255.0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 28 broadcast  10.42.42.255
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  3 router  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  6 dns-server  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 client provides name: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPDISCOVER(wlan0) 38:1f:8d:cb:82:c8 
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 tags: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPOFFER(wlan0) 10.42.42.11 38:1f:8d:cb:82:c8 
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 next server: 10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  1 option: 53 message-type  2
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 54 server-identifier  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 51 lease-time  12h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 58 T1  6h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 59 T2  10h30m
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  1 netmask  255.255.255.0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 28 broadcast  10.42.42.255
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  3 router  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  6 dns-server  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 available DHCP range: 10.42.42.10 -- 10.42.42.40
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 client provides name: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPREQUEST(wlan0) 10.42.42.11 38:1f:8d:cb:82:c8 
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 tags: wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 DHCPACK(wlan0) 10.42.42.11 38:1f:8d:cb:82:c8 wlan0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 next server: 10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  1 option: 53 message-type  5
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 54 server-identifier  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 51 lease-time  12h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 58 T1  6h
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 59 T2  10h30m
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  1 netmask  255.255.255.0
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option: 28 broadcast  10.42.42.255
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  3 router  10.42.42.1
Nov  9 00:57:55 dnsmasq-dhcp[15]: 1716252376 sent size:  4 option:  6 dns-server  10.42.42.1
Nov  9 00:57:56 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.11
Nov  9 00:57:56 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[I 221109 00:57:57 web:2239] 200 POST /v1/url_config (10.42.42.11) 96.40ms
[... cut ...]
{'token': 'AAAAAA', 'softVer': '1.0.0', 'productKey': '0qjbwe2ybnlf1out', 'protocolVer': '2.2', 'baselineVer': '40.00', 'hid': '381f8dcb82c8', 'modules': '[{"type":9,"softVer":"1.0.0","online":true}]', 'devAttribute': 515, 'cadVer': '1.0.2', 'cdVer': '1.0.0', 'options': '{"isFK":false}', 't': 27}
[LOG] ==== End client request body ===
tuya.device.active - response: {'success': True, 't': 1667955477}
[I 221109 00:57:57 web:2239] 200 POST /d.json?a=tuya.device.active&et=1&t=27&uuid=HyU6lxvMSEcy&v=4.4&sign=f2ace86f98b28b0f2e7c5061f32ec63c (10.42.42.11) 3.50ms
```

Cossid, 2022-11-09:
Did you copy all the other json files out of another device folder?  Make sure you have all of them (there are a couple existing profiles missing some json response files).

closed 2022-12-01 by Cossid

59zCez, 2023-12-18:
HI, I got 4 Bulbs that are KHSUIN 16W, they are the BR30, they look a little different that the ones from the OP, I have tried to load the firmware with no success, see attached log from couldcutter - it seems like it should work and it does not... I have opened one of the bulbs to verify the chipset, indeed is BK7231T.
product:
https://www.amazon.com/Smart-Light-Bulbs-Equivalent-Changing/dp/B08D3LCBMN/ref=sr_1_1_sspa?crid=19WVQZ5UPUEX4&keywords=KHSUIN%2BBR30%2B16w&qid=1701368848&s=hi&sprefix=khsuin%2Bbr30%2B16w%2Ctools%2C81&sr=1-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

I have loaded one using local tuya and got some useful information that may help (specifications,debuging)... not sure if there is anything missing from the profile already listed/loaded under couldcutter:

Tried to load OpenBeken-v1.17.262_bk7231t.ug.bin and ESPHome-Kickstart-v23.08.29_bk7231t_app.ota.ug.bin and no success.
I dont own a android equipment so I cant try a dump., is there anything that can be done?
I was able to make it work using TuyaLocal but I really preferred either ESPhome or Tasmota.

Tuya developer shows this as the Product:
Device Name	Product
Smart Bulb 1	bk_lm96xe1_16w

{
  "result": {
    "active_time": 1702902546,
    "category": "dj",
    "create_time": 1702869514,
    "custom_name": "Smart Bulb 1",
    "icon": "smart/icon/001458092465966zkPdc/710695a223fb43108cfb908c49b6587a.png",
    "id": "xxxxxxxxxxxxxx",
    "ip": "96.255.49.102",
    "is_online": true,
    "lat": "",
    "local_key": "xxxxxx",
    "lon": "",
    "model": "",
    "name": "Smart Bulb",
    "product_id": "fslmzu1jritagk4z",
    "product_name": "bk_lm96xe1_16w",
    "sub": false,
    "time_zone": "-05:00",
    "update_time": 1702902546,
    "uuid": "xxxxxxxxx"
  },
  "success": true,
  "t": 1702913369714,
  "tid": "xxxxxxxxxxxxxxxxx"

[KHSUIN 16W.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708551/KHSUIN.16W.txt)
[Debuging from tuya developer.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708557/Debuging.from.tuya.developer.txt)
[Standard Status set.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708558/Standard.Status.set.txt)
[specifications.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/13708559/specifications.txt)