tuya-cloudcutter/tuya-cloudcutter #173: Issue when device tries to connect to cloudcutterflash AP

deanfourie1, 2022-11-06:
Im getting the following when the device tries to connect to the AP. Looks like it attempts to connect but fails with

```
[W 221106 04:22:38 iostream:1406] SSL Error on 12 ('10.42.42.19', 26997): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02
```

Any ideas?
Thanks

Cossid, 2022-11-06:
That means the exploit didn't work and the profile did not match your device.  Full logs would show your device did not have an AP prefix starting with A- at that point, which is necessary for the exploit.

Please post what device this is, what module it has, what Tuya firmware is currently on it (if joined to SmartLife), or a firmware dump if you have have pulled one already.

deanfourie1, 2022-11-06:
> That means the exploit didn't work and the profile did not match your device.  Full logs would show your device did not have an AP prefix starting with A- at that point, which is necessary for the [...]

To be honest I don't know much about it, but the brand is Brilliant. An Australian company.

Also, regarding the profile. I did not specify a profile when running run_detach.sh as there is nothing in the documentation regarding this?

Do I need it?

deanfourie1, 2022-11-06:
> That means the exploit didn't work and the profile did not match your device.  Full logs would show your device did not have an AP prefix starting with A- at that point, which is necessary for the [...]

I believe it is a Generic Tuya device based on the GU10 RGBW bulbs. ESP8285 I believe.

Cossid, 2022-11-06:
It forces you to choose one after you run run_detach.sh if you didn't provide one in advance.  It made you select a mfg + model, which in combination is a profile.

If it is an ESP8285, Tuya-CloudCutter will not work on it.  CloudCutter is currently for Beken BK7231T and BK7231N based modules.

deanfourie1, 2022-11-06:
> It forces you to choose one after you run run_detach.sh if you didn't provide one in advance.  It made you select a mfg + model, which in combination is a profile.
> 
> If it is an ESP8285, Tuya-CloudCutter will not work on it.  CloudCutter is currently for Beken BK7231T and BK7231N based modules.

Bugger, stuck with lemons

closed 2022-11-21 by Cossid