tuya-cloudcutter/tuya-cloudcutter #174: LV300/400//480/600ST LED ceiling light

Elkropac, 2022-11-06:
Hi, 
is it possible to cut this one?
https://www.aliexpress.com/item/1005003439859979.html
It's CB3L chip

Do i have to select some model from menu in run_detach.sh?

[image]

Cossid, 2022-11-06:
Can you provide either a firmware dump or if the device is paired to Smart Life, a firmware version (Smart Life -> Device -> edit button in upper right -> Device Update -> List versions on that page).

Elkropac, 2022-11-06:
I haven't paired it with smartlife yet, i went directly to liberating the device ;) 

I have ewlink installed on my phone, is it ok to use this to get the info you want?

Cossid, 2022-11-06:
I am not certain what options are in ewlink.  If it has the ability to display current firmware on the device, that would probably be sufficient.

Elkropac, 2022-11-06:
Ok, i will try it. Will it be still possible to use cutter, even after pairing with official cloud?

Cossid, 2022-11-06:
Yes.

Elkropac, 2022-11-06:
Ok, i will have to use Smart life.
 It's really painful to start the pairing, i saw it's AP two times, but i failed after my phone connected back to wrong network

Elkropac, 2022-11-06:
Ok, it finished pairing somehow, even it said it failed, thats why it did not show AP again.
It wants me to update firmware, is it safe to do, won't it prevent me from using cutter?

Cossid, 2022-11-06:
No, do not update firmware if asked.  It could jeopardize your chances of cloudcutter working.

Elkropac, 2022-11-06:
In device update it says "Version 1.2.2", nothing more

Cossid, 2022-11-06:
There is no current profile that matches that firmware, a firmware dump is needed, and even then it will take some time, as the existing tools do not yet work well with this device.

Elkropac, 2022-11-06:
Ok, i will try to connect to chip, no promises

Elkropac, 2022-11-12:
Not sure, what to do.
I connected my usb flash device using this diagram
https://developer.tuya.com/en/docs/iot/cb3l-module-datasheet?id=Kai51ngmrh3qm

Then connected RST to ground . I cannot read from device using uartprogram  from   hid_download_py

kuba2k2, 2022-11-12:
Here's a guide you can use:
https://docs.libretuya.ml/docs/platform/beken-72xx/dumping/

I'm not sure why are you connecting RST to ground. Use bk7231tools as instructed in this guide, and it will work.

Elkropac, 2022-11-12:
The guide also says to connect CEN to GND to reset the board. CEN is RST, pin 1 , right?

Elkropac, 2022-11-12:
Or is it the pin 3?
There are two ports named "CEN"
https://docs.libretuya.ml/boards/cb3l/#pinout

kuba2k2, 2022-11-12:
> The guide also says to connect CEN to GND to reset the board. CEN is RST, pin 1 , right?

But you're connecting it to GND for a short time only. Run the program, then reset the chip, then disconnect CEN from GND. It should start reading if you connected it properly.

It shouldn't matter which CEN you're using. Tuya didn't really make it clear which is the actual reset pin, maybe they're just both wired together.

Elkropac, 2022-11-12:
When i reset chip, i loose /dev/ttyUSB4 decice and program quits.

I'm trying now to have CEN connected to ground while i plug USB flasher to usb port, then disconnect CEN and start the program - but i still cannot read.

Should i get some communication on minicom, while it boots?

Is it ok to have chip connected to the light driver without soldering it out?

kuba2k2, 2022-11-12:
> When i reset chip, i loose /dev/ttyUSB4 device and program quits.

Then you have to power the chip externally, from another 3.3V power supply. Otherwise it draws too much current and resets the USB-TTL adapter.

Elkropac, 2022-11-12:
I think, i have another usb-ttl module, i will use it for RX-TX (+ground) and this module for power(+ground) - could it work?

kuba2k2, 2022-11-13:
Yes, it should.

Elkropac, 2022-11-13:
Oh yeah, it's reading something. I'm always so nervous when connecting multiple powered devices together via shared ground

Elkropac, 2022-11-13:
Here is dump with my wifi name and password edited out (at least plaintext representation), hope it does not break it for you
[lvl300_light.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9997914/lvl300_light.zip)

Elkropac, 2022-11-17:
@kuba2k2 I have flashed it with your forked esphome and it works great, thanks

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

Elkropac, 2022-12-01:
Hi, thanks for your work. I'm sorry, having it disassembled and with wires soldered to it, I have flashed it with libretuya/esphome already. I'm considering to buy more units, so I might try it later

divadiow, 2024-11-29:
> Here is dump with my wifi name and password edited out (at least plaintext representation), hope it does not break it for you [lvl300_light.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9997914/lvl300_light.zip)

this firmware offers an update to v1.2.2 from v1.0.0

[image]

[1.2.2_boot.txt](https://github.com/user-attachments/files/17962991/1.2.2_boot.txt)
[1.2.2_lvl300_light.zip](https://github.com/user-attachments/files/17962993/1.2.2_lvl300_light.zip)

Cossid, 2024-11-29:
Updated with split versioned profiles.