tuya-cloudcutter/tuya-cloudcutter #175: Hama E27 CW Wifi 00176584

jf-wm, 2022-11-06:
I connected the bulb ([manufacturer link](https://uk.hama.com/00176584/hama-wlan-led-lamp-e27-10w-dimmable-bulb-for-voice-app-control-white)) with tuya using a dummy AP. I hope you can extract the profile out of the dumped flash. I have several other of these lamps, so I can check if the profile works when a profile is available. 
[dump.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9946705/dump.bin.zip)

Cossid, 2022-11-06:
LSC 3000272-970715 profile should match your device.

Cossid, 2022-12-01:
A new profile has been verified and added for this device.

closed 2022-12-01 by Cossid

jf-wm, 2022-12-10:
I finally tested the LSC and the new profile you created. The LSC profile doesn't work but the new profile works just fine on the first try.

frostworx, 2023-02-22:
Hi. I received my 4x 00176584 for 5€ order today and found both this project and this issue quickly.
Of course, I immediately tested it and I can confirm the first device I tried was perfectly "freeable".
Thank you both for your great work! ❤

Liionboy, 2024-01-09:
Today i will teardown and extract profile, now i buy bulb

Cossid, 2024-01-09:
> Had ordered the same bulbs but it seems that the profile doesn't work anymore. The Tuya-app writes version 3.3.31 but I can't find a profile with this version...

3.3.31 is either ESP or patched beken.