tuya-cloudcutter/tuya-cloudcutter #176: LSC Outdoor Spot lights

RF197, 2022-11-08:
[image]

Data dump attached.

Please tell me if you need more info, I had to "dissect" these lights a little....
They are basically three RGB spotlights, with one "master" (containing the CB2L module) and two "slaves" which are connected with a cable.

[Led_spots_extract_dir.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9965100/Led_spots_extract_dir.zip)

Cossid, 2022-11-08:
The iHome IH-BW948-999 profile should work for this device.

Cossid, 2022-12-01:
The dump attached to this issue is extracted only and does not include the storage, so no schema can be generated/verified.  You can try the iHome IH-BW948-999, or supply a full 2M bin and we can re-process this.

RF197, 2022-12-04:
Dear Cossid, thanks for the hard work by the way.

I extracted what I could from the raw 2M dump I made, but that process might have gone wrong. Here's the raw dump, maybe that'll clear things up?

Otherwise I can try again, but these lights are _lets sat_ not openable without destroying them....
[lsc_outdoor_spots_raw.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10148766/lsc_outdoor_spots_raw.zip)

Cossid, 2022-12-04:
Yes, that is exactly what we needed.  A profile for this device has been added.

closed 2022-12-04 by Cossid