tuya-cloudcutter/tuya-cloudcutter #178: Wifi Error while flashing Nous A1 Smart Socket

denizkoekden, 2022-11-09:
Hello,
I tried to flash my Nous A1 Socket (BK7231N-Chipset) with Raspberry Pi 3. I used the specific instructions for Raspberry Pi 3 but I always get this error: "Error: Connection activation failed: 802.1X supplicant took too long to authenticate" I also tried the "-r" flag without success.

The Scripts detects my Plug in AP-Mode then tries to connect to it but fails with the above message. Any Idea what this could be? I try reflashing and start over with blank system again but it's weird. Maybe someone has an Idea.

Also which Template should I use for the Nous A1?

Regards

Cossid, 2022-11-10:
Nous A1 has not been seen here yet.  You would at minimum either need to post the Tuya firmware version installed (can be found if joined to Smart Life or via a serial boot log), or a firmware dump of the device.  LightLeak does not work on BK7231N devices yet.

denizkoekden, 2022-11-10:
So the Module has Mainmodule Version V1.1.12 and Other Type Version 1.0.0. I guess I have to flash the Module via Serial. Also there is already an Thread about the Device here: https://www.elektroda.com/rtvforum/topic3901733.html

Cossid, 2022-11-10:
There is not currently a compatible profile for that device.

Cossid, 2022-12-10:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-10 by Cossid

cosote, 2022-12-30:
I also have some NOUS A1 plugs with latest mainmodule version v1.1.12.
Please find the dump with SSID wlan and password 12345678 attached for profile generation.
[NOUS_A1_1.1.12.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10323533/NOUS_A1_1.1.12.bin.zip)

Cossid, 2022-12-30:
@cosote Thanks, I have added this as Nous A1 Smart Plug, please let me know if it works or not.  I can't tell if this is using a patched SDK or not.  The compile date is after their known patch, so it's not outside the realm of possibility that this device is patched.

Cossid, 2022-12-30:
@cosote I have removed the profile.  Unfortunately, it appears this firmware has been patched and is not vulnerable to CloudCutter.  Serial flashing will be required for this device.

kuba2k2, 2022-12-30:
I'd suggest adding a new label for patched devices.

Cossid, 2022-12-30:
Ha, I was just doing that.

cosote, 2022-12-31:
@Cossid Ok, cool, thx - I've flashed it via serial with latest version of OpenBK7231T_App and all is running fine.
But I also have 11 other Nous A1 plugs on original firmware (probably 1.0.0 or so), that haven't been updated yet to 1.1.12.
I will pull that firmware and upload it here in a couple Minutes 👍.

cosote, 2022-12-31:
@Cossid Please find attached here the dump from stock/not updated Nous A1 plug. It looks to me it's version 1.1.8.
I was dissecting it, but could find a doc how to prepare a profile/schema or if I had it, how to use it with tuya-cloudcutter manually.
[NOUS_A1_1.1.8.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10327099/NOUS_A1_1.1.8.bin.zip)

Cossid, 2022-12-31:
@cosote Thanks!  I have added a profile for 1.1.8, which is indeed still vulnerable.

cosote, 2022-12-31:
It worked perfectly - thx a lot & have a happy new year :-D