tuya-cloudcutter/tuya-cloudcutter #180: [Profile request]  ENSHINE GU10 Smart Bulb

hectork2, 2022-11-15:
Hi 
i teardowned my LED Bulb and I dump firmware with bk7231tools and try make profile 
but i understand There is no good guide to make
i try used this help to make profile with this topic, but i can not 
https://www.elektroda.com/rtvforum/viewtopic.php?p=20107873#20107873

Is it possible to create a profile for ENSHINE GU10? 

Product :
Full Name. : ENSHINE GU10 Smart Bulb, Tunable White and RGB Colour WiFi Light Bulbs, 5W, 470 Lumens
https://www.amazon.co.uk/Enshine-Dimmable-Changing-Compatible-Required/dp/B08DXJ5CSW

[ENSHINE GU10 Smart Bulb-dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10016289/ENSHINE.GU10.Smart.Bulb-dump.zip)

hectork2, 2022-11-16:
Thank you , But when i run shell  
i give error for "null byte detected in payload"

 ---> Using cache
 ---> b1cb250f57cc
Step 8/10 : FROM python-deps AS cloudcutter
 ---> b1cb250f57cc
Step 9/10 : COPY src /src
 ---> Using cache
 ---> d04ef6606f29
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> e02fe5f7aff9
Successfully built e02fe5f7aff9
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231T_UG_1.14.39.bin
 > OpenBK7231T_UG_1.14.39.bin

[?] Select the brand of your device: Enshine
   Avatar
   Ayococr
   Calex
   CASENL
   CLOUDY_BAY
   DIY
 > Enshine
   Euarne
   Fcmila
   Feit
   FLSNT
   Gosund
   Helloify

[?] Select the article number of your device: GU10-Bulb
 > GU10-Bulb

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
ome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-15AE", trying to connect..
Device 'wlan0' successfully activated with 'da3544bb-7c48-454c-b9b7-01d3afe3fafe'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 344, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 213, in __exploit_device
    device_config = exploit_device_with_config(args, exploit_profile)
  File "/src/cloudcutter/exploit.py", line 89, in exploit_device_with_config
    payload = check_valid_payload(payload)
  File "/src/cloudcutter/exploit.py", line 57, in check_valid_payload
    raise ValueError('At least one null byte detected in payload!')
ValueError: At least one null byte detected in payload!

Cossid, 2022-11-16:
Interesting, thanks for following up.  That is indeed a problem, and I'm not sure there is a solution.  I've requested feedback in the PR thread from the cloudcutter team.  I'll follow up if there is progress.

hectork2, 2022-11-16:
Can I ask you a question ?
How did you create the profile?
Because according to the tutorial on the site and used pyload.py
It is different from the profile you created

https://www.elektroda.com/rtvforum/viewtopic.php?p=20107873#20107873

[payload_builder.py.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10023001/payload_builder.py.zip)

pipenv run python haxomatic.py ../file/dump_extract_dir/dump_app_1.00_decrypted.bin
 
[!] Loading and disassembling code - may take a moment
[+] Code loaded!
[!] Searching for post-vuln code patterns
[+] Found a post-vuln code pattern match!
[+] Matched instructions: 
        0xac14c: adds r4, #0xfc
        0xac14e: ldr r3, [r4, #0x50]
        0xac150: ldr r1, [pc, #0xb0]
        0xac152: adds r0, r6, #0
[+] Identified lan object register as r4
[!] Searching for JSON object register
[+] Identified JSON object register as r7
[!] Searching for ty_cJSON_Parse function address
[+] ty_cJSON_Parse address: 0xba9e0
[!] Searching for mf_cmd_process gadget address
[+] mf_cmd_process gadget address (THUMB): **0xa2e01**
[!] Searching for a mov r0, r7 intermediate gadget
[+] Found usable intermediate gadget at address **0xb0078**:
        0xb0078: ldr r3, [r4]
        0xb007a: adds r0, r7, #0
        0xb007c: blx r3
[+] Payload gadgets (THUMB): intermediate_gadget_addr=0xb0079 mf_cmd_gadget_addr=0xa2e01

[image]

 resultat =>

 python3 ../payload_builder.py  
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUnqCSJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QkJCQkEgCQBBIAkAQSAJAEEgCQBBIAkAQSAJAEEgCQBBIAkAQSAJAA=="
}

Cossid, 2022-11-16:
You used the address instead of the thumb for the first value.  The last line of successful output from haxomatic are the addresses you should be using in payload_builder.

hectork2, 2022-11-16:
i comment code  cheking payload in exploit.py and run again 
now  i have new error for ssl 

Step 9/10 : COPY src /src
 ---> 66822fe1227c
Step 10/10 : WORKDIR /src
 ---> Running in e4d8c46bafbf
Removing intermediate container e4d8c46bafbf
 ---> 84b00a9f53ef
Successfully built 84b00a9f53ef
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231T_UG_1.14.39.bin
 > OpenBK7231T_UG_1.14.39.bin

[?] Select the brand of your device: Enshine
   Avatar
   Ayococr
   Calex
   CASENL
   CLOUDY_BAY
   DIY
 > Enshine
   Euarne
   Fcmila
   Feit
   FLSNT
   Gosund
   Helloify

[?] Select the article number of your device: GU10-Bulb
 > GU10-Bulb

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-15AE", trying to connect..
Device 'wlan0' successfully activated with 'da3544bb-7c48-454c-b9b7-01d3afe3fafe'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Using  and
Exploit run, saved device config to!
output=/work/configured-devices/pr8tiGMyaVmn.deviceconfig
Saved device config in /work/configured-devices/pr8tiGMyaVmn.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-15AE", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-15AE", trying to connect..
Device 'wlan0' successfully activated with 'da3544bb-7c48-454c-b9b7-01d3afe3fafe'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Nov 16 15:22:28 dnsmasq[15]: started, version 2.80 cachesize 150
Nov 16 15:22:28 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Nov 16 15:22:28 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Nov 16 15:22:28 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Nov 16 15:22:28 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:7b:86:af and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
[... 18 lines cut ...]
Nov 16 15:22:38 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Nov 16 15:22:38 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221116 15:22:38 iostream:1406] SSL Error on 12 ('10.42.42.39', 59474): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Nov 16 15:22:40 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Nov 16 15:22:40 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221116 15:22:40 iostream:1406] SSL Error on 12 ('10.42.42.39', 59475): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Nov 16 15:22:41 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Nov 16 15:22:41 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221116 15:22:42 iostream:1406] SSL Error on 12 ('10.42.42.39', 59476): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Nov 16 15:22:43 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Nov 16 15:22:43 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221116 15:22:43 iostream:1406] SSL Error on 12 ('10.42.42.39', 59477): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Nov 16 15:22:44 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
[... cut ...]
[W 221116 15:22:45 iostream:1406] SSL Error on 12 ('10.42.42.39', 59478): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Nov 16 15:22:46 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.39
Nov 16 15:22:46 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221116 15:22:46 iostream:1406] SSL Error on 12 ('10.42.42.39', 59479): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Using PSK v1 - Received PSK ID version 02

Cossid, 2022-11-16:
That won't work, that check is there for a reason.  This is the first case of "bad luck" where the device is vunerable, but at a place in the code we can't exploit due to the delivery method in use.  Essentially, it would require reworking the exploit entirely as I understand it, and it's not likely to happen anytime soon for a single device.

hectork2, 2022-11-16:
> That won't work, that check is there for a reason. This is the first case of "bad luck" where the device is vunerable, but at a place in the code we can't exploit due to the delivery method in use. [...]

Unfortunately, I have about 20 count of this device

Cossid, 2022-11-16:
> can you explain how to create another file to make profile , such as tuya.device.active and ...

There are only 2 unique files per device profile.

The first and most important is the "profile" file, which needs to be generated generally per device, but is sometimes shared.  This is currently done via haxomatic (when it works, currently only for specific BK7231T devices) and payload_builder.py (which is incomplete, only can output for BK7231T devices currently).  I am currently working on an improved and combined version of these that will also be more universal, but it is not complete yet.  It also will not resolve the issue with this specific device anyway.

The second file is "tuya.device.active.json", and can often be shared between similar devices.  There are two ways to generate this file, and I'm only familiar with one of them (the same instructions as you linked to in the elecktroda thread) where you join the device to Smart Life, then dump it, and extract data from the decrypted settings (which is sometimes fragmented).  The alternate way, which I have no experience with is intercepting auth keys from a tuya server and then communicating directly with it via the API using those keys.  This process is only needed if you intent to use Tuya's firmware in cut mode.  If you intend to replace it with custom firmware, an existing file from any other profile will work fine.

For the rest of the files residing in the same directory, they are generic files and can be copied from any other profile, just make sure all the needed ones are present (it doesn't hurt to include all of them that are currently available).

Cossid, 2022-11-20:
After discussion and assist from @notkmhn we found the effected address was one where we had a choice of two and the alternate appears to work.  I tested locally it it looks good.  I'll try to get the tools updated to reflect this method.

Can you please pull the update from that branch and try again?

hectork2, 2022-11-21:
> After discussion and assist from @notkmhn we found the effected address was one where we had a choice of two and the alternate appears to work. I tested locally it it looks good. I'll try to get the [...]

Thank you , yea , i answer today

Cossid, 2022-11-22:
Sorry, I was migrating to the organization repos, and had to relocate this.
Now located at https://github.com/tuya-cloudcutter/tuya-cloudcutter/tree/gh-180

closed 2022-11-30 by Cossid