tuya-cloudcutter/tuya-cloudcutter #183: Access point scan pattern is not robust

harsh2029, 2022-11-18:
I was trying out exploit for one of my CB2S based Smart plugs as commented [here](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/120).
During the process, the smart plug creates `A-B172` access point.
However, the cloudcutter script detects another AP which is someone's WiFi SSID `MOTO3C7A-5G`.
`MOTO3C7A-5G` has pattern `"A-"` but it does not start with `"A-"`.
cloudcutter most of the time detects `MOTO3C7A-5G` first before `A-B172` and fails as shown below.
```
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "MOTO3C7A-5G", trying to connect..
Error: Connection activation failed: (7) Secrets were required, but not provided.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "MOTO3C7A-5G", trying to connect..
Error: Connection activation failed: (7) Secrets were required, but not provided.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "MOTO3C7A-5G", trying to connect..
Error: Connection activation failed: (7) Secrets were required, but not provided.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "MOTO3C7A-5G", trying to connect..
Error: Connection activation failed: (7) Secrets were required, but not provided.
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "MOTO3C7A-5G", trying to connect..
Error: Connection activation failed: (7) Secrets were required, but not provided.
Error, could not connect to SSID.
Failed to connect, please run this script again

```

I believe, the pattern matching can be made robust enough to avoid such issues.

Cossid, 2022-11-18:
Can you please try modifying [common.sh line 3](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/922709203f782bc8586654d2c199a4ee2631eea6/common.sh#L3)

from
`COMBINED_AP_PREAMBLE=$(cat ap_preambles.txt | grep -v '#' | sort -u | awk '{print "-e \""$0"\"" }' | tr '\n' ' ')`
to
`COMBINED_AP_PREAMBLE=$(cat ap_preambles.txt | grep -v '#' | sort -u | awk '{print "-e \"^"$0"\"" }' | tr '\n' ' ')`

harsh2029, 2022-11-18:
> Can you please try modifying [common.sh line 3](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/922709203f782bc8586654d2c199a4ee2631eea6/common.sh#L3)
>
> from `COMBINED_AP_PREAMBLE=$(cat [...]

Thanks!
In addition to your change, I also have to remove fixed string matching option `-F` from line 45 to make it work.
`AP_MATCHED_NAME=$(nmcli -t -f SSID dev wifi list --rescan yes | eval grep $COMBINED_AP_PREAMBLE | sort -u)`

closed 2022-11-22 by Cossid