tuya-cloudcutter/tuya-cloudcutter #184: TreatLife SL10 Firmware v1.0.2 - SSL Error, unable to flash

juniortivotyro, 2022-11-19:
Hello, I purchased a new 4 pack of SL10 bulbs from Amazon, and upon trying to run the tuya-cloudcutter exploit on them, I get the following error:

```
Using PSK v1 - Received PSK ID version 02
SSL Error on 12 ('10.42.42.11', 59776): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
```
Upon looking through different issues, it seems that the profile is incompatible with whatever firmware version is on these bulbs. I believe this is the same issue as referenced in #88 which seems to have people being able to flash them successfully. The fact that it isn't allowing me to makes me think this may be a newer firmware than was referenced in that issue. I paired the bulb to Smart Life, and it seems the Main/MCU versions are both 1.0.2.

I don't currently have the ability to dump the firmware on one, but I do have a 3D printer and I could make the jig needed to do so if required, though I am a bit hesitant to do so since I haven't been able to successfully connect to similar chips through serial by soldering without bricking it.

What I've tried:

- SL10-V2 profile (with/without specified Device ID/Localkey)
- SL10-OMNI-TP120 profile (with/without specified Device ID/Localkey)
- SL20 profile (with/without specified Device ID/Localkey)
- Tuya-Generic lightbulb profiles  (with/without specified Device ID/Localkey)

Here is a log file of my last unsuccessful run: [log.log](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10047874/log.log)

Cossid, 2022-11-19:
Sounds like they have a real variety in the SL10 line.  There are currently 2 SL10 dumps and they are for firmware versions 1.0.1 and 1.0.8, so yours would be a 3rd now.  There is not currently a profile for 1.0.2 bulbs that I'm aware of.

If you have a spare ESP/ESP32 dev board, could you try the combination of the [android app](https://github.com/tuya-cloudcutter/cloudcutter-android) and [lightleak](https://github.com/tuya-cloudcutter/lightleak) (where you install lightleak on a dev board and install the android app on a phone, grant it location permissions manually) and try dumping the firmware wirelessly?  Assuming this is still BK7231T like the other devices.  If it is BK7231N, it will not work with lightleak yet.

juniortivotyro, 2022-11-19:
Thank you for the quick reply :smile: I didn't know about that method of dumping. I'll look into it and try to dump a fresh bulb.

juniortivotyro, 2022-11-19:
It took some effort, but I was able to get the ESP32 firmware to compile and flash with it seemingly working

What I changed to make it compile for esp32 (in customap.cpp):
```
//#if defined(ESP32)
//	WiFi.onEvent(onStationConnected, SYSTEM_EVENT_STA_CONNECTED);
//#elif defined(ESP8266)
//	WiFi.onEvent(onStationConnected, WIFI_EVENT_SOFTAPMODE_STACONNECTED);
//#else
	WiFi.onEvent(onStationConnected, ARDUINO_EVENT_WIFI_AP_STACONNECTED);
//#endif
```

Serial output shows that it's starting the AP, the AP shows on my android device, and the CloudCutter application can connect to it. However, the app eventually gives an error that the device isn't responding to pings. Is there something I may be doing wrong? Could my changes to the code have caused this error? I wasn't able to get it to compile with any semblance of functionality with any other changes/lack of changes

Cossid, 2022-11-19:
cloudcutter-android and lightleak are still in early phases, so there are still a lot of rough edges.  Unfortunately, I can't speak to that code change with much authority, I can only state for me with esp32, SYSTEM_EVENT_AP_STACONNECTED defines to integer 16.  If that is the same on arduino, and it probably is, it should be fine.

Additionally, the device running lightleak and the target device should be very close in proximity, as AP mode doesn't have a very strong signal.  I think the next release (unknown timeframe) will have longer timeout, as that appears to be a common occurrence so far.

Thank you for taking the time to try it at least.

Cossid, 2022-11-19:
Also, I should clarify, a few requirements for lightleak.  1) you must remove the device from smart life and reset to factory defaults when asked.  2) You should be using the "Lightleak -BK7231T (Lightleak)" profile, not any of the versioned ones.

juniortivotyro, 2022-11-19:
It seems that the issue I'm having with lightleak compiling is related to using v2.0.0 of the ESP32 WiFi library, so that at least seemingly answers why "ARDUINO_EVENT_WIFI_AP_STACONNECTED" works

Still at a bit of a loss on what to do with getting the device firmware

juniortivotyro, 2022-11-19:
Okay, so turns out the issue I was having with connecting to the Lightleak ap was due to a VPN to my local network, which has the subnet 10.0.0.1/24

It just got to the step "Connect to smart device wifi", going to continue the steps

Cossid, 2022-11-19:
Thanks for following up with that.  I'll put in a request to use the 10.42.42.0/24 range like regular cloudcutter does.

juniortivotyro, 2022-11-20:
The furthest I've been able to get is a failure at checking if device is exploitable. Is it possible that means it could be using the BK7231N chip?

Cossid, 2022-11-20:
Honestly, I can't really answer, that's further than I make it with lightleak so far (but it has worked for others).

I know you mentioned not soldering at this point, but could you open one device and confirm which wifi module this device has?  It usually starts with a "W" or a "C", W is BK7231T, C is BK7231N

juniortivotyro, 2022-11-20:
Here is an image of the LED boardFrom: Cossid ***@***.***>Sent: Saturday, November 19, 2022, 6:04 PMTo: tuya-cloudcutter/tuya-cloudcutter ***@***.***>Cc: juniortivotyro ***@***.***>; Author ***@***.***>Subject: Re: [tuya-cloudcutter/tuya-cloudcutter] TreatLife SL10 Firmware v1.0.2 - SSL Error, unable to flash (Issue #184)
Honestly, I can't really answer, that's further than I make it with lightleak so far (but it has worked for others).
I know you mentioned not soldering at this point, but could you open one device and confirm which wifi module this device has?  It usually starts with a "W" or a "C", W is BK7231T, C is BK7231N

juniortivotyro, 2022-11-20:
Okay, that didn't seem to work. Either way, It's a WB8P

Cossid, 2022-11-20:
No problem, that's enough information to verify this is still a BK7231T device.

juniortivotyro, 2022-11-20:
Even if I were 100% sure I wanted to solder or use the jig, I don't see a way to remove the boards without destructive methods

kuba2k2, 2022-11-20:
> The furthest I've been able to get is a failure at checking if device is exploitable. Is it possible that means it could be using the BK7231N chip?

That means it's probably not exploitable, or at least not with the parameters currently available in Lightleak. I currently don't have a solution for that.

juniortivotyro, 2022-11-20:
Alright, I bit the bullet and dumped the fw, and it seems like it was a success! I will attach it now, though I don't know how to verify if it's correctly dumped
[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10050500/dump.zip)

Cossid, 2022-11-20:
Unfortunately, that is incomplete.  Can you try again using [bk7231tools](https://github.com/tuya-cloudcutter/bk7231tools) and the following parameters:
-s 0 -c 512
for sizing along with whatever you need for com port and timeout values.

juniortivotyro, 2022-11-20:
[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10050595/dump.zip)
How's this?

Cossid, 2022-11-20:
Looks good.

Try copying one of the existing SL10 profiles, and replacing the contents of 'profile' with
```
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQRMqCiJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QUFBQYdaCQCHWgkAh1oJAIdaCQCHWgkAh1oJAIdaCQCHWgkAh1oJAA=="
}
```

I'm not going to create a PR as we're currently transitioning to a universal format, and it will be included there when we're able to roll it out.

juniortivotyro, 2022-11-20:
That worked perfectly, thank you!

closed 2022-11-30 by Cossid