tuya-cloudcutter/tuya-cloudcutter #185: BlitzWolf BW-CLT1 Ceiling Light dump

blakadder, 2022-11-21:
Blitzwolf BW-CLT1 ceiling light (available at but now branded as BlitzWill: https://www.banggood.com/custlink/vvKWlyt6aE)
WB3L software version 1.1.9
[BW-CLT1_dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10060522/BW-CLT1_dump.zip)

deepcoder, 2022-11-22:
Looks very similar to this unit I converted using physical connection to MCU for flashing. As is shown in picture, MCU is a WB3S rather than WB3L in the Blitzwolf. Two units working well, very pleased with them running OpenBK7231T firmware. I did have to reflash one of the two units as it lost its config and hung, however both have been solid since. While it is easy to open the units I have, hope you find a way to install firmware without opening the unit. Stupid me, I set one of the open units down on the floor and accidentally lightly kicked it, knocked or broke something on circuit board. Now I have a spare with bad circuit board ;-) Thanks for your work!

```

Build on Sep 13 2022 07:56:30 version 1.12.55
Build on Oct 29 2022 17:49:13 version 1.14.86

P6 PWM 5 warm white
P8 PWM 4 is cool white
P26 PWM 3 is blue
P24 PWM 2 is green
P9 PWM 1 is red
```

https://smile.amazon.com/gp/product/B081QGS8CK

[image]

blakadder, 2022-11-22:
This is CLT1 internals
[image]

closed 2022-11-30 by Cossid