tuya-cloudcutter/tuya-cloudcutter #191: Arlec ALD095HA - RGBCW Downlight (AU) - WB2S BK7231T

machship-mm, 2022-11-25:
Arlec ALD095HA downlight (available from [Bunnings Australia](https://www.bunnings.com.au/arlec-9w-720lm-rgb-cct-grid-connect-smart-led-downlight_p0335074))

Firmware version according to Tuya app is V2.9.16.

I have successfully flashed OpenBK7231T_App using uart, however, no matter what I do, this device will not update to new firmware using Raspberry Pi for tuya-cloudcutter. 

I've tried the PHOPOLLO/SBE2606 profile and it all says it succeeds, but the firmware is not replaced. Tried changing timeout to close to 90 seconds from 30, but it still just never updates.

I also tried haxomatic and was given this profile, but again, it just hangs and never updates:
```
[!] Matched pattern for BK7231T version 2, payload version 1
[!] Searching for ty_cJSON_Parse
[+] Payload prep gadget (THUMB): 0xba65d
[!] Searching for mf_cmd_process
[+] Payload pwn gadget (THUMB): 0xad515
[+] Profile:
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQV2mCyJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QUFBQRXVCgAV1QoAFdUKABXVCgAV1QoAFdUKABXVCgAV1QoAFdUKAA=="
}
```

[Arlec_ALD095HA.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10091851/Arlec_ALD095HA.zip)

I have about 40 of these so I'd really love to not have to pull them all apart! Thanks all

Cossid, 2022-11-25:
The profile is correct.  Make sure you don't have MQTT or DNSMASQ running (they are running by default on raspberry pi os), as they are required for firmware flashing.

machship-mm, 2022-11-26:
Thanks. Turns out it was the mosquitto broker running as a service. Once disabled, the PHOPOLLO/SBE2606 worked perfectly! 

Thanks guys.

closed 2022-11-26 by machship-mm