tuya-cloudcutter/tuya-cloudcutter #194: Seeking profile for Tuya NAS-WR01W plug with power monitoring

cawidtu, 2022-11-26:
I recently bought an "unbranded" Tuya PowerPlug NAS-WR01W power plug that seems to come with a BK7231T chip. I was unable to convert it using the current tuya-cloudcutter profiles, but could dump its flash memory using the amazing Lightleak concept. I am attaching the dump and some picture of the device. Can anyone help and create a profile for the device?

[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10096237/dump.zip)
[image]
[image]
[image]

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

cawidtu, 2022-12-02:
Thank you. It works. I flashed 3 devices with OpenBK7321T and accidentally once used the profile for the NAS-WR07 (tuya-generic-nas-wr07w-16a-smart-plug) - which still worked. Are the profiles for the NAS-WR01 and NAS-WR07 exchangeable?

kuba2k2, 2022-12-02:
They use the same firmware, but it's possible that their schema differs. For flashing, schema doesn't matter, but if you wanted local control (cloudcutting with stock firmware), it might have not worked well enough.