tuya-cloudcutter/tuya-cloudcutter #195: Information on a few Treatlife switches

rickyelopez, 2022-11-27:
Hey guys, bought a bunch of switches and used this tool to cut the cloud connection, will be flashing libretuya tomorrow. Figured it would be good to add the profiles that worked for them. Had some trouble with this tool the last time I used it, but it was smooth sailing this time. That trick about using `tuya-convert`'s startup script to kill services holding ports was golden, worked a treat this time too.

- `DS02` Smart 3-way Dimmer Switch successfully expoited with `DS02F` profile
- `DS01C` Smart Dimmer Switch successfully exploited with `DS02F` profile
- `SS01S` Smart Switch successfully exploited with `1.1.8 - BK7231T / oem_bk7231s_rnd_switch` profile
- `SS01` Smart 3-way Switch - No luck here, tried a bunch of profiles. Any suggestions for other profiles to try? If not, what data do I need to provide to generate a profile? It failed with `SSL Error on 12`

kuba2k2, 2022-11-27:
About the SS01, you can join it in the Tuya app and see what firmware version it's running. Alternatively, you can try Lightleak PoC, since WB3S is BK7231T.

rickyelopez, 2022-11-27:
Thanks for the tip @kuba2k2! Unfortunately, no luck :cry:. I paired it to the app and confirmed v1.1.3 on both MCUs. I tried that profile but still hit the `SSL Error on 12`. Any other suggestions?

Cossid, 2022-11-27:
The only 1.1.3 we have is for a curtain switch, and judging by the class name it is not compatible.  Would it be possible for you to get a dump of the firmware?  As kuba2k2 mentioned, you might be able to dump it wirelessly via lightleak, in which case we would be able to generate a profile from that.

rickyelopez, 2022-11-27:
Hmm I'm having trouble finding lightleak, googling is not turning anything up. Can you provide a link? Happy to get a dump.

Cossid, 2022-11-27:
It's still in the early phases and not documented.  It requires a dev board (ESP8266, ESP32, or Beken) as an intermediate device, and an android device to run the app.
https://github.com/tuya-cloudcutter/cloudcutter-android for the app (see releases to download the apk).  You will need to manually grant location access.
https://github.com/tuya-cloudcutter/lightleak for the intermediate device.

rickyelopez, 2022-11-27:
Wow that's a neat project! I'm trying to flash it onto a WB3S because that's the only chip I have here. It flashes successfully but the AP doesn't come up, even after a reset. Any suggestions?

kuba2k2, 2022-11-27:
Not unless we see logs :smile:

rickyelopez, 2022-11-27:
haha fair, not sure what logs to show you though? I don't see anything on the uart, not sure where else to look?

Cossid, 2022-11-27:
I did ESP32, so I can't say for sure.  How did you flash the WB3S version?  Which file and offset did you use?

kuba2k2, 2022-11-27:
> haha fair, not sure what logs to show you though? I don't see anything on the uart, not sure where else to look?

The Lightleak platformio program is set to output logs on UART1, the same one as used for flashing. So check there if you checked on UART2 before.

rickyelopez, 2022-11-27:
> I did ESP32, so I can't say for sure. How did you flash the WB3S version? Which file and offset did you use?

I flashed using platformio, just did `pio run -e bk7231t` and then `pio run -t upload -e bk7231t`.

> The Lightleak platformio program is set to output logs on UART1, the same one as used for flashing. So check there if you checked on UART2 before.

I checked using `pio device monitor -e bk7231t` which should be using uart1, I don't see anything except one null character on boot.

kuba2k2, 2022-11-28:
You can check if it prints any useful info on UART2. Should at least be the bootloader message with some weird numbers.

Cossid, 2022-12-10:
Closing as there is no actionable item here.  It should remain searchable as information.

closed 2022-12-10 by Cossid