tuya-cloudcutter/tuya-cloudcutter #196: Nedis Flood Light gives error message when trying to use cloudcutter

atxbyea, 2022-11-27:
Trying to use cloudcutter on my PopOS laptop, getting an error message during the cutting phase, unsure if this is because the device does not exist in a profile (tried a few different profiles) or due to it being popOS (ubuntu based with networkmanager).

`{"uuid": "uuid", "auth_key": "authkey", "local_key": "localkey", "sec_key": "seckey", "device_id": "devid", "chip_family": "BK7231N"}`

`Using PSK v1 - Received PSK ID version 02
[W 221127 20:01:34 iostream:1406] SSL Error on 12 ('10.42.42.29', 54842): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)`

If needed I can provide dumps or other information.

kuba2k2, 2022-11-27:
This error means an incompatible profile. Pair the device with the Tuya app and check the software version.

atxbyea, 2022-11-27:
`Main Module V1.0.2`
`MCU Module V1.0.2`

`PIR Floodlight WIFILOFS20FBK`

kuba2k2, 2022-11-27:
The only 1.0.2 profiles we currently have is a switch and some vendor-built bulb firmware. So most likely there's no profile for your light yet.

atxbyea, 2022-11-27:
Great, thanks for the feedback, I'll check back later :-)

Cossid, 2022-11-27:
Do you know which module is in this floodlight, and does it have a TuyaMCU?

atxbyea, 2022-11-28:
I have not opened it up no, I only know the CHIP based on the identification above.

kuba2k2, 2022-11-28:
If you haven't opened it up, you don't know the chip for sure. The fact that you got `chip_family` as BK7231N only means that you selected a BK7231N profile.

jkwim, 2022-12-19:
Can somebody share a log of a successful attempt with Received PSK ID version 02 as shown in below log?

`Dec 15 16:53:00 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.17
Dec 15 16:53:00 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221215 16:53:01 iostream:1404] SSL Error on 12 ('10.42.42.17', 64539): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)`

Is it correct to assume that any device that reponds with PSK ID version 02 will not work?

Cossid, 2022-12-19:
Correct, if your AP prefix does not change to "A-" and you receive SSL errors, that means the profile you selected was not compatible with your device.

jkwim, 2022-12-19:
> Correct, if your AP prefix does not change to "A-" and you receive SSL errors, that means the profile you selected was not compatible with your device.

My question was actually, has any device which responds with 'PSK ID version 02' worked with correct profile.

Is there something that is needed to handle 'PSK ID version 02' on the host side? 

I am asking the question having found this from "tuya-convert" team:

[https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02](url)

Cossid, 2022-12-19:
If you get any SSL error at all, that means the profile you selected is not compatible.  It doesn't necessarily mean that the device can't be cloud cut, it just means the profile you selected cannot be used for that device.

Cossid, 2022-12-19:
Closing issue as this isn't a bug, just an incompatible profile.  The app will not give a more clear message stating when a profile is not compatible.

closed 2022-12-19 by Cossid

szupi-ipuzs, 2023-08-09:
Sorry to dig up an old thread, but I have just bought the same device and was able to succesfully cloudcut it, using ume-motion-security-light profile. However, my device has V1.1.7 firmware versions (both main and module), which seems to be the latest available.
So, @atxbyea, you can simply allow Tuya app to upgrade the firmware and then cloudcutting it might become possible....

Coincidently, the exact same profile also worked on Brennenstuhl WFD 3050 P, which seems to have exact same firmware, down to the same DPIds and values.