tuya-cloudcutter/tuya-cloudcutter #202: Provided device profile path /work/device-profiles/bneta-iot-smart-wifi-led-e27-rgbct-bulb/combined.json is not a directory

bonuzzz, 2022-12-02:
One more error there:
```
Using interface wlan0 with hwaddr b8:27:eb:37:a0:4e and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Dec  2 13:02:47 dnsmasq-dhcp[15]: 842012285 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec  2 13:02:47 dnsmasq-dhcp[15]: 842012285 client provides name: wlan0
Provided device profile path /work/device-profiles/bneta-iot-smart-wifi-led-e27-rgbct-bulb/combined.json is not a directory
Oh no, something went wrong with detaching from the cloud! Try again I guess..
```

bonuzzz, 2022-12-02:
the same with other profile
Provided device profile path /work/device-profiles/immax-neo-e14-rgbcct-bulb/combined.json is not a directory

kuba2k2, 2022-12-02:
Oops, I must have missed that. Should be fixed in 7211831bb44f773bb767601f5aa7a97cdff89f03.

bonuzzz, 2022-12-02:
It works! Thanks!

closed 2022-12-02 by bonuzzz