tuya-cloudcutter/tuya-cloudcutter #204: run_flash.sh issues? Provided device profile path xxxx is not a directory

remlei, 2022-12-02:
`Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.88MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 3aacf23a10f6
Step 3/10 : FROM base AS python-deps
 ---> 3aacf23a10f6
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> b79d3a87f0b7
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 4e1f1a6ee6e9
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> dcdadaa83711
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> e4f1996c0cfa
Step 8/10 : FROM python-deps AS cloudcutter
 ---> e4f1996c0cfa
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 379d1a43af07
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> a7f3ec67f95a
Successfully built a7f3ec67f95a
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231T_UG_1.15.125.bin
 > OpenBK7231T_UG_1.15.125.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Tuya Generic
   QIACHIP
   Qiachip
   STITCH
   Spectrum
   Stitch
   Topgreener
   TreatLife
 > Tuya Generic
   UME
   UltraPro
   Woox
   iHome
   ihome

[?] Select the article number of your device: LSPA9 Smart Plug
   DS-101 2-gang Light Switch
   DS-102 2-gang Light Switch
   E27 A60 RGBCWW V1
   E27 RGBCT WB2L M1 Bulb
   EDM-01AA-EU Dimmer Switch
   LSPA7 Plug
 > LSPA9 Smart Plug
   LSPS5 Smart Wifi Switch 3-Gang
   LV300 LV400 LV480 LV600ST LED Square LED Ceiling Light
   MSL7 Fairy Light Strip v1.0.0
   MSL7 Fairy Light Strip v1.1.0
   Mini Smart Switch
   NAS-WR01W Smart Plug

[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_dltj
 > 1.0.3 - BK7231T / oem_bk7231s_dltj

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
.Found access point name: "SmartLife-2667", trying to connect..
Error: Connection activation failed: (11) 802.1X supplicant took too long to authenticate.
..Found access point name: "SmartLife-2667", trying to connect..
Device 'wlxe84e0632002d' successfully activated with 'f3ec3ffd-6622-4b72-91e6-7f055fc5efcd'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/LwtxsDwFTsyE.deviceconfig
Saved device config in /work/configured-devices/LwtxsDwFTsyE.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
Found access point name: "SmartLife-2667", trying to connect..
Error: Connection activation failed: (11) 802.1X supplicant took too long to authenticate.
.....Found access point name: "SmartLife-2667", trying to connect..
Error: Connection activation failed: (11) 802.1X supplicant took too long to authenticate.
.Found access point name: "SmartLife-2667", trying to connect..
Device 'wlxe84e0632002d' successfully activated with 'f3ec3ffd-6622-4b72-91e6-7f055fc5efcd'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlxe84e0632002d
Dec  2 14:53:46 dnsmasq[15]: started, version 2.80 cachesize 150
Dec  2 14:53:46 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Dec  2 14:53:46 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Dec  2 14:53:46 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlxe84e0632002d
Dec  2 14:53:46 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlxe84e0632002d with hwaddr e8:4e:06:32:00:2d and ssid "cloudcutterflash"
wlxe84e0632002d: interface state UNINITIALIZED->ENABLED
wlxe84e0632002d: AP-ENABLED 
Provided device profile path /work/device-profiles/tuya-generic-lspa9-smart-plug/combined.json is not a directory
Oh no, something went wrong with updating firmware! Try again I guess..
`

I dont know whats the issue here

Cossid, 2022-12-02:
Please do a `git update`, I believe this was fixed earlier today.

zillion42, 2022-12-02:
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'NetworkManager.service'.
Authenticating as: tobi,,, (tobi)
Password:
==== AUTHENTICATION COMPLETE ===
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-"
Found access point name: "SmartLife-3EE0", trying to connect..
Error: Insufficient privileges.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Could not load profile tuya-generic-nas-wr01w-smart-plug. Are you sure the profile file exists and is a valid combined JSON?
Oh no, something went wrong with running the exploit! Try again I guess..
tobi@introvision2:~/tuya-cloudcutter$ git pull
Already up to date

Cossid, 2022-12-02:
> Error: Insufficient privileges.

You probably need to run elivated

zillion42, 2022-12-02:
No, weird. I tried it by ssh and by terminal in ubuntu, I have sudo twice, once in the beginning and once for network manager.

kuba2k2, 2022-12-02:
Oops, that's another mistake I made. Are you specifying profile on commandline? Try using the interactive menu.
`tuya-generic-nas-wr01w-smart-plug.json` that file shouldn't exist there, what is it?

zillion42, 2022-12-02:
I downloaded it from https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/tree/master/devices
because I tried to solve the problem, ok I will try interactive

Cossid, 2022-12-02:
Yeah, you can't download just the device json without the profile json referenced within.  Should work fine from the menu.

zillion42, 2022-12-02:
```
Successfully built 2daddc23190b
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

Traceback (most recent call last):
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/src/.venv/lib/python3.9/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/local/lib/python3.9/socket.py", line 954, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connection.py", line 358, in connect
    self.sock = conn = self._new_conn()
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f961d2cf220>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/src/.venv/lib/python3.9/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/src/.venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/src/.venv/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='tuya-cloudcutter.github.io', port=443): Max retries exceeded with url: /api/devices.json (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f961d2cf220>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/src/get_input.py", line 253, in <module>
    cli(obj={})
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/src/get_input.py", line 219, in choose_profile
    device_slug = ask_device_base(api_get("devices.json"))["slug"]
  File "/src/get_input.py", line 13, in api_get
    with requests.get(f"https://tuya-cloudcutter.github.io/api/{path}") as r:
  File "/src/.venv/lib/python3.9/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/src/.venv/lib/python3.9/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/requests/adapters.py", line 565, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='tuya-cloudcutter.github.io', port=443): Max retries exceeded with url: /api/devices.json (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f961d2cf220>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
cat: profile.txt: No such file or directory
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'NetworkManager.service'.
Authenticating as: tobi,,, (tobi)
Password:
==== AUTHENTICATION COMPLETE ===
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-"
```

zillion42, 2022-12-02:
I also tried:
[?] How do you want to choose the device?: By manufacturer/device name
   By manufacturer/device name
   By firmware version and name
  > From device-profiles (i.e. custom profile)

kuba2k2, 2022-12-02:
Do you have internet connection on the box you're using?

zillion42, 2022-12-02:
Would be super Nice if this would work...
Do I have internet connection ? Sure, except I stopped dns port 53 so, it didn't even built the docker image in the beginning.
I had to add 8.8.8.8 to resolve.conf

zillion42, 2022-12-02:
Do you wish to terminate systemd-resolve? [y/N] y

kuba2k2, 2022-12-02:
For now you need internet connection to at least download the profile first. We'll think about a possibility of pre-downloading all profiles locally.

zillion42, 2022-12-02:
I have internet:

```
tobi@introvision2:~/tuya-cloudcutter$ nslookup google.com
Server:         127.0.0.53
Address:        127.0.0.53#53

Non-authoritative answer:
Name:   google.com
Address: 142.250.203.110
Name:   google.com
Address: 2a00:1450:400a:808::200e

tobi@introvision2:~/tuya-cloudcutter$ wget https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/tuya-generic-nas-wr01w-smart-plug.json
--2022-12-02 19:13:42--  https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/tuya-generic-nas-wr01w-smart-plug.json
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘tuya-generic-nas-wr01w-smart-plug.json’

tuya-generic-nas-wr01w-smart-plug.js     [ <=>                                                                  ] 209,41K  --.-KB/s    in 0,05s

2022-12-02 19:13:43 (3,78 MB/s) - ‘tuya-generic-nas-wr01w-smart-plug.json’ saved [214435]

tobi@introvision2:~/tuya-cloudcutter$
```

zillion42, 2022-12-02:
I don't know if the docker image has internet

kuba2k2, 2022-12-02:
For some reason your Docker doesn't have internet. Alternatively you can download both device JSON and its profile JSON, put them in a subdirectory in device-profiles, and choose `From device-profiles (i.e. custom profile)`.

zillion42, 2022-12-02:
there is no profile for NAS-WR01W in the profiles directory
https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/tree/master/profiles

zillion42, 2022-12-02:
it's a nous a1
https://github.com/arendst/Tasmota/discussions/17209

zillion42, 2022-12-02:
ahh, its: "oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00"

kuba2k2, 2022-12-02:
Here's [NAS-WR01W](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/devices/tuya-generic-nas-wr01w-smart-plug.json). Its profile is [oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00](https://github.com/tuya-cloudcutter/tuya-cloudcutter.github.io/blob/master/profiles/oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00.json).

kuba2k2, 2022-12-02:
There's no guarantee your Nous A1 is compatible with that NAS-WR01W.

zillion42, 2022-12-02:
```
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: From device-profiles (i.e. custom profile)
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile
```

I copied both to device profiles, it won't show:

```
tobi@introvision2:~/tuya-cloudcutter$ ls device-profiles/
oem-bk7231s-dltj-1.3.5-sdk-1.0.2-40.00.json  README.md  schema  tuya-generic-nas-wr01w-smart-plug.json
```

zillion42, 2022-12-02:
sorry, I'm trying this for the first time

kuba2k2, 2022-12-02:
You have to put them in a subdirectory.

kuba2k2, 2022-12-02:
See here: https://github.com/tuya-cloudcutter/tuya-cloudcutter/tree/main/device-profiles

zillion42, 2022-12-02:
```
device-profiles/
├─ your-device-name/
│  ├─ your-device-name.json
│  ├─ oem-bk7231n-dctrl-switch-1.1.0-sdk-2.3.1-40.00.json
```

[image]

kuba2k2, 2022-12-02:
The files are way too big. Are you sure you didn't save the HTML of GitHub page? You have to press "Raw" to download the JSON file.

zillion42, 2022-12-02:
```
obi@introvision2:~/tuya-cloudcutter$ ./run_flash.sh
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.72MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> f4a8e1262ec3
Step 3/10 : FROM base AS python-deps
 ---> f4a8e1262ec3
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 0366218cbb83
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 3c55c27a6d5e
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 877222ed0957
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> e72bd74d3d13
Step 8/10 : FROM python-deps AS cloudcutter
 ---> e72bd74d3d13
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 8f8a7be398b2
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 2daddc23190b
Successfully built 2daddc23190b
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: From device-profiles (i.e. custom profile)
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-nas-wr01w
 > tuya-generic-nas-wr01w

Traceback (most recent call last):
  File "/src/get_input.py", line 253, in <module>
    cli(obj={})
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/src/get_input.py", line 232, in choose_profile
    device, profile = load_profile(profile_dir)
  File "/src/get_input.py", line 102, in load_profile
    with open(join(profile_dir, "device.json"), "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/work/device-profiles/tuya-generic-nas-wr01w/device.json'
cat: profile.txt: No such file or directory
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
.........Found access point name: "SmartLife-3EE0", trying to connect..
Device 'wlx00aaab002c30' successfully activated with '854cb6fd-a295-461d-923d-1c6a9d0362e0'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Could not load profile . Are you sure the profile file exists and is a valid combined JSON?
Oh no, something went wrong with running the exploit! Try again I guess..
tobi@introvision2:~/tuya-cloudcutter$ 
```

zillion42, 2022-12-02:
**The files are way too big. Are you sure you didn't save the HTML of GitHub page? You have to press "Raw" to download the JSON file.**

ooops

kuba2k2, 2022-12-02:
> Could not load profile . Are you sure the profile file exists and is a valid combined JSON?
Oh no, something went wrong with running the exploit! Try again I guess..

Fixed in latest commit, do git pull and try again.

zillion42, 2022-12-02:
ok I will git pull, but I think there is something wrong with the docker image network manager. I reports:

`Error: Connection activation failed: (5) IP configuration could not be reserved (no available address, timeout, etc.).`

```
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: From device-profiles (i.e. custom profile) manufacturer/device name
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-nas-wr01w-smart-plug
 > tuya-generic-nas-wr01w-smart-plug

Traceback (most recent call last):
  File "/src/get_input.py", line 253, in <module>
    cli(obj={})
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/src/.venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/src/.venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/src/get_input.py", line 232, in choose_profile
    device, profile = load_profile(profile_dir)
  File "/src/get_input.py", line 102, in load_profile
    with open(join(profile_dir, "device.json"), "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/work/device-profiles/tuya-generic-nas-wr01w-smart-plug/device.json'
==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
.Found access point name: "SmartLife-3EE0", trying to connect..
Error: Connection activation failed: (5) IP configuration could not be reserved (no available address, timeout, etc.).
.Found access point name: "SmartLife-3EE0", trying to connect..
```

zillion42, 2022-12-02:
Same problem:

.....Found access point name: "SmartLife-3EE0", trying to connect..
Error: Connection activation failed: (5) IP configuration could not be reserved (no available address, timeout, etc.).
.Found access point name: "SmartLife-3EE0", trying to connect..

```
tobi@introvision2:~/tuya-cloudcutter$ git pull
remote: Enumerating objects: 27, done.
remote: Counting objects: 100% (27/27), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 27 (delta 9), reused 22 (delta 4), pack-reused 0
Unpacking objects: 100% (27/27), 15.81 KiB | 899.00 KiB/s, done.
From https://github.com/tuya-cloudcutter/tuya-cloudcutter
   d013c30..8776957  main       -> origin/main
Updating d013c30..8776957
Fast-forward
 INSTRUCTIONS.md                              |  8 +++---
 common_run.sh                                |  8 ++++--
 device-profiles/README.md                    | 16 +++++++++++
 profile-building/generate_profile_classic.py |  9 ++++--
 run_detach.sh                                | 20 +++++++------
 run_flash.sh                                 | 42 +++++++++++++++-------------
 src/get_input.py                             | 42 ++++++++++++----------------
 7 files changed, 85 insertions(+), 60 deletions(-)
tobi@introvision2:~/tuya-cloudcutter$ ./run_flash.sh 
Checking UDP port 53... Occupied by systemd-resolve with PID 5547.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.26MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> f4a8e1262ec3
Step 3/10 : FROM base AS python-deps
 ---> f4a8e1262ec3
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 0366218cbb83
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 3c55c27a6d5e
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 877222ed0957
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> e72bd74d3d13
Step 8/10 : FROM python-deps AS cloudcutter
 ---> e72bd74d3d13
Step 9/10 : COPY src /src
 ---> fb43cf636530
Step 10/10 : WORKDIR /src
 ---> Running in eedc66c28ced
Removing intermediate container eedc66c28ced
 ---> 69fae37bc519
Successfully built 69fae37bc519
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: From device-profiles (i.e. custom profile) manufacturer/device name
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-nas-wr01w-smart-plug
 > tuya-generic-nas-wr01w-smart-plug

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
.....Found access point name: "SmartLife-3EE0", trying to connect..
Error: Connection activation failed: (5) IP configuration could not be reserved (no available address, timeout, etc.).
.Found access point name: "SmartLife-3EE0", trying to connect..
```

zillion42, 2022-12-02:
at least no 
Traceback (most recent call last):
  File "/src/get_input.py", line 253, in <module>

kuba2k2, 2022-12-02:
Then it's a problem beyond cloudcutter, we can't fix whatever's wrong with your docker installation.

zillion42, 2022-12-02:
thx a lot for your support... its Ubuntu 20.04.5 LTS standard,  I installed docker.io with apt but it also suggests snap docker, is there any difference ?

```
Command 'docker' not found, but can be installed with:

sudo snap install docker     # version 20.10.17, or
sudo apt  install docker.io  # version 20.10.12-0ubuntu2~20.04.1

See 'snap info docker' for additional versions.
```

zillion42, 2022-12-02:
can I buy you a coffee ?

Cossid, 2022-12-02:
Does it have more than one wireless network card? or is a virtualized network driver?

You can try passing in a specific wifi adapter, with the `-w` flag, `nmcli device status` should show you all devices available.

zillion42, 2022-12-02:
```
tobi@introvision2:~$ iwconfig
lo        no wireless extensions.

enp2s0    no wireless extensions.

wlx00aaab002c30  IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

docker0   no wireless extensions.
```

Cossid, 2022-12-02:
try running `sudo bash run_flash.sh -w wlx00aaab002c30`

If the device is virtualized, it may be missing the ability to create an AP mode, which is a requirement here.

zillion42, 2022-12-02:
weirdly, this worked before:

```
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
```

zillion42, 2022-12-02:
partial Success ! <3

```

tobi@introvision2:~/tuya-cloudcutter$ sudo bash run_flash.sh -w wlx00aaab002c30
[sudo] password for tobi: 
Checking UDP port 53... Occupied by systemd-resolve with PID 911.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.27MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> f4a8e1262ec3
Step 3/10 : FROM base AS python-deps
 ---> f4a8e1262ec3
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 0366218cbb83
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 3c55c27a6d5e
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 877222ed0957
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> e72bd74d3d13
Step 8/10 : FROM python-deps AS cloudcutter
 ---> e72bd74d3d13
Step 9/10 : COPY src /src
 ---> Using cache
 ---> fb43cf636530
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 69fae37bc519
Successfully built 69fae37bc519
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: From device-profiles (i.e. custom profile)
   By manufacturer/device name
   By firmware version and name
 > From device-profiles (i.e. custom profile)

[?] Select device profile: tuya-generic-nas-wr01w-smart-plug
 > tuya-generic-nas-wr01w-smart-plug

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
.....Found access point name: "SmartLife-3EE0", trying to connect..
Device 'wlx00aaab002c30' successfully activated with '854cb6fd-a295-461d-923d-1c6a9d0362e0'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/oQYjGy1rmQsp.deviceconfig
Saved device config in /work/configured-devices/oQYjGy1rmQsp.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Scanning for known AP SSID prfixes: "A-" "atomi-smart-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" "UltraPro-" 
Found access point name: "SmartLife-3EE0", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
............Found access point name: "SmartLife-3EE0", trying to connect..
Device 'wlx00aaab002c30' successfully activated with '854cb6fd-a295-461d-923d-1c6a9d0362e0'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlx00aaab002c30
Dec  2 19:38:45 dnsmasq[15]: started, version 2.80 cachesize 150
Dec  2 19:38:45 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Dec  2 19:38:45 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Dec  2 19:38:45 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlx00aaab002c30
Dec  2 19:38:45 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
nl80211: kernel reports: (extension) channel is disabled
Could not set channel for kernel driver
Interface initialization failed
wlx00aaab002c30: interface state UNINITIALIZED->DISABLED
wlx00aaab002c30: AP-DISABLED 
wlx00aaab002c30: Unable to setup interface.
wlx00aaab002c30: interface state DISABLED->DISABLED
wlx00aaab002c30: AP-DISABLED 
wlx00aaab002c30: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlx00aaab002c30 wasn't started
nl80211: deinit ifname=wlx00aaab002c30 disabled_11b_rates=0
Oh no, something went wrong with updating firmware! Try again I guess..
```

Cossid, 2022-12-02:
Are you in a region where channel 1 is not available?  If so, you can manually override the channel is `/src/setup_apmode.sh`

zillion42, 2022-12-02:
germany ?

Cossid, 2022-12-02:
Hmm, channel 1 should work everywhere.  This is certainly down to the driver for your wireless adapter the the host environment's network management, I'm afraid there is not much additional help we can offer here.  CloudCutter currently requires NetworkManager to operate the necessary AP mode.  You may have a wireless device that does not support promiscuous mode.

zillion42, 2022-12-02:
I used to be wlan0.
aircrack-ng also used to work with this adapter

```
tobi@introvision2:~/tuya-cloudcutter$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 009: ID 148f:2573 Ralink Technology, Corp. RT2501/RT2573 Wireless Adapter
```

zillion42, 2022-12-02:
[image]
I will take a break and try again tomorrow, thx a lot for your support ! Please bear with me, I'm trying ! 
Will report back here.

remlei, 2022-12-03:
> Please do a `git update`, I believe this was fixed earlier today.

thanls finally able to flash all of my tuya smartplugs.

closed 2022-12-03 by remlei

zillion42, 2022-12-03:
I have this situation now:

```
└─$ sudo ./run_flash.sh
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.02MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 70fa42c638b3
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> cd9186bd4747
Step 3/10 : FROM base AS python-deps
 ---> cd9186bd4747
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 0ff49417dd7b
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 540933d58616
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 630b8e2226a2
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 4dc9e7f21316
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 4dc9e7f21316
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 92f7556b328d
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 3a75b040b02f
Successfully built 3a75b040b02f
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231N_UG_1.15.125.bin
 > OpenBK7231N_UG_1.15.125.bin

[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Tuya Generic
   QIACHIP
   Qiachip
   STITCH
   Spectrum
   Stitch
   Topgreener
   TreatLife
 > Tuya Generic
   UME
   UltraPro
   Woox
   iHome
   ihome

[?] Select the article number of your device: LV300 LV400 LV480 LV600ST LED Square LED Ceilin[?] Select the article number of your device: NAS-WR01W Smart Plug
   LSPA9 Smart Plug
   LSPS5 Smart Wifi Switch 3-Gang
[... 228 lines cut ...]
Dec  3 17:48:01 dnsmasq[15]: query[A] m2.tuyaeu.com from 10.42.42.23
Dec  3 17:48:01 dnsmasq[15]: config m2.tuyaeu.com is 10.42.42.1
^CTraceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 371, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 221, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmare=True)
  File "/src/cloudcutter/__main__.py", line 186, in __configure_local_device_or_update_firmware
    tornado.ioloop.IOLoop.current().start()
  File "/src/.venv/lib/python3.9/site-packages/tornado/platform/asyncio.py", line 215, in start
    self.asyncio_loop.run_forever()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 601, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1869, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.9/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Oh no, something went wrong with updating firmware! Try again I guess..

```

Cossid, 2022-12-03:
The profile you chose is not compatible with your device.  If the AP prefix doesn't change to A-, it is not compatible.

zillion42, 2022-12-03:
hmm... I linked a picture to my pcb, wait, it's here
It says NAS-WR01W6-T V7

https://user-images.githubusercontent.com/36650866/204391975-50b75025-5755-4ab0-83db-c2f285407630.jpg

Cossid, 2022-12-03:
That does not mean it has the same firmware (or even the same chip).  When it comes to Tuya Generics, there are many devices that share a common model name that are not the same.  Nous usually is distinct from Tuya Generic as well.

zillion42, 2022-12-03:
you think I should try to flash it with OpenBK7231N_UG_1.15.125.bin with a TTL adaptor ? I found this thread in polish:
https://www-elektroda-pl.translate.goog/rtvforum/topic3917832.html?_x_tr_sl=auto&_x_tr_tl=de&_x_tr_hl=de&_x_tr_pto=wapp

Cossid, 2022-12-03:
You should know which chip is actually in the device first, does the chip read BK7231T or BK7231N?  You've been attempting to use a BK7231T profile this whole time.

zillion42, 2022-12-03:
ohh, no according to the polish thread it's a N. Even with my macro camera I couldn't make out what the chip actually read.
https://photos.app.goo.gl/dMd2XSHbiHZZK6MZ9

Cossid, 2022-12-03:
Can't help you there.  The thread you linked to has both T and N devices, and the pictures you supplied are not good enough to read any markings on your chip.

zillion42, 2022-12-03:
It's N.
Which profile should I use ?
[image]

zillion42, 2022-12-03:
I think it reads 
BEKEN 
BK7231N
NU1379FQ

Cossid, 2022-12-03:
Still depends on which firmware version it has on it.

You can serial flash without knowing though.

Cossid, 2022-12-03:
I would recommend you try "By firmware version and name" and try different versions of bk7231n oem_bk7231n_plug (device selection after doesn't really matter)

zillion42, 2022-12-03:
I somehow can't read the firmware version in the tuya app. It only shows me the available update version, not the module version, Is there maybe another way I could determine. I mean, I've been trying this for 2 days now, and if possible I would not like to open them all with a dremel and start soldering.

[12 more replies]