tuya-cloudcutter/tuya-cloudcutter #205: Unable to flash custom firmware - Could not load profile

christi-ko, 2022-12-02:
Just was able to use run_deatach.sh to cut my Aubess device from the cloud (Thank you for the new added profile ...).
 In the next step i tried to use run__flash.sh but not able to run it sucessfull.

`sudo ./run_flash.sh 'wlan0' 'aubess-20a-plug' 'OpenBK7231N_UG_1.15.125.bin'`

I am getting the following message at the end:

`Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Could not load profile aubess-20a-plug. Are you sure the profile file exists and is a valid combined JSON?
Oh no, something went wrong with running the exploit! Try again I guess..`

Can one of you give me a direction how to solve this topic, or what i am doing wrong?

kuba2k2, 2022-12-02:
Please run `git pull`, this has most likely been fixed few minutes ago.

Cossid, 2022-12-02:
Also, the input will have changed

Try `sudo ./ruh_flash.sh -w wlan0 -p aubess-20a-plug -f OpenBK7231N_UG_1.15.125.bin` after updating

christi-ko, 2022-12-02:
I think it did work sucessfull ...

`[DEVICE OTA] Responding to device OTA HTTP request range: bytes=456736-587247/587248
[I 221202 18:12:34 web:2271] 206 GET /files/OpenBK7231N_UG_1.15.125.bin (10.42.42.29) 45.36ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=456736-587247/587248
[I 221202 18:12:39 web:2271] 206 GET /files/OpenBK7231N_UG_1.15.125.bin (10.42.42.29) 34.36ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=456736-587247/587248
`

closed 2022-12-02 by kuba2k2