tuya-cloudcutter/tuya-cloudcutter #206: Occupied by systemd-resolve

TomyOne, 2022-12-04:
Well it will stop resolver and then it cannot resolve DNS name, hmmm...

`
tom@ubuntu22:~/tuya-cloudcutter$ ./run_flash.sh 
Checking UDP port 53... Occupied by systemd-resolve with PID 3481.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  22.93MB
Step 1/10 : FROM python:3.9-slim-buster AS base
Get "https://registry-1.docker.io/v2/": dial tcp: lookup registry-1.docker.io: Temporary failure in name resolution
Failed to build Docker image, stopping script
`

Cossid, 2022-12-04:
You're using self-hosted DNS?  This isn't stopping DNS clients, it's stopping a DNS server.

You can try disabling that check in [safety_checks.sh ](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/ee7451bc08cb081c40396b1e2f51c8b3a8c3e71e/safety_checks.sh#L67) by adding a # to the beginning of that line.

TomyOne, 2022-12-04:
Hi there, No I don't have self-hosted DNS.

Ah yea Thanks, disabling that check in safety_checks.sh worked.

Apparently it also stopped DNS Client in Ubuntu 22.04....

Cossid, 2022-12-04:
I've adjusted it to only check for MQTT for now, as that is the only one known to for sure cause problems.  We'll re-evaluate adding others back if we see more tickets about failed cutting.

closed 2022-12-04 by Cossid