tuya-cloudcutter/tuya-cloudcutter #207: FEIT bulb — SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC

caver01, 2022-12-05:
At long last, I feel good about the progress I have made setting up tuya-cloudcutter on a Pi3B+. I am trying to flash OpenBK firmware [OpenBK7231T_UG_1.15.133.bin]

After reading everything I can, setting up a brand new Pi image to get this going, I feel pretty good about my progress, and I think I am getting all the way to the actual exploit with my bulb—I think it is requesting the firmware update, but then I get the SSL error. Is there anything I can do? Has my bulb been updated to a newer SSL Key or something from Tuya cloud? I think I am really close!

Thanks. Hoping I can make this work.

`Dec  5 18:36:31 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.15
Dec  5 18:36:31 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 221205 18:36:31 iostream:1404] SSL Error on 12 ('10.42.42.15', 58463): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129) `

kuba2k2, 2022-12-05:
You're using an incorrect profile, the exploit didn't run at all. Find out your firmware version, [as instructed here](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#finding-your-device).

caver01, 2022-12-05:
Really? The bulb has bpa8000rgbw2 printed right on the bulb, so I am selecting by manufacturer FEIT bulb: feit-bpa8000rgbw2p

Perhaps the “2” vs “2p”

kuba2k2, 2022-12-05:
It's possible that the same model ships with different chips, or different firmware versions.

caver01, 2022-12-05:
Ok. I just checked via SmartLife app so I can try again using “by firmware” instead of by device manufacturer to see if that works. . .

caver01, 2022-12-05:
SmartLife app says it’s v1.2.16, but when I select that during the run_flash.sh, and pick FEIT, it only lets me pick OM60 RGBW CA AG 3 Bulb which. . . is not what my bulb has written on it.

kuba2k2, 2022-12-05:
1.2.16 is probably not WB2L, it's BK7231N. Try 1.2.16 / BK7231N version.

caver01, 2022-12-05:
It does go as far as Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/FAAXY8FRaBlO.deviceconfig

Then it reconnects
.Found access point name: "SmartLife-7B27", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.. . . .. . . . . . . .

if I restart the bulb at this point it will connect, but then goes to SSL error above.

It does have “WB2L” written on the module inside.

caver01, 2022-12-05:
I just tried again with SmartLife, re-connected this bulb (which again, is the FEIT BPA800. . .   Checked the firmware with smartlife app and it says 2.9.15!

Ok, so if I now try to use that firmware version, I don’t get to pick FEIT as the manufacturer.

Cossid, 2022-12-05:
If you're flashing a 3rd party firmware, it doesn't matter which mft/model you chose, that is only if you need to use it with Tuya firmware in cut mode.  Just select whatever is available.

caver01, 2022-12-05:
I think I figured that out—and just succeeded in OTA flash witth open BK v 1.15.133!

One learning for pi-specific setup:

do:
sudo raspi-config

open advanced options, network, and switch it to Network Manager (instead of DHCPD). This cleaned up some weird errors I was getting during setup.

Thanks for helping me figure this out. Very exciting to finally succeed here. Next for me is to load the proper tasmota-like config now that I can see my bulb using openBK device page

closed 2022-12-05 by caver01