tuya-cloudcutter/tuya-cloudcutter #208: building_profile_Ubuntu 22.04.1 LTS

GuidoMeinke, 2022-12-06:
Hi I've got a problem building the profile of my tuya based smart device.
here the issue:
@guido-Ubuntu:~/tuya-cloudcutter/profile-building$ python3 build_profile.py Lumary-Smart-Slim-Light-Panel-A3 eu
Traceback (most recent call last):
  File "/home/guido/tuya-cloudcutter/profile-building/build_profile.py", line 7, in <module>
    import pull_schema
  File "/home/guido/tuya-cloudcutter/profile-building/pull_schema.py", line 19, in <module>
    import sslpsk2 as sslpsk
ModuleNotFoundError: No module named 'sslpsk2'

@guido-Ubuntu:~/tuya-cloudcutter/profile-building$ python2 build_profile.py Lumary-Smart-Slim-Light-Panel-A3 eu
  File "build_profile.py", line 33
    print(f"[+] Processing {file=} as {base_name}")
                                                 ^
SyntaxError: invalid syntax

Can't install python

Cossid, 2022-12-06:
Please run `python2 build_profile.py -h` for information on firmware filename requirements.

GuidoMeinke, 2022-12-06:
guido@guido-Ubuntu:~/tuya-cloudcutter/profile-building$ python2 build_profile.py -h
  File "build_profile.py", line 33
    print(f"[+] Processing {file=} as {base_name}")
                                                 ^
SyntaxError: invalid syntax

Cossid, 2022-12-06:
Oh, you need python 3.x, 2 doesn't support strings in that format.

GuidoMeinke, 2022-12-06:
python3 build_profile.py -h
Traceback (most recent call last):
  File "/home/guido/tuya-cloudcutter/profile-building/build_profile.py", line 7, in <module>
    import pull_schema
  File "/home/guido/tuya-cloudcutter/profile-building/pull_schema.py", line 19, in <module>
    import sslpsk2 as sslpsk
ModuleNotFoundError: No module named 'sslpsk2'

Cossid, 2022-12-06:
I'll generate a profile from the other ticket you opened.
I'll also look into adding a pyproject.toml file to the profile builder directory to aid with pip package management.

If you still want to run, you likely need to do:
```pip install bk7231tools
pip install cryptodome
pip install sslpsk2
pip install capstone```

Don't recall if there are others or not.

GuidoMeinke, 2022-12-07:
> 

I'm a newbie in Linux, but when i try to install sslpsk2 it runs in error
  pip install sslpsk2
Collecting sslpsk2
  Using cached sslpsk2-1.0.2.tar.gz (2.1 MB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: sslpsk2
  Building wheel for sslpsk2 (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
...etc

same at cryptodome
 pip install cryptodome
ERROR: Could not find a version that satisfies the requirement cryptodome (from versions: none)
ERROR: No matching distribution found for cryptodome

Cossid, 2022-12-07:
Sorry, you likely don't need to sintall cryptodome via pip, I think that might actually be a stock package.

Not sure about sslpsk2, ever seen that error before, but I think sslpsk2 might not work with python 3.11 yet, so you'd need >=3.0 and < 3.11.

Cossid, 2022-12-07:
I have added pyproject.toml files and poetry lock files.
You should be able to install dependencies now with either `poetry install` or `pip install .`

closed 2022-12-07 by Cossid