tuya-cloudcutter/tuya-cloudcutter #209: Successfully detach LSPA9 Smart Plug but it keeps turning off after ~30s

dj-fiorex, 2022-12-07:
Hello to everybody, i successfully detach a LSPA9 Smart Plug, this is the log
```
[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
 > 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch   <----- Tried this but fail!

[?] Select the brand of your device: Tuya Generic
   Aubess
   Topgreener
 > Tuya Generic

[?] Select the article number of your device: LSPA7 Plug
 > LSPA7 Plug

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. 
Scanning for known AP SSID prfixes: 
.Found access point name: "SmartLife-3022", trying to connect..
Device 'wlp2s0' successfully activated with '41230d9b-925c-4d0c-80a6-31e79d3cfad1'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/u1M5tPRvKKA5.deviceconfig
Saved device config in /work/configured-devices/u1M5tPRvKKA5.deviceconfig
==> Turn the device off and on again once. 
Scanning for known AP SSID prfixes: 
.................................Found access point name: "A-3022", trying to connect..
Device 'wlp2s0' successfully activated with '4ee48e98-7d1e-4b9d-986e-ba0eb3424b7e'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
Using WLAN adapter: wlp2s0
Dec  7 14:55:42 dnsmasq[15]: started, version 2.80 cachesize 150
Dec  7 14:55:42 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Dec  7 14:55:42 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Dec  7 14:55:42 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp2s0
Dec  7 14:55:42 dnsmasq[15]: read /etc/hosts - 7 addresses
Configuration file: /dev/stdin
Using interface wlp2s0 with hwaddr 78:92:9c:ea:e1:c9 and ssid "cloudcutterflash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
Dec  7 14:55:48 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec  7 14:55:48 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp2s0) 38:1f:8d:e4:30:22 
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 tags: wlp2s0
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp2s0) 10.42.42.34 38:1f:8d:e4:30:22 
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 59 T2  10h30m
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  1 netmask  255.255.255.0
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 28 broadcast  10.42.42.255
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  3 router  10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option:  6 dns-server  10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlp2s0) 38:1f:8d:e4:30:22 
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 tags: wlp2s0
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 DHCPOFFER(wlp2s0) 10.42.42.34 38:1f:8d:e4:30:22 
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 next server: 10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  1 option: 53 message-type  2
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 54 server-identifier  10.42.42.1
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 51 lease-time  12h
Dec  7 14:55:51 dnsmasq-dhcp[15]: 1220565492 sent size:  4 option: 58 T1  6h
[... 33 lines cut ...]
{'type': '["timezone"]', 't': 1670424989}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1670424990, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}, 'timezone': {'ackId': '0-0', 'validTime': 1800, 'time': 1670424990, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}}, 't': 1670424990, 'success': True}
[I 221207 14:56:30 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=7GvvsLfXKPmtwKAIs8Vh&et=1&t=1670424989&v=2.0&sign=02fc8e218190813bdfe43a7935f56c27 (10.42.42.34) 92.29ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.uuid.pskkey.get&et=1&t=1670424992&uuid=u1M5tPRvKKA5&v=1.0&sign=80a6d953a83027ef849e532a540549b5
[LOG] ==== Client request body ===
{'t': 1670424992}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1670424993, 'success': True}
[I 221207 14:56:33 web:2271] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1670424992&uuid=u1M5tPRvKKA5&v=1.0&sign=80a6d953a83027ef849e532a540549b5 (10.42.42.34) 95.63ms
device_id='<device_id>', ip='<wrong_ip>', local_key='<local_key>'
Device should be successfully onboarded on WiFi AP!

```
two highlights:

1. I selected the profile based on the firmware preinstalled on the smart plug, i know that i have an LSPA9 
2. in the log file. when i say <wrong_ip> this is because that ip wasn't an ip of my wifi but the ip inside the linux ap, i don't know if this is relevant or not but i needed to reconnect to the right WiFi after the process with the tuya app

I can control the device with tinytuya, i also installed tuya-local in my home-assistant and i can turn on/off, read voltage but not read current and load.
But the main problem is that if i turn on the switch (with the physical button or with tuya-local) it automatically turn off after ~ 30 seconds

Anyone could help me? What is the best way in this case? stay with a detached fw or flash a custom one? and in this case witch one is better? 

Thanks to everybody

Cossid, 2022-12-07:
We're finding there are a lot of _different_ plugs with the model LSPA9 that use different power monitoring chips and pin configurations.  Custom firmware is _usually_ the way to go because then you can control the configuration instead of relying on similar Tuya models.  However you'll have to figure out the pin settings no matter which firmware you go with.

As for 'better', that's a relative term and you'll get biased answers.  Both currently available are great options, it's more of a preference thing.  ESPHome has stronger direct HA integration and more custom modules, whereas OpenBeken takes a more Tasmota-like approach where everything that comes with the firmware is what you get after configuration.

Cossid, 2022-12-07:
I can't speak with authority on that, unfortunately.

Cossid, 2022-12-07:
As to the strange control, I'm guessing the schema that got applied to your device doesn't match and there is an inverted pin and Tuya's firmware has an action triggered when it thinks something is being held down when it really isn't.  This is why 3rd part firmware is usually the better option if you don't have an exact device/schema match.

dj-fiorex, 2022-12-07:
Thanks @Cossid  i just tried to flash a custom FW, based on CB2S that seems the chip inside ([look here](https://zry.io/archives/799)) the process seems to end with success. i read in the terminal the progress of the download of the custom FW but there wasn't any text like "Complete" or "Success" the program simply terminated.

Now i tried to power off  and then power on the device but nothing happen. The device is not connected to the network and if put it in pairing mode i will see the network named "A-3022", that is the same name for when i used the detach process

Cossid, 2022-12-07:
Did you use the correct firmware?  Did you use the correct type of firmware (\_UG\_)?  CB2S would be BK7231N.  The fact you can still put it in pairing mode and get an `A-` means the new firmware did not validate.

dj-fiorex, 2022-12-07:
Yes i used CB2S as board and used the file with _ug_ in the name generated by platformio. Then i used as platform for cloudcutter the saved config of the detach. Maybe is this the error?

Cossid, 2022-12-07:
I'm not certain what that means.  I guess I have never detached first and then reused those values, I just go straight to flashing usually, personally.  You can just re-run the flash from scratch, and post the full log please.

dj-fiorex, 2022-12-07:
This is the log 
```
/run_flash.sh 
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Building cloudcutter docker image..
Sending build context to Docker daemon  23.04MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 174676d1b146
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> fc344b80d9b9
Step 3/10 : FROM base AS python-deps
 ---> fc344b80d9b9
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> d605a3599fe4
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 809cb709e7cb
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> a288d92ed67b
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 9da1901bd04b
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 9da1901bd04b
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 9869d2566f1b
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 9a4c40938676
Successfully built 9a4c40938676
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: bk7231n_app.ota.ug.bin
 > bk7231n_app.ota.ug.bin

[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 0.0.3 - BK7231N / oem_bk7231n_light_st[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_ceiling_[?] Select the firmware version and name: 1.0.0 - BK7231N / oem_bk7231n_garage_d[?] Select the firmware version and name: 1.0.0 - BK7231T / HYS_03_018_bk7231t_t[?] Select the firmware version and name: 1.0.1 - BK7231T / HYS_03_010_RGB_soq7i[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.1 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select the firmware version and name: 1.0.10 - BK7231N / bk7231n_lowpower_co[?] Select the firmware version and name: 1.0.10 - BK7231T / oem_bk7231s_light_t[?] Select the firmware version and name: 1.0.2 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.0.2 - BK7231T / bk7231t_common_user_[?] Select the firmware version and name: 1.0.2 - BK7231T / oem_bk7231s_light_mi[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.0.3 - BK7231T / oem_bk7231s_light_ty[?] Select the firmware version and name: 1.0.4 - BK7231T / oem_bk7231s_ty_ffc_n[?] Select the firmware version and name: 1.0.5 - BK7231N / oem_bk7231n_dctrl_sw[?] Select the firmware version and name: 1.0.6 - BK7231N / oem_bk7231n_strip_rg[?] Select the firmware version and name: 1.0.7 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.0.8 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.0.9 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.1.0 - BK7231T / HYS_03_018_bk7231t_t[?] Select the firmware version and name: 1.1.0 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_control_[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_rnd_swit[?] Select the firmware version and name: 1.1.2 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.1.3 - BK7231T / oem_bk7231s_irbox_mo[?] Select the firmware version and name: 1.1.3 - BK7231T / oem_bk7231s_switch_c[?] Select the firmware version and name: 1.1.4 - BK7231T / oem_bk7231s_ty_ffc_d[?] Select the firmware version and name: 1.1.5 - BK7231T / oem_bk7231s_rnd_zero[?] Select the firmware version and name: 1.1.6 - BK7231T / oem_bk7231s_light_ff[?] Select the firmware version and name: 1.1.6 - BK7231T / oem_bk7231s_switch1_[?] Select the firmware version and name: 1.1.7 - BK7231T / oem_bk7231s_light_pi[?] Select the firmware version and name: 1.1.71 - BK7231T / bk7231t_common_user[?] Select the firmware version and name: 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.6 - BK7231T / oem_bk7231s_light_ty
   1.1.6 - BK7231T / oem_bk7231s_strip_ty
   1.1.6 - BK7231T / oem_bk7231s_switch1_zero_kjd
   1.1.7 - BK7231N / oem_bk7231n_plug
   1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
   1.1.71 - BK7231T / bk7231t_common_user_config_ty
 > 1.1.8 - BK7231N / oem_bk7231n_plug
   1.1.8 - BK7231T / oem_bk7231s_rnd_switch
   1.1.9 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.2.1 - BK7231N / oem_bk7231n_control_switch
   1.2.16 - BK7231N / oem_bk7231n_light_ty
   1.2.7 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty

[?] Select the brand of your device: Tuya Generic
   Aubess
   Topgreener
 > Tuya Generic

[?] Select the article number of your device: LSPA7 Plug
 > LSPA7 Plug

[... 78 lines cut ...]
[LOG] ==== Client request body ===
{'t': 1670454473}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1670454474, 'success': True}
[I 221207 23:07:54 web:2271] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1670454473&uuid=EqU1xgGFbGNH&v=1.0&sign=36388ef2af1a5ccebd4664f59db93890 (10.42.42.34) 88.46ms
[... cut ...]
tuya.device.upgrade.status.update - response: {'success': True, 't': 1670454477}
[I 221207 23:07:57 web:2271] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=8eodnPMOYaw2I2aM8Tv0&et=1&t=1670454477&v=4.1&sign=49010e80880570b95e5e4d7dc06d1bc7 (10.42.42.34) 5.17ms
[I 221207 23:08:09 web:2271] 200 GET /files/bk7231n_app.ota.ug.bin (10.42.42.34) 11994.00ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-481039/481040
carmelo@carmelo-desktop:~/Desktop/tuya-cloudcutter$ 

```

dj-fiorex, 2022-12-07:
i think i bricked the device with this last attempt, now the smart plug doesn't even turn on/off and when plugged in there is a blu led

Cossid, 2022-12-07:
That last log looks like a successful flash.  Are you sure ESPHome isn't running/connecting and you just have an incorrect configuration?

dj-fiorex, 2022-12-07:
what do you mean with this last sentence?

dj-fiorex, 2022-12-07:
yeah, i mean with this simple config i should at least have the web ui no? 
no AP mode, it is connected to my wifi but the page at the ip address can't be reached

Cossid, 2022-12-07:
No, you don't have web_server or captive_portal enabled, so there is no webpage to visit.  You need to send an OTA with a new firmware with those features.  You're beyond CloudCutter here, so I'm closing this ticket.  Please move on to the ESPHome github or discord for ESPHome support.

closed 2022-12-07 by Cossid

dj-fiorex, 2022-12-08:
Thanks a lot for your time @Cossid, how can i add my device with a photo? So someone in the future will knows already what to do :)

Cossid, 2022-12-08:
We only add profiles that also come with a full working schema.  The issue here is your device doesn't have a full working schema, and the one you tried wasn't 100% compatible.  The only way we can add a new schema is with a full firmware dump where we can verify the schema.

For now, they anybody in a similar situation will need to do by firmware version and name like you did, and it will continue to guide them the same way it did you.

dj-fiorex, 2022-12-08:
Ok, good news is, i have another 3 of these, never connected never used so I will try again to dump the firmware