tuya-cloudcutter/tuya-cloudcutter #213: LSC Bulb with firmware Version 5.0.8

LuisThe0ne, 2022-12-08:
Hi!
I just tried to flash a LSC Filament Bulb (3001686-970709) but there is no exploit for firmware Version 5.0.8.
Is there any option for me to create a profile?
Thanks

LuisThe0ne, 2022-12-08:
I tried the dedicated Profile as well as the 2.6.9 Firmware Profile

LuisThe0ne, 2022-12-08:
I used 1.1.6 because that one bulb had an older firmware.

LuisThe0ne, 2022-12-08:
@kuba2k2 I did not find 5.0.8 that's the problem.

kuba2k2, 2022-12-08:
Let me rephrase it: where did you see 5.0.8? Can you post a screenshot?

If one bulb had 1.1.6, was the other one updated to 5.0.8, or did it come with 5.0.8 from the factory?

LuisThe0ne, 2022-12-08:
I did not find 5.0.8 anywhere.
But I have four bulbs one had 1.1.6 and the others have 5.0.8.

kuba2k2, 2022-12-08:
Let's try again: how do you know they have 5.0.8? Can you post a screenshot?

LuisThe0ne, 2022-12-08:
I am sorry that I was so stupid and I did not understand what ya mean.
[image]
I know it from the tuya app. :)

Cossid, 2022-12-08:
Can you confirm those 5.0.8 bulbs are actually beken modules?  LSC used to sell ESP8266 devices.

If you aren't able to disassemble to verify, you can usually check by putting the bulb in slow blink mode (6x on/off) and using a wifi scanner to the the mac address/bssid of the AP, and check it against a mac address registry.  It will either come up as "Tuya Smart Inc" or "Espressif Inc".  If it comes up as no match, it's probably Espressif, and you can subtract 0x2 from the first hex value and it will likely match Espressif.

LuisThe0ne, 2022-12-08:
Great, thanks!
It looks like I got an esp based device.
Any ideas on how to flash those?
Tuya convert doesn't do what it should.

Cossid, 2022-12-08:
Unfortunately, there is currently no solution for ESP devices which cannot work with Tuya-Convert.  Technically speaking, the ESP devices appear vulnerable to the same exploit CloudCutter uses, but there has been no progress on reverse engineering getting a usable process yet.

closed 2022-12-08 by Cossid