tuya-cloudcutter/tuya-cloudcutter #215: Flashing problem after dnsmasq.pid "Permission denied" error

sverd, 2022-12-08:
Hello,
I tried using cloudcutter on a more or less unbranded smart plug. Following the instructions I used the Tuya app to get the firmware version (1.1.8) en to "detach" the plug. 

When I try to flash the OpenBKN7231N firmware to the plug things start out as described. However at a certain point, just before the actual flashing should take place I get the message "dnsmasq: failed to open pidfile /src/dnsmasq.pid: Permission denied". Followed by setting up the accesspoint.
At that point everything just halts. The accesspoint seems to be available (checked with another device) but the plug does not connect. 

I think this is related to the dnsmasq error, but I have no clue what is causing the error and how it may be resolved. As there is no /src/dnsmasq.pid on my OpenSuSE Leap system I think it relates to a file in the Docker container but I 'm not sure.
Does anyone have an idea what might be going on?

Cossid, 2022-12-09:
Are you running elevated or as a regular user?  The file it is referring to should be from within the docker container, so you shouldn't see it.

I'm guessing you likely have udp port 53 (DNS server) or 67 (DHCP server) active on your host machine.  We had a check for that, but it appeared it may have been not just turning off dns servers, but also dns clients.  You can try [uncommenting lines 67 & 68 in safety_checks.py](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/bc4109501f4bf375ec23c8f8c91ece5f9754932b/safety_checks.sh#L67) and see if that resolves your issue or not.  You may have to re-start dns services when done.

Cossid, 2022-12-10:
I have re-enabled the safety checks for DNS/DHCP by default.  You will be prompted to shop those services before running CloudCutter.

closed 2022-12-10 by Cossid

X-Ryl669, 2023-03-11:
Doesn't work also for me.
When in docker, and trying to run `dnsmasq --no-resolv --interface=wls1 --bind-interfaces --listen-address=10.42.42.1 --except-interface=lo --dhcp-range=10.42.42.10,10.42.42.40,12h --address=/#/10.42.42.1 -x $(pwd)/dnsmasq.pid --log-dhcp --log-queries --log-facility=/dev/stdout` it hangs forever.

If I kill the dnsmasq process, it start the ap (but obviously, the device can't connect since there is no DHCP server).
If I start a DHCPd server on this interface that offers 10.42.42.10 up to 40 and then the device connects (visible with nmap and wireshark), but no firmware update happens.

Please notice that I've also disabled apparmor that had rules on  dnsmasq's rights to write anywhere, so it's not a apparmor deny going on here.

X-Ryl669, 2023-03-11:
If running the same command out of the container dnsmasq starts (on my host, not in the docker), I see progress, the device starts to emit HTTP requests but fails with SSL errors:
```
Using WLAN adapter: wls1
Configuration file: /dev/stdin
Using interface wls1 with hwaddr 1c:4b:d6:aa:28:d3 and ssid "cloudcutterflash"
wls1: interface state UNINITIALIZED->ENABLED
wls1: AP-ENABLED 
404 GET /generate_204 (10.42.42.11) 2.87ms
404 GET / (10.42.42.11) 0.89ms
404 GET /generate_204 (10.42.42.11) 1.25ms
404 GET / (10.42.42.11) 0.86ms
404 GET /gslb/gslb/getbucket.asp?list=thm.market.xiaomi.com%2Cthm.market.xiaomi.com&ver=3.0 (10.42.42.11) 2.22ms
SSL Error on 17 ('10.42.42.11', 57692): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /gslb/?ver=4.0&type=wifi&uuid=0&list=app.chat.xiaomi.net%2Capp.chat.global.xiaomi.net%2Cresolver.msg.xiaomi.net&sdkver=30&osver=25&os=MIX%202%3AV9.2.7.0.NDECNEK&mi=3&key=11cdc5de0ffba952ce7146afdb3e6447 (10.42.42.11) 0.76ms
404 GET /generate_204 (10.42.42.11) 0.97ms
404 GET / (10.42.42.11) 0.99ms
404 GET /generate_204 (10.42.42.11) 0.98ms
404 GET / (10.42.42.11) 0.87ms
SSL Error on 16 ('10.42.42.11', 57708): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 1.10ms
404 GET / (10.42.42.11) 1.02ms
SSL Error on 16 ('10.42.42.11', 57714): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57716): [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57718): [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57720): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57722): [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57724): [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57726): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57728): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57734): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57736): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57740): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 1.13ms
404 GET / (10.42.42.11) 0.85ms
SSL Error on 16 ('10.42.42.11', 57746): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 1.00ms
404 GET /generate_204 (10.42.42.11) 0.95ms
404 GET /generate_204 (10.42.42.11) 0.95ms
SSL Error on 16 ('10.42.42.11', 57764): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57768): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57772): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 1.71ms
404 GET / (10.42.42.11) 0.99ms
SSL Error on 16 ('10.42.42.11', 57778): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57780): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 1.33ms
404 GET / (10.42.42.11) 1.04ms
SSL Error on 16 ('10.42.42.11', 57792): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57796): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
SSL Error on 16 ('10.42.42.11', 57800): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
404 GET /generate_204 (10.42.42.11) 0.93ms
404 GET / (10.42.42.11) 0.88ms
SSL Error on 16 ('10.42.42.11', 57806): [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1129)
```

Cossid, 2023-03-11:
@X-Ryl669 I have no idea what you're doing here, that's not a Tuya device, the calls it is making are nothing like a Tuya device, and even reference xiaomi in one of the calls.  Cloudcutter kills dnsmasq on the host because it runs it's own version in the container, which does include DHCP.  If it's hanging, you have something conflicting or configured wrong.

kuba2k2, 2023-03-11:
I think the logs are from an Android device that's connected to the network:
`MIX%202%3AV9.2.7.0.NDECNEK` - this points to a Mi Mix 2 with MIUI v9.2.7.0 on Android Nougat...

The /generate_204 requests are also typical for Android devices.

X-Ryl669, 2023-03-12:
Well, sorry, I wasn't clear. Let me explain better:
1. I have a tuya device (IR remote with CBU v1.3 and firmware v2.0.0 as reported in Tuya/SmartLife app) and I've built a new firmware for it using LibreTuya (esphome). I want to flash it with my firmware. 
2. So I've run this repository's tuya cloud cutter script on my Arch Linux system. 
3. The process went smoothly until the part that should start an AP, and it failed exactly like this issue is describing: dnsmasq that's running in the docker instance issue a Permission denied while trying to write in `/whatever/path`
4. I found out that this is known and due to AppArmor preventing dnsmasq to write anywhere.
5. Yet, the process had failed. So I disabled AppArmor and re-run the script
6. Now, at the same position in the script (in the `setup_ap.sh` script in the docker process), it hangs. The dnsmasq in the docker container hangs so the script can't continue.
7. While hacking around, I've tried to kill the dnsmasq process to let the script continue. It does, but since there is no DHCP server running, the device can't connect.
8. So, I've started a DHCP server on the host (not the container) so it offers a lease for the device (and, as you've found out, I've used an Android phone to check if it worked by connecting to the created AP). 
9. I was able to check that the tuya device connected to the AP by running `nmap` on the host on the `10.42.42.0/24`subnet and found 2 devices (tuya and my smartphone)
10. I've captured the network traffic with wireshark on the host and I've posted the connections above. 
11. Yet, nothing happened (no firmware upgrade) so I had to kill the script
12. Next try, I've stopped the DHCP server, re-run the script until it hanged on dnsmasq again. This time I've runned the dnsmasq on the host when it happened (the host version doesn't hang) and killed the container hung dnsmasq's process
13. The container's AP was then created and the tuya device was connected (this time, I've tried without my phone)
14. While capturing the network traffic, I can see that the device is assigned an IP address (`10.42.42.24`). It's making a few ARP requests. But nothing happens: the device doesn't do anything else. I don't know if I have to reset it in that state.
15. I've scanned the opened port and only the port 6668 is opened on the device.

Right now, I don't know what to do next to make it flash its firmware. 

Is the dnsmasq software in the docker special ? 
BTW, I had to reset & hold the push button for it to blink fast and reset again to get my computer to connect to A-something, and the second time, it does not work like this, I have to reset X number of times before it goes back in fast blink mode and a new reset then for the computer to connect to it again.

Cossid, 2023-03-12:
There is nothing special about the dnsmasq software, it is bog standard.  I'm not sure what you mean by dnsmasq hanging, dnsmasq runs in the background along with hostapd and tornado web server.  The script waits for specific requests from the device, so it's probably not hung, it just hasn't made the correct requests yet (there are a few devices that behave like this).

Please run with the `-v` flag and post full logs, as it will give us a better understanding of what is and isn't happening.

Further, when you get to the point where dnsmasq and hostapd are running, if the device isn't connecting, try following these instructions:

On any device on the same network as you're device running this script, please log into the Smart Life app ('Try as Guest' works fine if you do not already have an account)
- Start the add device procedure (hit '+' in the upper-right and select 'Add Device')
- Under 'Add Manually' select 'Socket (Wi-Fi)'
- Enter your network credentials as instructed
- Do not follow any of the instructions about putting a device into pairing mode, instead select 'next' until it asks the status of the indicator and select 'Blink Slowly'
- Select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.

Sometimes we run into a bit of a chick and egg problem where the wifi device is both sending the instructions for the device to connect to wifi as well as needing to listen for new connections, and it can't do those at the same time, so it sends the connect data, then opens hostapd, and sometimes the device tries to connect before hostapd is fully operational.  Some Tuya devices will try to connect more than once or wait 60 seconds and try again, but some only try the one time and give up.  This process separates that responsibility so hostapd can just sit that and listen while your phone sends the "connect" instruction instead.

X-Ryl669, 2023-03-12:
What I mean by hang, is that the next command in the setup_ap.sh script isn't executed (so the control isn't returned to the bash script) so no hostapd is launched. If I launch the dnsmasq command inside the container, I don't get any bash prompt back (until I press Ctrl + C obviously). Yet, if I launch the same command on the host, I do get a prompt back and the log to stdout is being filled. I've no idea why it's struck, and since I don't have strace or gdb in the container, I can't figure out what's wrong.

Thank you for the updated instructions, I'll try them tonight and report any success or failure.

X-Ryl669, 2023-03-14:
Here's the log with -v: 
```bash
tuya-cloudcutter]$ sudo ./tuya-cloudcutter.sh -v
[!] WARNING: Selected wifi AP support: oui
AP support is manditory for tuya-cloudcutter to work.  If this is blank or 'no' your adapter doesn't support this feature.
Press any key to continue, or CTRL+C to exitBuilding cloudcutter docker image
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  23.18MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 5896ec7c9c41
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 1c5fd1b3c03b
Step 3/10 : FROM base AS python-deps
 ---> 1c5fd1b3c03b
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 1d7a2e61d29b
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 677b1a276043
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 292f9cb8404e
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> f649f31a0ed4
Step 8/10 : FROM python-deps AS cloudcutter
 ---> f649f31a0ed4
Step 9/10 : COPY src /src
 ---> Using cache
 ---> 975f921f0b08
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> b4278005e4b4
Successfully built b4278005e4b4
Successfully tagged cloudcutter:latest
Successfully built docker image
1) Detach from the cloud and run Tuya firmware locally
2) Flash 3rd Party Firmware
[?] Select your desired operation [1/2]: 2
Loading options, please wait...
[?] Select your custom firmware file: bk7231n_app.ota.ug.bin
 > bk7231n_app.ota.ug.bin

[?] How do you want to choose the device?: By firmware version and name
   By manufacturer/device name
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 2.0.0 - BK7231N / oem_bk7231n_irbox_mol_ty
   1.3.19 - BK7231N / oem_bk7231n_light_ty
   1.3.2 - BK7231T / oem_bk7231s_ceiling_light_ty
   1.3.2 - BK7231T / oem_bk7231s_dltj
   1.3.2 - BK7231T / oem_bk7231s_light_pir_globe
   1.3.21 - BK7231N / oem_bk7231n_light_ty
   1.3.5 - BK7231T / oem_bk7231s_dltj
 > 2.0.0 - BK7231N / oem_bk7231n_irbox_mol_ty
   2.0.0 - BK7231N / oem_bk7231n_strip_ty
   2.0.2 - BK7231N / bk7231n_common_user_config_ty
   2.0.6 - BK7231T / WB3S_RMD_Night_Light
   2.1.1 - BK7231T / HYS_03_018_bk7231t_r1e1zch4rds9ypqq
   2.1.6 - BK7231N / bk7231n_common_user_config_ty
   2.9.12 - BK7231T / oem_bk7231s_light_ty

[?] Select the brand of your device: Tuya Generic
   Aubess
[... 34 lines cut ...]
================================================================================

common.sh: ligne 50: service : commande introuvable
Scanning for open Tuya SmartLife AP
.......
Found access point name: "A-984C", trying to connect...
Périphérique « wls1 » activé avec « 852afc5c-7e75-4b26-aef1-04ba6cd14543 ».
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
Flashing custom firmware...

================================================================================
Wait for up to 10-120 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
================================================================================

Using WLAN adapter: wls1
```
It hangs here (in the container, it's struck on dnsmasq)

X-Ryl669, 2023-03-14:
Following your process worked! 

However, I needed to switch constantly between my home wifi and cloudcutterflash AP because tuya needs to connect to internet to download the pictures on the pages, and does not allow editing the SSID in their dialog. So the processed ended up as:
1. Connect to 'cloudcutterflash', add a WIFI socket
2. Edit the password to 'abcdabcd'
3. Click next
4. Connect to home's WIFI for ip connection
5. Select blink slowly
6. When asked to connect to the device's AP, connect to A-XXX
7. Profit.

Cossid, 2023-03-14:
Yeah, that sounds to me like AppArmor is interfering still.  Not much we can do about the host interfering.