tuya-cloudcutter/tuya-cloudcutter #216: Daybetter 800lm RGB+2700-6500K bulb (WB2L/BK7231T, Tuya 2.9.6)

dwildstr, 2022-12-10:
These are inexpensive E26 bulbs with Beken MCUs and RGBCT LEDs driven by a BP5758D chip. Confusingly, there are actually two different hardware builds inside of identical-looking bulbs; one version of the Daybetter bulb uses a CB2L/BK7231N and is still on 1.2 firmware. I'm including the 2MB .bin file and the json in the zip below.

[daybetter.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10198918/daybetter.zip)

Cossid, 2022-12-10:
Unfortunately, this is an invalid dump due to a bug in LightLeak.  Please run it again, but without hitting `read storage`.  `read flash` is the only option you should run.

dwildstr, 2022-12-10:
Thanks, this should be a cleaner dump.
[daybetter-2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10199085/daybetter-2.zip)

Cossid, 2022-12-10:
That one is correct.  It has been added to CloudCutter as `Daybetter A19 800lm RGBCT Bulb 2.9.6`, and I have renamed the other one to be `Daybetter A19 800lm RGBCT Bulb 1.2.16` so it is clearer up-front.

Between that and the filename checks, it should be harder to mix them up.

closed 2022-12-10 by Cossid

kuba2k2, 2022-12-10:
Sorry about this bug, I'll try to fix it ASAP.