tuya-cloudcutter/tuya-cloudcutter #217: TreatLife Smart Light Switch - TWB3S1-ZR

EliteScouter, 2022-12-14:
The device was paired. Full dump included.

[image]
[image]
[image]

[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10225018/dump.zip)

Cossid, 2022-12-14:
Do you know if this is the single-pole or 3-way version of this switch?

Cossid, 2022-12-14:
Little bit of searching showed this is the single-pole version.
This device has been added as TreatLife SS01S Single Pole Smart Light Switch

Thanks!

closed 2022-12-14 by Cossid