tuya-cloudcutter/tuya-cloudcutter #219: Lenovo Smart LED Lightstrip profile request

etnguyen03, 2022-12-16:
Hello, this is a dump of the [Lenovo smart LED lightstrip, 2m version](https://www.lenovo.com/us/en/p/accessories-and-software/smart-home-devices/smart-home-essentials_smart-lighting/zg38c03842). [Teardown here](https://www.elektroda.com/rtvforum/viewtopic.php?p=20335928).

Interestingly I tried to use `build_profile.py` and I got this error right after the script got a token -- maybe something to look into as well?

```
Traceback (most recent call last):
  File "/home/ethan/Documents/tuya-cloudcutter/profile-building/build_profile.py", line 41, in <module>
    pull_schema.run_directory(extracted_location)
  File "/home/ethan/Documents/tuya-cloudcutter/profile-building/pull_schema.py", line 344, in run_directory
    run(directory, output_file_prefix, uuid, auth_key, product_key, firmware_key, software_version, baseline_version, cad_version, cd_version, protocol_version, region, token)
  File "/home/ethan/Documents/tuya-cloudcutter/profile-building/pull_schema.py", line 264, in run
    response = connection.request(url, params, data, "POST")
  File "/home/ethan/Documents/tuya-cloudcutter/profile-building/pull_schema.py", line 58, in request
    result = json.loads(response_body)["result"]
  File "/usr/lib/python3.10/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.10/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Unterminated string starting at: line 1 column 11 (char 10)
```

[Lenovo_Smart-LED-Lightstrip-SE-243SC_led-strip.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10241818/Lenovo_Smart-LED-Lightstrip-SE-243SC_led-strip.bin.zip)

Cossid, 2022-12-16:
That error might be because you are not in the US region, and didn't pass an alternate region.  Usually you get a token expired error, but they may have a different response we haven't handled.

I was able to get a schema without issue against the US server with my US account.

Profile has been added as Lenovo SE-243SC Smart LED Lightstrip

etnguyen03, 2022-12-16:
Hmm weird because I am in the US region, with a US account, and I wasn't able to get a response.

Cossid, 2022-12-16:
@etnguyen03 Do you have Tuya services firewalled off by any chance?

Cossid, 2022-12-16:
I've added a friendlier error message, but the only way to get the error you got is if the script can't communicate with Tuya's API, or it received a response that Tuya's API would not send (they follow a specific structure for everything including failures).

closed 2022-12-16 by Cossid