tuya-cloudcutter/tuya-cloudcutter #220: LVGESS WHDZ03 Smart Plug - Got A-xxxx SSID but firmware upgrade not successful

jkwim, 2022-12-18:
I attempted flashing LVGESS WHDZ03 Smart Plug (Aliexpress) and the firmware upgrade step was not successful.

To define the issue I will use exerpts from the session screen log.

1. Firmware upgrade appears to get triggered:

`[I 221218 10:27:38 web:2271] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1671359257&uuid=bxZjtkeUCxVP&v=1.0&sign=8ecc076a0c4ac653836c1ab514065910 (10.42.42.27) 116.64ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671359260} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.get&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359259&v=4.4&sign=ef20c3f20c0021b74873932938cfabac
[LOG] ==== Client request body ===
{'type': 0, 't': 1671359259}
[LOG] ==== End client request body ===
tuya.device.upgrade.get - response: {'result': {'url': '[http://10.42.42.1:80/files/OpenBK7231N_UG_1.15.188.bin](http://10.42.42.1/files/OpenBK7231N_UG_1.15.188.bin)', 'hmac': 'BF88ABCC45CC6A7844943340C8F3A9998927D01DC82A120862F60CCAE7B3AE0A', 'version': '9.0.0', 'size': '593408', 'type': 0}, 'success': True, 't': 1671359261}
[I 221218 10:27:41 web:2271] 200 POST /d.json?a=tuya.device.upgrade.get&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359259&v=4.4&sign=ef20c3f20c0021b74873932938cfabac (10.42.42.27) 19.98ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.status.update&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359261&v=4.1&sign=6a94ea8edb667989068d1cee2ae3a1f2
[LOG] ==== Client request body ===
{'type': 0, 'upgradeStatus': 2, 't': 1671359261}
[LOG] ==== End client request body ===
!!! Endpoint default response not found - tuya.device.upgrade.status.update
tuya.device.upgrade.status.update - response: {'success': True, 't': 1671359261}
[I 221218 10:27:41 web:2271] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359261&v=4.1&sign=6a94ea8edb667989068d1cee2ae3a1f2 (10.42.42.27) 99.66ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.silent.get&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359270&v=4.4&sign=cb6047ba0f357865622688fd516fc60d
[LOG] ==== Client request body ===
{'subId': None, 't': 1671359270}
[LOG] ==== End client request body ===
tuya.device.upgrade.silent.get - response: {'result': {'url': '[http://10.42.42.1:80/files/OpenBK7231N_UG_1.15.188.bin](http://10.42.42.1/files/OpenBK7231N_UG_1.15.188.bin)', 'hmac': 'BF88ABCC45CC6A7844943340C8F3A9998927D01DC82A120862F60CCAE7B3AE0A', 'version': '9.0.0', 'size': '593408', 'type': 0}, 'success': True, 't': 1671359272}`

2. Server responding
`[I 221218 10:27:54 web:2271] 200 POST /d.json?a=tuya.device.timer.count&devId=LVpeLLrWLtPVnvjony8s&et=1&t=1671359274&v=4.0&sign=64e109968e5fb300c3a05d6fa5c47aaa (10.42.42.27) 103.98ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671359277} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[I 221218 10:28:02 web:2271] 200 GET /files/OpenBK7231N_UG_1.15.188.bin (10.42.42.27) 20279.96ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-593407/593408
Dec 18 10:28:07 dnsmasq-dhcp[15]: 1220565492 available DHCP range: 10.42.42.10 -- 10.42.42.40
Dec 18 10:28:07 dnsmasq-dhcp[15]: 1220565492 client provides name: wlan0
Dec 18 10:28:07 dnsmasq-dhcp[15]: 1220565492 DHCPDISCOVER(wlan0) fc:67:1f:c1:68:59
Dec 18 10:28:07 dnsmasq-dhcp[15]: 1220565492 tags: wlan0`

(Pardon me for the format. The line breaks dissapear when viewed for some reason.)

3. At the end the device did not get updated but when changed in to Access Point mode gives a "A-xxxx" access point and IP network 192.168.175.1

So basically the firmware update has not taken place.

1. Can somebody share a log from a successful attempt for me to compare?

2. I used profile tuya-generic-lspa7-plug and accordingly assumed BK7231N. I do not know the exact chip that this device has.

What is the way forward (I can bring up the access point A-xxxx)?

If I flash BK7231T firmware would I end up in a no-return situation?

Cossid, 2022-12-18:
More complete logs would be more helpful.  Even your response is chopped up in what appears to be an unusual way.

You got an A- prefix, the the exploit is successful, and whatever chip the profile you used has is what chip is inside your device.  If you flash the wrong firmware, you will get a device that will not respond and will require serial flashing to recover.
Where is your device in relation to the device running cloudcutter?  You may have a wireless range issue.  The fact you are getting dnsmasq/dhcp entries at the *end* indicates your device is likely dropping connection and reconnecting.

jkwim, 2022-12-18:
> More complete logs would be more helpful. Even your response is chopped up in what appears to be an unusual way.
>
> You got an A- prefix, the the exploit is successful, and whatever chip the profile [...]

Thanks. I will move the device closer to the server and try again.

jkwim, 2022-12-20:
> Where is your device in relation to the device running cloudcutter? You may have a wireless range issue. The fact you are getting dnsmasq/dhcp entries at the **end** indicates your device is likely dropping connection and reconnecting.

You were right!
After I moved the two devices closer the flashing was successful.
Thanks!

However I wonder what the following means (got this on both sessions failed and successful):

`!!! Endpoint default response not found - tuya.device.upgrade.status.update`

Cossid, 2022-12-20:
Some devices ask for a status update after requesting a firmware update, and some don't.  It's not critical it receives a valid response to that request.

closed 2022-12-20 by Cossid