tuya-cloudcutter/tuya-cloudcutter #223: Original Firmware Tuya Wifi Smoke Detector

ipn00tic, 2022-12-20:
Original Firmware Tuya Wifi Smoke Detector

Shop: https://pt.aliexpress.com/item/4000818367545.html?spm=a2g0o.order_list.order_list_main.71.2a8fcaa4xzrb6i&gatewayAdapt=glo2bra
[Smoke_DUMP.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10273061/Smoke_DUMP.zip)

Cossid, 2022-12-20:
Would it be possible get pictures of the actual module/chip in this unit?  This identifies as BL7231NL, where we have not seen one designated 'NL' before.  It appears to be using a standard N SDK, but with changed strings for NL, and it would be nice to know if we're dealing with something different here or not.  The bootloader looks like standard BK7231N.

I'll have a profile up shortly, but would appreciate verification on if it works or not if you haven't already serial flashed.

Cossid, 2022-12-20:
Added as Tuya Generic TYSM-01 Smoke Detector, please let us know if you're able to verify it as working as expected.

ipn00tic, 2022-12-20:
I already made the flash for openbeken but it has problems of autonomy, it is draining the batteries in less than 24 hours. I can go back to the stock firmware to try some alternative.

Cossid, 2022-12-21:
Yes, you should at least be able to run cut stock firmware now.

Closing as complete, please re-open if you do have any issues using the cloudcutter profile.

closed 2022-12-21 by Cossid

Cossid, 2022-12-21:
Also, for reference, this firmware is compiled as BK7231NL instead of the usual BK7231N.  It appears the L stands for Low-Power, so there are special things done to reduce power consumption.  Unfortunately, I can't provide much detail beyond that.