tuya-cloudcutter/tuya-cloudcutter #224: BlitzWill BW-CLT2 failed to build profile with haxomatic

blakadder, 2022-12-20:
As instructed by build_profile.py

`RuntimeError: Unknown pattern, please open a new issue and include the bin.`
[blitzwill_bw---clt2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10273385/blitzwill_bw---clt2.zip)

Cossid, 2022-12-21:
Profile builder has been updated, please try again.

blakadder, 2022-12-22:
I see the builder doesn't give the pin layout for this dump

Cossid, 2022-12-22:
Correct, there is no user_param_keys in storage or within the app, which means the settings for this are compiled directly into the code instead of through a configuration (this one has a very specific device configuration of `BL_RGBCW_LIGHT_12KEY_NOKEY_MCU` which means the firmware was compiled for this device only, not shared.).  There is no known way to handle these other than doing a normal new device procedure probing.

closed 2022-12-22 by blakadder