tuya-cloudcutter/tuya-cloudcutter #227: Profile Building error - Tuya-Generic_MiniSwitch_ZN330893-01

ElGranPepo, 2022-12-22:
Used Lightleak to extract the bin file, trying to create my own profile as I test along:

pi@cloudcutter:~/tuya-cloudcutter/profile-building $ python build_profile.py Tuya-Generic_MiniSwitch_ZN330893-01.bin
[+] Processing file='Tuya-Generic_MiniSwitch_ZN330893-01.bin' as Tuya-Generic_MiniSwitch_ZN330893-01
Missing app RBL container. Using a scan pattern instead
Traceback (most recent call last):
  File "/home/pi/tuya-cloudcutter/profile-building/build_profile.py", line 37, in <module>
    extract.run(file)
  File "/home/pi/tuya-cloudcutter/profile-building/extract.py", line 46, in run
    bk7231tools.__main__.dissect_dump_file(input)
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/__main__.py", line 213, in dissect_dump_file
    _, code = __scan_pattern_find_payload(dumpfile, partition_name=missing, layout=layout,
  File "/home/pi/.local/lib/python3.9/site-packages/bk7231tools/__main__.py", line 129, in __scan_pattern_find_payload
    raise ValueError(f"Could not find end of partition for {partition.name}")
ValueError: Could not find end of partition for app

Cossid, 2022-12-22:
Not the error you're experiencing (but would run into later), there should only be one underscore in the file name, separating the mfg and model, anything else should be dashes.

The error you are getting is actually from bk7231tools about not understanding the dump.  Are you sure you have a valid 2MiB dump?  Don't use the 'read storage' option, 'read flash' is all you should be using.

ElGranPepo, 2022-12-22:
I was playing with the options, does each button create a new file or is the original file get overwritten?  maybe have each button create a separate file.

the instructions say: 

 There should be no spaces, and both a manufacturer and model must be present, separated by an underscore (_). 

this would make me think you need two underscores.

Thank you

kuba2k2, 2022-12-22:
> I was playing with the options, does each button create a new file or is the original file get overwritten? maybe have each button create a separate file.

Since v0.6.0 they all write to the same file. So, in theory, pressing all of them will give you a complete file, just like pressing "Read Flash" only.

ElGranPepo, 2022-12-22:
the Pi is reading the dump but is erroring out during the process:

cloudcutter:~/tuya-cloudcutter/profile-building $ python build_profile.py Tuya-Generic_MiniSwitch_ZN330893-01.bin US
[+] Processing file='Tuya-Generic_MiniSwitch_ZN330893-01.bin' as Tuya-Generic_MiniSwitch_ZN330893-01
RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xdd20]
                extracted to /home/pi/tuya-cloudcutter/profile-building/Tuya-Generic_MiniSwitch_ZN330893-01
        0x129f0a: app - [encoding_algorithm=NONE, size=0xe17e0]
                extracted to /home/pi/tuya-cloudcutter/profile-building/Tuya-Generic_MiniSwitch_ZN330893-01
Storage partition:
        0x1ee000: 68 KiB - 7 keys
        - 'gw_bi'
        - 'user_param_key'
        - 'gw_wsm'
        - 'is_stride'
        - 'gw_di'
        - 'gw_ai'
        - 'wd_channel_mem'
                extracted all keys to /home/pi/tuya-cloudcutter/profile-building/Tuya-Generic_MiniSwitch_ZN330893-01/Tuya-Generic_MiniSwitch_ZN330893-01_storage.json
Storage area `user_param_key`:
        - found! Extracted to /home/pi/tuya-cloudcutter/profile-building/Tuya-Generic_MiniSwitch_ZN330893-01/Tuya-Generic_MiniSwitch_ZN330893-01_user_param_key.json
[!] Matched pattern for BK7231T version 2, payload version datagram
[!] Searching for intermediate address

Killed

Cossid, 2022-12-22:
> There should be no spaces, and both a manufacturer and model must be present, separated by an underscore (_).
> 
> this would make me think you need two underscores.

This states manufacturer and model should be separated by an underscore, how does that imply more than one?  I guess I'll try to rephrase it.  There should only be 1 underscore, separating manufacturer and model, everything else should be dashes instead of spaces.

> [!] Searching for intermediate address
> 
> Killed

What is the actual error?  That's never the last line to come out of haxomatic.

ElGranPepo, 2022-12-22:
> This states manufacturer and model should be separated by an underscore, how does that imply more than one? I guess I'll try to rephrase it. There should only be 1 underscore, separating [...]

nothing else is being presented, it just brings me back to the command line.  I did notice that my WinSCP connection times out while running the command.  Is this CPU intensive?  should this work on a PI or do i need a more powerfule system?

thanks

ElGranPepo, 2022-12-22:
Adding dump and pics:

[Generic-AP-SMT-Breaker02-1CH_ZN330893.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10288144/Generic-AP-SMT
[image]
-Breaker02-1CH_ZN330893.zip)

ElGranPepo, 2022-12-22:
Adding Dump and Pic
[Generic-AP-SMT-Breaker02-1CH_ZN330893.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10288216/Generic-AP-SMT-Breaker02-1CH_ZN330893.zip)

[image]

Cossid, 2022-12-22:
> > What is the actual error? That's never the last line to come out of haxomatic.
> 
> nothing else is being presented, it just brings me back to the command line. I did notice that my WinSCP connection times out while running the command. Is this CPU intensive? should this work on a PI or do i need a more powerful system?
> 
> thanks

Yes, on a low powered CPU like a raspberry pi, that process appears to take about 45 seconds.  It's mostly IO intensive I believe.  On a more powerful pc, it only takes 2-3 seconds.

ElGranPepo, 2022-12-22:
> Yes, on a low powered CPU like a raspberry pi, that process appears to take about 45 seconds. It's mostly IO intensive I believe. On a more powerful pc, it only takes 2-3 seconds.

OK any logs i can provide to troubleshoot?  was hoping to make this Pi dedicated to this (have a couple of devices i want to dump).

Will also try a pi4 and see if that makes a difference

Also uploaded dump, let me know if you have issues using it.

Thanks again!!

Cossid, 2022-12-22:
I added a message about it taking up to a minute on low powered CPUs.  My 45 second quote was running from a Pi 4B 4GB model.  I'm unsure how long it might take on a lower model.

I have added the profile for this device as Tuya Generic AP-SMT-Breaker02-1CH Relay

Cossid, 2022-12-22:
> OK any logs i can provide to troubleshoot? was hoping to make this Pi dedicated to this (have a couple of devices i want to dump).

Is it actually exiting back to bash without a message, or are you killing it with CTRL+C/X?  It's python so if it's crashing it should spit out a backtrace or an error we specifically raise at the very least.

ElGranPepo, 2022-12-22:
> Is it actually exiting back to bash without a message, or are you killing it with CTRL+C/X? It's python so if it's crashing it should spit out a backtrace or an error we specifically raise at the [...]

Its exiting to bash without a message.  

I took the SDCard out and moved it to a pi4b 4GB i have and the process worked. im guessing the Pi 3 is too underpowered to run the script.

Cossid, 2022-12-22:
Strange, there is no technical reason for that to happen that I know of.

ElGranPepo, 2022-12-22:
> I added a message about it taking up to a minute on low powered CPUs. My 45 second quote was running from a Pi 4B 4GB model. I'm unsure how long it might take on a lower model.
> 
> I have added the profile for this device as Tuya Generic AP-SMT-Breaker02-1CH Relay

thank you for uploading the profile.

Of note some interesting things I haven't read about:

- in order to get the device into pairing mode you have to turn the switch on/off 20 times.
  - first 10 times will set the device to pair in Bluetooth bode (rapid blinking light)
  - second set will turn device WIFI pairing on (slow blinking light)

Cossid, 2022-12-22:
If you pause for about 2 seconds after the 5th time, you should only need to do it a total of ten times.

There are two general methods for devices that need power cycles to get into pairing modes:
3x to get into quick pairing mode, wait about 2 seconds then 3x more to get into slow pairing mode
5x to get into quick pairing mode, wait about 2 seconds then 5x more to get into slow pairing mode

However, several firmwares has slightly different settings.  Some you don't need to pause in the middle at all, some you have to pause longer.

ElGranPepo, 2022-12-22:
> If you pause for about 2 seconds after the 5th time, you should only need to do it a total of ten times.
>
> There are two general methods for devices that need power cycles to get into pairing modes: [...]

doesn't seem to work that way, tried many times to get it to work with less clicks until I figured out how to get them to pair correctly.  

for uploading firmware, is there a preference between ESPHome and OpenBekken?

Cossid, 2022-12-22:
Some things require a longer pause, up to 5 seconds, but I've never seen anything with documentation other than 6x and 10x for the upper end.

As for ESPHome vs OpenBeken, it's a matter of preference.  ESPHome has more components and better HA integration, but requires up-front configuration.  OpenBeken is easy to get up with basic components, but doesn't have a lot of customization available.

tony-fav, 2022-12-22:
One of my bulbs needed 4x then pause (fast flash) then 4x again to get to AP. Think it was the Stitch bulbs

JesseMcL, 2022-12-22:
> Strange, there is no technical reason for that to happen that I know of.

@Cossid I ran into this yesterday, kernel oom_killer kills the process during haxomatic due to out of memory. Pi3 only has 1GB. For me i copied the dump/working files over to another system (Ubuntu WSL2 on windows) and it succeeded. Copy the files back to the pi3 and it works.

Cossid, 2022-12-23:
@JesseMcL Thank you for that data, I'll see if there is anything I can do to make it more memory efficient.

Cossid, 2022-12-23:
@ElGranPepo @JesseMcL I have removed old caching code from when haxomatic was still using arm assembly instruction conversions instead of binary bytematching.  On my RPi4B 4G, it now runs nearly instantaneously.  Please do a git update and try again on a Pi 3 and let me know how it goes.

JesseMcL, 2022-12-23:
> @ElGranPepo @JesseMcL I have removed old caching code from when haxomatic was still using arm assembly instruction conversions instead of binary bytematching. On my RPi4B 4G, it now runs nearly instantaneously. Please do a git update and try again on a Pi 3 and let me know how it goes.

@Cossid clean pull works beautifully on Pi3 now.

Cossid, 2022-12-23:
@JesseMcL Thanks for the confirmation!

Closing this a complete, between fixing haxomatic on pi3 and adding more documentation and errors for incorrect filenames.

closed 2022-12-23 by Cossid