tuya-cloudcutter/tuya-cloudcutter #228: Specturm NYMPHEA 24W

GalGavu, 2022-12-22:
Hi,
I bought this one https://spectrumsmart.pl/pl/p/Oprawa-LED-Wi-FiBluetooth-NYMPHEA-24W-1840lm-CCT-DIM-230V-130st-IP20-390x70mm-BIALY-okragla/87

and I was naive that one of the profiles there would work with OpenBeken. I run cloudcutter and basically from logs I think it flashed openBeken because script finished with:
```
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-464975/464976
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1671744597} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[I 221222 21:29:59 web:2271] 206 GET /files/OpenBK7231T_UG_1.15.221.bin (10.42.42.15) 57.88ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=359968-464975/464976
[I 221222 21:30:04 web:2271] 206 GET /files/OpenBK7231T_UG_1.15.221.bin (10.42.42.15) 54.68ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=359968-464975/464976
[I 221222 21:30:07 web:2271] 206 GET /files/OpenBK7231T_UG_1.15.221.bin (10.42.42.15) 55.57ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=359968-464975/464976
[I 221222 21:30:13 web:2271] 206 GET /files/OpenBK7231T_UG_1.15.221.bin (10.42.42.15) 52.80ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=359968-464975/464976

kuba2k2, 2022-12-22:
If it still can be put in AP mode and shows A- SSID, you didn't break anything and you can just run cloudcutter again.

GalGavu, 2022-12-22:
@kuba2k2 so why it's not ending up with OpenBeken AP?

kuba2k2, 2022-12-22:
Because the firmware didn't flash successfully. Why - is already above my knowledge level.

GalGavu, 2022-12-22:
Ah so basically I can run cloudcutter but not do much with it right now :) I'll wait - maybe someone will help with flashing it successfully.

kuba2k2, 2022-12-22:
It's most likely a bad wifi connection or just a random issue. You can try again and see if it helps.

blakadder, 2022-12-22:
sometimes you need to try quite a few times until it properly takes

Cossid, 2022-12-23:
Usually (but not always), that means the device you are trying to flash is too far away from the device sending the firmware and is suffering from a poor signal.  There are indeed some devices that just take multiple tries no matter how physically close they are though.  As others said, you should be able to retry until successful, but keep the devices as close to each other if possible.

bart-sofomo, 2022-12-23:
Brought laptop as close as possible to light and it worked :) Thanks for helping me out -> now figuring out pins.

GalGavu, 2022-12-23:
Brought laptop as close as possible to the light - flashed perfectly with Openbeken. Thanks for helping me out.

closed 2022-12-23 by GalGavu