tuya-cloudcutter/tuya-cloudcutter #229: support other vendors as tuya?

WiredLife, 2022-12-23:
i try to build a profile for a ledvance bulb and ran into this error:
`pi@raspberrypi:~/tuya-cloudcutter/profile-building $ python build_profile.py LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.bin
[+] Processing file='LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.bin' as LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xdd40]
                extracted to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
        0x129f0a: app - [encoding_algorithm=NONE, size=0xfa220]
                extracted to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
Storage partition:
        0x1ee000: 68 KiB - 6 keys
                extracted all keys to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb_storage.json
[+] Searching for known exploit patterns, this may take up to a minute on low powered CPUs
[+] Matched pattern for BK7231T version 2, payload type datagram
[+] Searching for datagram payload address
[+] datagram payload address gadget (THUMB): 0xAAE15
[+] Searching for finish address
[+] Finish address gadget (THUMB): 0xB7F5D
[+] uuid: 1b16da1cb27fb3ea
[+] auth_key: wzIMV0Y7uZ4lv1mhlAxj1DJy0O8VSdAc
[+] ap_ssid: LDV SMART+
[+] swv: 1.0.4
[+] bv: 40.00
[+] firmware key: keywmrsuax733arg
[+] product key: keywmrsuax733arg
[+] SDK: 1.0.8
[+] Device class: oem_bk7231s_light4_4to5_deppon60A
[!] No token provided.  On the same network, please log into Smart Life, start the add device procedure, select 'Socket (Wi-Fi)', enter your network credentials, next until it asks the status of the indicator and select 'Blink Slowly', select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.  A new token should be sent to your network, and this script will continue.
[!] Note: this will join the device to your account.  You can safely delete it afterwards.
[+] Waiting for multicast token from app...
Using token: 3QOoANDC product_key: keywmrsuax733arg firmware_key: keywmrsuax733arg
{'t': 1671783420, 'success': False, 'errorCode': 'NOT_EXISTS', 'errorMsg': 'Product Not exist'}
[!] Unable to build complete profile as schema remains missing.`

after using the ledvance app i got this:

`pi@raspberrypi:~/tuya-cloudcutter/profile-building $ python build_profile.py LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.bin
[+] Processing file='LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.bin' as LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xdd40]
                extracted to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
        0x129f0a: app - [encoding_algorithm=NONE, size=0xfa220]
                extracted to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb
Storage partition:
        0x1ee000: 68 KiB - 6 keys
                extracted all keys to /home/pi/tuya-cloudcutter/profile-building/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb_storage.json
[+] Searching for known exploit patterns, this may take up to a minute on low powered CPUs
[+] Matched pattern for BK7231T version 2, payload type datagram
[+] Searching for datagram payload address
[+] datagram payload address gadget (THUMB): 0xAAE15
[+] Searching for finish address
[+] Finish address gadget (THUMB): 0xB7F5D
[+] uuid: 1b16da1cb27fb3ea
[+] auth_key: wzIMV0Y7uZ4lv1mhlAxj1DJy0O8VSdAc
[+] ap_ssid: LDV SMART+
[+] swv: 1.0.4
[+] bv: 40.00
[+] firmware key: keywmrsuax733arg
[+] product key: keywmrsuax733arg
[+] SDK: 1.0.8
[+] Device class: oem_bk7231s_light4_4to5_deppon60A
[!] No token provided.  On the same network, please log into Smart Life, start the add device procedure, select 'Socket (Wi-Fi)', enter your network credentials, next until it asks the status of the indicator and select 'Blink Slowly', select 'Go to Connect', then in your wifi selection screen, hit the back button to return to Smart Life.  A new token should be sent to your network, and this script will continue.
[!] Note: this will join the device to your account.  You can safely delete it afterwards.
[+] Waiting for multicast token from app...
Using token: ttldU47b product_key: keywmrsuax733arg firmware_key: keywmrsuax733arg
[!] Unable to get a response from Tuya API, or response was malformed.
[!] http://a.tuyaeu.com/d.json must not be blocked in order to pull data from the Tuya API.
Traceback (most recent call last):
  File "/home/pi/tuya-cloudcutter/profile-building/tuya_api_connection.py", line 35, in request
    response_body_json = json.loads(response_body)
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Unterminated string starting at: line 1 column 11 (char 10)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/tuya-cloudcutter/profile-building/build_profile.py", line 41, in <module>
    pull_schema.run_directory(extracted_location)
  File "/home/pi/tuya-cloudcutter/profile-building/pull_schema.py", line 255, in run_directory
    run(directory, output_file_prefix, uuid, auth_key, product_key, firmware_key, software_version, baseline_version, cad_version, cd_version, protocol_version, token)
  File "/home/pi/tuya-cloudcutter/profile-building/pull_schema.py", line 175, in run
    response = connection.request(url, params, data, "POST")
  File "/home/pi/tuya-cloudcutter/profile-building/tuya_api_connection.py", line 45, in request
    if (response_body_json is not None):
UnboundLocalError: local variable 'response_body_json' referenced before assignment`

Cossid, 2022-12-23:
Can you supply the bin for this?  LSC has a few devices that appear to only be registered in the CN region.  From what I can parse from your logs, LSC is generating a token for the EU region, and the machine you are running profile builder on couldn't communicate in the standard way.  I'm not sure if LSC is using a different endpoint or a different protocol, and I can't determine that from logs alone.

WiredLife, 2022-12-23:
its a ledvance device as the log suggest ;)
here you go
[LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10296829/LEDVANCE_AC33911---A60-B27-MultiColor-Bulb.zip)

Cossid, 2022-12-30:
I spent a bit of time, but can't figure out what api address this must be using.  I tried using the Tuya API with the LEDVANCE app, and it didn't work, which means there must be another API endpoint, but I couldn't find it.  I couldn't see anything different in the data the device was sending/requesting, so I don't know what is needed to support LEDVANCE.  They are the only vendor that seems to lock their stuff out of Smart Life app that I've seen so far.

Cossid, 2023-01-16:
Adding #wontfix to this one and closing until better information is discovered.
I tried using the LEDVANCE app and cannot add the bulb there either, even via bluetooth.  Unfortunately, the only way I know of to handle these devices is adding with an active account from China, which I do not have, and cannot automate.

Cossid, 2023-01-16:
If you would like to flash this device with 3rd party firmware, you can use this unverified profile (it will work for flashing, but it may or may not work for regular cutting use with Tuya firmware).  I applied a generic RGBCT bulb schema, so it could work with regular cutting, but I have no easy way to verify.  Flashing should go fine.

[ledvance-ac33911-rgbct-bulb.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10421619/ledvance-ac33911-rgbct-bulb.zip)
Extract this to your device-profiles directory and use `-p ledvance-ac33911-rgbct-bulb`

closed 2023-01-16 by Cossid

Cossid, 2023-02-07:
A full verified profile for this device has been added as LEDVANCE SmartPlus AC33911 RGBCT Bulb.  I found the script we have in place actually works, but instead of opening the Tuya Smart Life app, you MUST open the LEDVANCE Smart+ Wifi app instead.  I wasn't able to verify before because I couldn't locate the app, but I have since found it is region locked out of the US, and I had to download it from an alternate source, in which I found simply using the token from the LEDVANCE app was sufficient.